AltBeacon/android-beacon-library #658: Problem detecting beacons in background at low latency

Salvatore-Giordano, 2018-03-08:
My app should detect an iBeacon with low latency (~20 seconds max).

I'm using a BootstrapNotifier in order to receive callbacks, but I have some problems. On android 8 I get this log 
```
W/JobInfo: Specified interval for 1 is +5m10s0ms. Clamped to +15m0s0ms
03-08 10:23:29.119 8332-8430/it.tecnoinnovazioni.signal.staging W/JobInfo: Specified flex for 1 is 0. Clamped to +5m0s0ms
03-08 10:23:29.119 8332-8430/it.tecnoinnovazioni.signal.staging W/JobInfo: Specified interval for 1 is +5m10s0ms. Clamped to +15m0s0ms
03-08 10:23:29.119 8332-8430/it.tecnoinnovazioni.signal.staging W/JobInfo: Specified flex for 1 is 0. Clamped to +5m0s0ms
```

I have also some problems on Huawei devices like issue #554.

Any suggestion? Maybe I'm not using the right way to make such a scan.

The aim is to notify the app when detecting a beacon and then, after the user opens the app and performs a task, restart the scan.

davidgyoung, 2018-03-09:
@Salvatore-Giordano, for project-specific help, can you please post a question on StackOverflow.com?  We need to reserve this forum for bug reports and feature requests with the library itself, so I'm closing this issue.

There are limitations in how often you can do background scans on Android 8, which is generally limited to every 15-25 minutes.  For details as to why this is true, see: http://www.davidgyoungtech.com/2017/08/07/beacon-detection-with-android-8

On Huawei, a vendor-specific Android modification does two things (1) kills background services after a certain amount of time in the background.  (2) Blocks bluetooth scans after they have been started a certain number of times in the background.  I still have no good fix for the Huawei issue, but I am open to suggestions in #554 

You may wish to explore implementing a ForegroundService to allow more frequent bluetooth scanning in the background.

For further project-specific discussion, please open a question on StackOverflow.com.

closed 2018-03-09 by davidgyoung

Salvatore-Giordano, 2018-03-15:
Ok thank you @davidgyoung . 
Do you know if exists some official documentation for this kind of issues? We need to explain the problem to our customers ..

davidgyoung, 2018-03-15:
There is no documentation more official than the two links I shared.  This is an open source project, so for issues, GitHub issue threads are as official as it gets.