AltBeacon/android-beacon-library #659: Too many wakeups on Android Oreo

tomacco, 2018-03-13:
Android vitals includes a section in the Google Developer Console that warns you about "excessive application wake ups". I understand that the library heavily relies on AlarmManager in versions prior Android Oreo, and `JobScheduler` in Oreo and greater versions. I also understand that the library has an "Auto wakeup" functionality (alarms every 5 mins) that can be used by holding a reference to an instance of the `RegionBootstrap` class. Please correct me if any of my assumptions are wrong or incomplete.

We use the AltBeacon library as a dependency of our own library, our library allows you to configure content from cloud and trigger it based on geofences or beacons (plus all analytics related to this, plus other features not relevant to this issue).

So in our current implementation we **do not use** de `RegionBootstrap` wake up functionality, (we completely rely on geofences to wake up the application and start scanning ), however, one of our clients has sent us the following report:

(Wakeups from Alarm Manager per hour).

[image]

In our library, we extensively use the ` GcmNetworkManager` class, for sync tasks, which internally uses `JobScheduler` in API 23+, but we never use AlarmManager directly at all.

Is there something we are not configuring properly in the AltBeacon Library?
Seems very strange that precisely on Oreo, where I understand AltBeacon extensively uses JobScheduler, is where there are more wake ups (compared to other versions).

Many thanks

davidgyoung, 2018-03-13:
The AlarmManager code is still active and exercised on Android 8, although it doesn't really do anything useful on that OS version.  It has not been removed simply because disabling it for Android 8 only requires more code, and in an attempt to keep things simple it is still there.  I had assumed it would not hurt anything.

I cannot think of any reason the counts should be higher on Android 8 than other OS versions.  An alarm gets set once for 5 minutes in the future when scanning, then rescheduled 

There should

  A new alarm is

tomacco, 2018-03-19:
@davidgyoung I'm investigating with the owner of the app. I'll keep you posted.

tomacco, 2018-05-23:
@agap Not really. Unfortunately our software is being used in a customer app, and they didn't say anything more regarding this, so we do not have any more info.

agap, 2018-05-23:
@tomacco the reason I was asking you about it is that I also found the similar case [on stackoverflow](https://stackoverflow.com/questions/48956533/excessive-wakeups-warned-by-android-vitals-for-google-play-services-geofences) and the common thing between yours, mine and the issue found on SO is that we were using geofences. I am under the impression that this spike in wakeup alerts could be attributed to geofence apis, not the altbeacon.

davidgyoung, 2019-01-10:
closing due to inactivity

closed 2019-01-10 by davidgyoung

DanielRFG, 2019-11-14:
StartupBroadcastReceiver is invoked each time passive scanner reach some Beacon. So if you configure scann interval to e.g 10 seconds in a background mode then you can see  excessive wake ups, because passive scanner will present result using StartupBroadcastReceiver each 10 seconds. David, Am i right? if actually that is, then you have to increase scanning time in your scanner configuration, but remmember... You will get scanning results from time range you configured. In some cases beacons from eg. 30 min. scanning range will useless. Generally when you need assign current location to a beacon scan result event.

Sorry for the grammar :)

davidgyoung, 2019-11-15:
@DanielRFG, this should not be a problem.  As soon as the first background wakeup comes, the library will start an immediate scan job to process the passive scan results.  This stops the passive scan and starts an active scan for the length of the background scan period.  At the end of that background scan period, passive scanning is disabled -- we know there are beacons around so there is no point in doing a passive scan to look for them.  Passive scanning is not resumed until a background scan period ends with no beacons visible.

DanielRFG, 2019-11-20:
@davidgyoung I have recently analyzed this process and it looks like passive scanning is started every time in OnStopJob (line 207 ScanJob.java) and when Scan job runtime expired (line 127 ScanJob.java). While the active scan is running, the results of the passive scan supplied by the StartupBroadcastReceiver are provided simultaneously. After onStopJob called,
passive scanning is restarted again. Note that, each time StartupBroadcastReceiver is called  with scan result cause scheduling new ScanJob to process passive scan result (there is 10s break in scheduling next scan job -> MIN_MILLIS_BETWEEN_SCAN_JOB_SCHEDULING = 10000L;).
 In extreme cases, StartupBroadcastReceiver can be called without longer interruption.

DanielRFG, 2019-11-20:
@davidgyoung Well you've right, but try to observe logs when you set "background scan on" interval to 10 seconds and break e.g 50 s. (but "background scann off" interval doesn't metter in this case  ). Then you can see that after scan job is finished, passive scanner i started again ( even if background scan period ends with visible beacons ). So in the worst case you can see that StartupBroadcastReceiver is called by the system, more or less, every 10 seconds.  I think the best solution here will be to protect scan job against too short scanning period or insert additional setting which configure ScanJob running timeout after wake up. Then you will can set scan job active e.g for 5 min but results will be presented every "bacground scan on" interval (in this case 10 seconds). then StartupBroadcastReceiver will be called every 5 min. I'm not sure im right, but it is worth to discuss this topic

davidgyoung, 2019-11-21:
I still don't believe this statement is true: "you can see that after scan job is finished, passive scanner i started again ( even if background scan period ends with visible beacons )"

In ScanJob.java, the `startPassiveScanIfNeeded` method will only start a passive scan in the else block below:

```
            for (Region region : regions) {
                RegionMonitoringState state = mScanState.getMonitoringStatus().stateOf(region);
                if (state != null && state.getInside()) {
                    insideAnyRegion = true;
                }
            }
            if (insideAnyRegion) {
                // TODO: Set up a scan filter for not detecting a beacon pattern
                LogManager.i(TAG, "We are inside a beacon region.  We will not scan between cycles.");
            }
            else {
                ...
            }
```

That else block only executes if every single monitored region is in an outside state.  If a beacon matching one of these regions was detected, that region would have an inside state. 

The case you describe may be possible, but only in a very specific case:  If a beacon matching a registered beacon parser  is detected, but it does not match any of the regions you have set up due to non-matching identifier patterns.

For the more general case, I still don't believe this will happen.

DanielRFG, 2019-11-21:
You've right, I forgot to instatiate the RegionBootstrap in Application oncreate body :)

DanielRFG, 2020-01-29:
@davidgyoung you've write that "As soon as the first background wakeup comes, the library will start an immediate scan job to process the passive scan results. This stops the passive scan and starts an active scan". But do we have any guarantee that Job will start immediately when app is in background? If Job will not start, then many wakeups are possible, beacause of active passive scaner. Maybe the better idea will be to stop passive scan in StartupBroadcastReceiver?

davidgyoung, 2020-01-29:
In my tests, yes, if a BroadcastReceiver fires, it can start an immediate Job.  That Job may take a finite amount of time to start (not sure the outermost amount, but maybe 500 ms?) and in that time there could be other packets detected that fire the BroadcastReceiver again.  In this case, the BroadcastReceiver is designed to queue these up so all get processed when the Job actually starts. 

If my above description is accurate, then I don't think there is any significant upside to stopping the scan in the Broadcast receiver, because additional wakeups over a half a second won't have significant performance impact.  But stopping the scan would would cause the additional packets to get missed, so I think there is a real downside without a significant upside.