AltBeacon/android-beacon-library #662: Foreground service mode

twanschik, 2018-03-24:
Hi, this is a propose for a new feature. 
It would be great if the AltBeacon library provided a foreground-service-scanning mode for library users similar to Estimote's `withScannerInForegroundService(Notification notification)` method. This could start the scanning proces wrapped in foreground service and leaves the decision to do so to the user of the library.

Is support for such a mode already planned and if yes is there some timeline?

Thanks

davidgyoung, 2018-03-24:
Thanks for the suggestion.  Yes, I am experimenting with such a a feature now.  The main trick is figuring out a way to fit it in with the existing library APIs so it is configurable without causing complications.  

No timeline yet.  In the meantime, you may be able to accomplish the same by starting a dummy foreground service in your app, and letting the library run ists scanner as a background service.  I have not tested this yet, but having one foreground service may keep the background service alive on Android 8.

twanschik, 2018-03-24:
Thanks for the answer David. I already tried to start a foreground service however I had no luck. The AltBeacon service still runs in background with all restrictions applied.

I already thought that it will be difficult to integrate sich a mode into AltBeacon. 

Just out of curiosity: what are the main points/problems one has to think of. Maybe I will try it myself and with a little bit of help I could get to a result pretty fast.

davidgyoung, 2018-03-24:
The trick is that the library may be configured a number of different ways -- as a scanning service in a separate process, as a (background) scanning service in the same process, or as a scheduled job (by default on Android 8+).  Configuration state is synchronized between the BeaconManager and these other process using multiple techniques.  The trick is to add yet another configuration setting in a way that does not overly complicate this process and break the other options.  My preference would.be to make a foreground service configuration exclusively in the app's AndroidManifest.xml declaration to keep things simple and unobtrusive.  But I have had trouble accessing app-specific resources declared in the manifest (e.g. intents/icons/activities) from an embedded Android library whose code exists in it's own aar file and has it's own classloader.

davidgyoung, 2019-01-10:
Requested featured added in version 2.15

closed 2019-01-10 by davidgyoung