AltBeacon/android-beacon-library #665: Passing null distance model update url prevents monitor callbacks from firing

kmerrell42, 2018-04-03:
### Expected behavior
Calling BeaconManager.setDistanceModelUpdateUrl(null); before adding a MonitorNotifier to the BeaconManager should force the BeaconManager to use the baked in model-distance-calculations.json file and monitoring callbacks should fire normally.

### Actual behavior
I see the following in the logs indicating that setting it to null is indeed triggering the mechanism to use the default model mapping, but I never receive Monitor.Notifier.didEnterRegion(region) callbacks.
_ModelSpecificDistanceCalculator: Cannot find match for this device.  Using default_

When I use a valid, non-null url, everything works as expected.

### Steps to reproduce this behavior

1. In onCreate() of the Application class, Instantiate the BeaconManager:  `BeaconManager beaconManager = BeaconManager.getInstanceForApplication(context);`
2. Call `BeaconManager.setDistanceModelUpdateUrl(null);`
3. Bind a BeaconConsumer to the BeaconManager and Add a MonitorNotifier in onBeaconServiceConnected (per the example code) that logs out in the implemented methods.
4. Launch the app

### Mobile device model and OS version
Pixel 2, Android 8.1.0, API 27

### Android Beacon Library version
2.13.1

kmerrell42, 2018-04-03:
FYI: The url arg for `BeaconManager.setDistanceModelUpdateUrl(null);` method is marked as @NonNull, but I was following the guidance from a the following issue:

[](https://github.com/AltBeacon/android-beacon-library/issues/357)

davidgyoung, 2018-04-03:
So to clarify, this all works as expected, the issue is that ithe @nonnull annotation should be changed to @nullable, yes?

kmerrell42, 2018-04-03:
It looks like making the change from a valid url to null was a red herring. Now my issue is that I am always in the  `0` (out) state for my region regardless of the distance model url change.

Please disregard everything above, but yes, removing the @NonNull would be an improvement. 

Thanks, and sorry about the false report :)

closed 2019-01-10 by davidgyoung