AltBeacon/android-beacon-library #666: App stops scanning just when screen is locked. It wakes up after 15min with the periodic job

Eylen, 2018-04-04:
### Expected behavior
The app should continue scanning for 15minutes after going to background or locking the screen.

### Actual behavior
When the app goes to background or the screen is locked, the app stops scanning inmediately.
The scanning job fires after the 15 minute period

### Steps to reproduce this behavior
Open the app in foreground, the scanning process is done. Then lock the screen or send the app to background without closing it, the scanning process stops inmediately.

### Mobile device model and OS version
Huawei P Smart
Android 8.0.0

### Android Beacon Library version
2.13.1

I'm using BackgroundPowerSaveMode, just in case it affects or changes the behaviour.

This is the log that I've been able to catch with beaconManager.setDebug(true):
```
04-04 12:45:50.760 4116-4116/ D/ScanJobScheduler: Not scheduling an immediate scan because we are in background mode.   Cancelling existing immediate scan.
04-04 12:45:50.761 4116-4116/ I/ScanJob: Using immediateScanJobId from manifest: 208352939
04-04 12:45:50.763 4116-4116/ I/ScanJob: Using periodicScanJobId from manifest: 208352940
04-04 12:45:50.763 4116-4116/ D/ScanJobScheduler: Scheduling ScanJob (job:208352940/<mypackage>/org.altbeacon.beacon.service.ScanJob) to run every 310000 millis
04-04 12:45:50.772 4116-5146/ D/ScanHelper: Beacon packet detected for: id1: 0040c159-12f6-4fc3-9189-87c069ffe5cf id2: 5 id3: 1 with rssi -60
    beacon detected : id1: 0040c159-12f6-4fc3-9189-87c069ffe5cf id2: 5 id3: 1
04-04 12:45:50.784 4116-5146/ D/ScanHelper: looking for ranging region matches for this beacon
    This region (id1: 0090c159-12f6-4fc3-9189-87c069ffe5cf id2: null id3: null) does not match beacon: id1: 0040c159-12f6-4fc3-9189-87c069ffe5cf id2: 5 id3: 1
04-04 12:45:50.785 4116-5146/ D/ScanHelper: This region (id1: 0070c159-12f6-4fc3-9189-87c069ffe5cf id2: null id3: null) does not match beacon: id1: 0040c159-12f6-4fc3-9189-87c069ffe5cf id2: 5 id3: 1
    This region (id1: 0060c159-12f6-4fc3-9189-87c069ffe5cf id2: null id3: null) does not match beacon: id1: 0040c159-12f6-4fc3-9189-87c069ffe5cf id2: 5 id3: 1
    This region (id1: 0080c159-12f6-4fc3-9189-87c069ffe5cf id2: null id3: null) does not match beacon: id1: 0040c159-12f6-4fc3-9189-87c069ffe5cf id2: 5 id3: 1
    This region (id1: 00a0c159-12f6-4fc3-9189-87c069ffe5cf id2: null id3: null) does not match beacon: id1: 0040c159-12f6-4fc3-9189-87c069ffe5cf id2: 5 id3: 1
    This region (id1: 00b0c159-12f6-4fc3-9189-87c069ffe5cf id2: null id3: null) does not match beacon: id1: 0040c159-12f6-4fc3-9189-87c069ffe5cf id2: 5 id3: 1
    This region (id1: 0050c159-12f6-4fc3-9189-87c069ffe5cf id2: null id3: null) does not match beacon: id1: 0040c159-12f6-4fc3-9189-87c069ffe5cf id2: 5 id3: 1
04-04 12:45:50.786 4116-5146/ D/ScanHelper: This region (id1: 00c0c159-12f6-4fc3-9189-87c069ffe5cf id2: null id3: null) does not match beacon: id1: 0040c159-12f6-4fc3-9189-87c069ffe5cf id2: 5 id3: 1
    This region (id1: 00d0c159-12f6-4fc3-9189-87c069ffe5cf id2: null id3: null) does not match beacon: id1: 0040c159-12f6-4fc3-9189-87c069ffe5cf id2: 5 id3: 1
    matches ranging region: id1: 0040c159-12f6-4fc3-9189-87c069ffe5cf id2: null id3: null
04-04 12:45:50.788 4116-4116/ D/CycledLeScannerForLollipop: got record
04-04 12:45:50.789 4116-4116/ I/ScanJob: Using periodicScanJobId from manifest: 208352940

********* Here I locked the screen **************

    onStopJob called for immediate scan org.altbeacon.beacon.service.ScanJob@b9cc647
04-04 12:45:50.789 4116-4116/ D/CycledLeScanner: stop called
    disabling scan
04-04 12:45:50.790 4116-4116/ D/CycledLeScanner: Done with scan cycle
04-04 12:45:50.790 4116-5139/ D/CycledLeScannerForLollipop: Stopping LE scan on scan handler
04-04 12:45:50.790 4116-4116/ D/ScanHelper: Calling ranging callback
04-04 12:45:50.792 4116-4116/ D/ScanHelper: Calling ranging callback
04-04 12:45:50.792 4116-4116/ D/ScanHelper: Calling ranging callback
04-04 12:45:50.792 4116-4116/ D/ScanHelper: Calling ranging callback
04-04 12:45:50.792 4116-4116/ D/ScanHelper: Calling ranging callback
04-04 12:45:50.792 4116-4116/ D/ScanHelper: Calling ranging callback
04-04 12:45:50.792 4116-4116/ D/ScanHelper: Calling ranging callback
04-04 12:45:50.792 4116-4116/ D/ScanHelper: Calling ranging callback
04-04 12:45:50.792 4116-4116/ D/ScanHelper: Calling ranging callback
04-04 12:45:50.793 4116-4116/ D/ScanHelper: Calling ranging callback
04-04 12:45:50.793 4116-4116/ D/CycledLeScanner: Scanning disabled. 
04-04 12:45:50.793 4116-4116/ D/CycledLeScanner: cancel wakeup alarm: PendingIntent{9d57fe3: android.os.BinderProxy@ff1d817}
04-04 12:45:50.797 4116-4116/ D/CycledLeScanner: Set a wakeup alarm to go off in 9223372036233392399 ms: PendingIntent{9d57fe3: android.os.BinderProxy@ff1d817}
    Stopping scanning previously left on.
    stopping bluetooth le scan
04-04 12:45:50.797 4116-4116/ D/CycledLeScannerForLollipop: Stopping scan
04-04 12:45:50.798 4116-4116/ D/CycledLeScanner: Destroying
04-04 12:45:50.798 4116-4116/ D/ScanJob: Scanning stopped
04-04 12:45:50.798 4116-5139/ D/CycledLeScannerForLollipop: Stopping LE scan on scan handler
04-04 12:45:50.798 4116-4116/ D/ScanJob: Checking to see if we need to start a passive scan
04-04 12:45:50.798 4116-4116/ I/ScanJob: We are inside a beacon region.  We will not scan between cycles.
04-04 12:45:50.800 4116-5139/ D/BluetoothLeScanner: could not find callback wrapper
04-04 12:45:50.800 4116-5139/ D/CycledLeScanner: Quitting scan thread
```

davidgyoung, 2018-04-04:
This sounds like expected behavior on Android 8.  As of this OS release, Android blocks long-running services in the background to scan for beacons as well as other activities.  This library accounts for this when an app is in the background by instead using a scheduled job every 15 minutes to scan for beacons.  you can read more about this here:

http://www.davidgyoungtech.com/2017/08/07/beacon-detection-with-android-8

Unfortunately, this is the most frequent background scan period allowed by Andorid 8+.  If you want to do constant scans in the background on Android 8, the  alternative is to use a foreground service to do the scanning.

Eylen, 2018-04-04:
I had read the post before, but I had understood that the expected behaviour was to start a continuos scan during 15 minutes after going to background or locking the screen, as it's the maximum period that Android 8 allows services to run in background I think. After that period, the periodic jobs would kick in to do the scanning in the background.

So is the expected behaviour to stop scanning in the same moment that the app goes to background or the screen is locked? Is there any way to "force" a scan during the allowed 15 minute period after the app goes to background and after that use the periodic job?

dietert, 2018-04-04:
Could it be that the alarm never will be called in this use case 

04-04 12:45:50.797 4116-4116/ D/CycledLeScanner: Set a wakeup alarm to go off in 9223372036233392399 ms

https://github.com/AltBeacon/android-beacon-library/blob/f9df15664c2d803d395fe2a6f726b639222f732b/src/main/java/org/altbeacon/beacon/service/scanner/CycledLeScanner.java#L502

davidgyoung, 2018-04-04:
Currently this is not possible with the library.  In theory, this could be an enhancement.  Perhaps there could be a configurable parameter for how long to *continue* with foreground scan periods for a certain amount of time after transitioning to the background.  (The max time would be limited by Android OS.)  This would be similar to the way iOS CoreLocation continues to range beacons every second for 5-10 seconds after transitioning to the background.

davidgyoung, 2018-04-04:
Yes, I think two enhancements to Android  background scanning are on the table:

1. Allow foreground scanning intervals to continue for a period of time after the app transitions to the background.   

2. Allow scan periods more frequent than 15 minutes in the background by use of a foreground service.

@Eylen, if you think the first enhancement would resolve this issue, I will create a new issue specifically for that enhancement and close this issue referencing the new one.

Eylen, 2018-04-04:
@davidgyoung I think that the first enhancement would be enough for most cases and useful for everyone, so it would be great to have it. Thanks!

The foreground service is something I'm planning to implement in the future, maybe activating the foreground service once you enter a region, but for now I think that the first enhancement would be enough.

davidgyoung, 2018-08-02:
See proposed solution in #717

closed 2018-08-02 by davidgyoung