AltBeacon/android-beacon-library #670: When using BootstrapNotifier BeaconManager.updateScanPeriods doesn't work

snydersaurus, 2018-04-17:
Testing on various incarnations of Android, 6/7/8.

In my Application that implements BootstrapNotifier, whenever I call `updateScanPeriods` (in `onCreate` ) an Exception is thrown saying the beacon consumer/service isn't bound yet. 

In this case where I want the Application to run un the background, at what point should I be calling `updateScanPeriods` so that it actually works?

Thank you

davidgyoung, 2018-04-17:
You only need to call `updateScanPeriods()` if the scanning service has already started, so it knows to pick up the changes. (In fact, as you've seen you can't call it without an exception if it has not started yet.)  So in your case you can just remove the call.  Once the scanning service starts, it will pick them up automatically.

closed 2018-04-17 by davidgyoung

snydersaurus, 2018-04-18:
Thank you @davidgyoung for the clarification. Much appreciated.