AltBeacon/android-beacon-library #671: didRangeBeaconsInRegion Method is never called in Android 8+

snti, 2018-04-17:
### Expected behavior
Find beacons in background, actually this behavior works in my galaxy S6 with Android 7.0.

### Actual behavior
The beacons appears only en foreground, not in background.

### Steps to reproduce this behavior
Download this project https://github.com/AltBeacon/android-beacon-library-reference
Configure the scan time and setBackgroundMode = true.
Run the project and go to the Ranging Activity. After that put the app in background.

### Mobile device model and OS version
Nexus 5X Android 8.1
Samsung Galaxy s8+ Android 8.0

### Android Beacon Library version
2.13.1

IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

davidgyoung, 2018-04-18:
@snti, the problem is that Android 8 introduced new restrictions on background bluetooth scanning that essentially only allow an app to scan once every 15 minutes when in the background.  You can read more about this here:

http://www.davidgyoungtech.com/2017/08/07/beacon-detection-with-android-8

You are not the first person to be surprised by this, and indeed this is a bug changed in Android.  I'm trying to figure out a better way to communicate this change for people building with this library.  The default library settings are to do a background scan every 10 minutes (so this being extended to 15 minutes on Android 8 would not be that much different.)  But if you change the backgroundBetweenScanPeriod like you did to be shorter, then it just won't work on Android 8.  Perhaps I should add a warning in the logs if the backgroundBetweenScanPeriod is ever set to anything below 15 minutes like this:

```
04-17 19:36:38.339 18038-18038/com.mybeaconapp W/BeaconManager: WARNING: Setting a short  backgroundBetweenScanPeriod has no effect on Android 8+, which is limited to scanning every ~15 minutes
```

Would such a warning have helped?

chintan-mishra, 2018-04-18:
Yes this warning will help a developers.

snti, 2018-04-18:
Thanks for the answer guys, @davidgyoung In the URL that you give me says: "The New Way: Fast Detections" with this I should find a solution for Android 8+ ?

Regards!

davidgyoung, 2018-04-18:
Yes, this library supports the new way automatically to the extent allowed by Android 8+.  But as that blog post describes it is essentially not possible to range beacons continuously in the background on Android 8+, unless you create an Android Foreground Service for your app.  If you are interested in that approach, see [here](https://github.com/AltBeacon/android-beacon-library/pull/663) for an experimental build that allows that.

davidgyoung, 2018-04-27:
Resolved via a warning in #677

closed 2018-04-27 by davidgyoung