AltBeacon/android-beacon-library #673: "Not allowed to start service Intent" crashes when app is in background

scm573, 2018-04-24:
Hi guys,

My app is suffering crashes when in background, on Android 8.0. (I've not tried on 8.1 though)

### Expected behavior
App can scan beacons and receive events when in background

### Actual behavior
App crashes when the library executes its callback.

### Steps to reproduce this behavior
1. Run the app on a device approximated to an active beacon.
2. Press home button and wait for its scanning.
3. Crash.

### Mobile device model and OS version
Nexus 5X / Sony Xperia Z1 Compact
Android 8.0.0

### Android Beacon Library version
2.13.1

Here's the stack trace:

> 04-24 11:30:10.010 9868-9868/com.company.product E/AndroidRuntime: FATAL EXCEPTION: main
    Process: com.company.product, PID: 9868
    java.lang.IllegalStateException: Not allowed to start service Intent { cmp=com.company.product/org.altbeacon.beacon.BeaconIntentProcessor (has extras) }: app is in background uid UidRecord{27a0755 u0a265 LAST bg:+1m23s166ms idle procs:1 seq(0,0,0)}
        at android.app.ContextImpl.startServiceCommon(ContextImpl.java:1505)
        at android.app.ContextImpl.startService(ContextImpl.java:1461)
        at android.content.ContextWrapper.startService(ContextWrapper.java:644)
        at org.altbeacon.beacon.service.Callback.call(Callback.java:60)
        at org.altbeacon.beacon.service.BeaconService.processExpiredMonitors(BeaconService.java:368)
        at org.altbeacon.beacon.service.BeaconService.access$200(BeaconService.java:69)
        at org.altbeacon.beacon.service.BeaconService$1.onCycleEnd(BeaconService.java:314)
        at org.altbeacon.beacon.service.scanner.CycledLeScanner.finishScanCycle(CycledLeScanner.java:242)
        at org.altbeacon.beacon.service.scanner.CycledLeScanner.scheduleScanCycleStop(CycledLeScanner.java:234)
        at org.altbeacon.beacon.service.scanner.CycledLeScanner$1.run(CycledLeScanner.java:230)
        at android.os.Handler.handleCallback(Handler.java:789)
        at android.os.Handler.dispatchMessage(Handler.java:98)
        at android.os.Looper.loop(Looper.java:164)
        at android.app.ActivityThread.main(ActivityThread.java:6541)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.Zygote$MethodAndArgsCaller.run(Zygote.java:240)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:767)

davidgyoung, 2018-04-24:
Interesting.  This is the first report I have heard like this.  Does your app use a Service to access the BeaconManager and start scanning?

scm573, 2018-04-24:
Hi @davidgyoung 

Yeah you're perfectly right, I implement the BootstrapNotifier and BeaconConsumer on a service class, which worked well before Oreo, but yes obviously it has been already affected by the OS changes.
What's more, to be honest I also implement GoogleApiClient on the same service class to get the location updates. I know it's a little beyond the discussion scope but if possible may I get some advices for such a situation. Is that a correct solution to replace my service with a JobScheduler?

davidgyoung, 2018-04-25:
Upon further inspection of the stack trace, it looks like the app using the library has set `beaconManager.setScheduledScanJobsEnabled(false)`.  I say this because the stack trace indicates that Callback.java is using a BroadcastIntent to deliver a region exit condition to your service (see snippet below), something that is forbidden in the background by Android 8, which is why you get the exception you see.  On Android 8+, `setScheduledScanJobsEnabled` must be set to true (the default for that platform) for results to be delivered in the background. 

It is legitimate to set this flag to false if you only care about foreground callbacks, or if you are using a foreground service to do the scanning.  So the Library cannot simply disable this.

I should probably add a comment to the JavaDocs and warning line in LogCat about this.  Here is a proposal to do that: #674.  I'm not sure these warnings would have helped in your case unless you had looked at the javadoc for the method, or tested the app on an Android 8+ device and noticed the warning in the log.

I am open to other suggestions to make this clearer.  Any thoughts, @scm573 ?

```
        boolean useLocalBroadcast = BeaconManager.getInstanceForApplication(context).getScheduledScanJobsEnabled();
        boolean success = false;

        if(useLocalBroadcast) {
            String action = null;
            if (dataName == "rangingData") {
                action = BeaconLocalBroadcastProcessor.RANGE_NOTIFICATION;
            }
            else {
                action = BeaconLocalBroadcastProcessor.MONITOR_NOTIFICATION;
            }
            Intent intent = new Intent(action);
            intent.putExtra(dataName, data);
            LogManager.d(TAG, "attempting callback via local broadcast intent: %s",action);
            success = LocalBroadcastManager.getInstance(context).sendBroadcast(intent);
        }
        else {
            Intent intent = new Intent();
            intent.setComponent(new ComponentName(context.getPackageName(), "org.altbeacon.beacon.BeaconIntentProcessor"));
            intent.putExtra(dataName, data);
            LogManager.d(TAG, "attempting callback via global broadcast intent: %s",intent.getComponent());
            try {
                context.startService(intent);
                success = true;
            } catch (Exception e) {
                LogManager.e(
                        TAG,
                        "Failed attempting to start service: " + intent.getComponent().flattenToString(),
                        e
                );
            }
        }
```

davidgyoung, 2018-04-27:
Closing this issue with  #674 as the resolution.

closed 2018-04-27 by davidgyoung