AltBeacon/android-beacon-library #680: Pixel 2 with Android 8.1 sleep mode long time until triggered enter region

arickatz, 2018-04-29:
### Expected behavior
immediate event trigger 
### Actual behavior
not triggering or takes 30 deconds
### Steps to reproduce this behavior
using the 2.13.1 version and bootstrap notifier on the application class
### Mobile device model and OS version
Pixel 2 with Android 8.1
### Android Beacon Library version
2.13.1 version 
IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

davidgyoung, 2018-04-29:
@arickatz, delays in triggering may be normal depending on the conditions under test.  Please add more detail.about your test case including:

* Is your app in the foreground or background?
* are you using BackgroundPowerSaver?
* Do you customize the scan periods?
* How exactly do you test this?

In addition to the above, please show a LogCat excerpt annotated with the time your beacon was detected 30 seconds later than you expected it to be.

Kitsopappas, 2018-05-02:
If you are customising the scan periods be sure to set the setBackgroundBetweenScanPeriod value greater than ~40000 based on your app's needs (On some tests I did, it throws a debug log that the background scanning is too fast for monitoring if you use a value < ~40000). I also use a Pixel 2 for thests with android O having BackgroundPowerSaver, setAndroidLScanningDisabled(true) and setBackgroundBetweenScanPeriod(40000) and the didEnter didExit events are firing perfectly.

davidgyoung, 2019-01-09:
closing due to inactivity

closed 2019-01-09 by davidgyoung