AltBeacon/android-beacon-library #681: didExitRegion event never called when the app is in the background with BackgroundPowerSaver enabled

Kitsopappas, 2018-05-04:
I have an app that is monitoring for beacons. When a beacon enters a region I make a bluetooth connection with this device (therefore it stops advertising) and the didExitRegion event fire. 

### Expected behavior
didEnterRegion and didExitRegion should be fired even if the app is in the background or not with BackgroundPowerSaver on.
### Actual behavior
If a device enters a region (While the app is in the background) and then the device stops advertising the didExitRegion event is never called (The Logcat shows a message "We are inside a beacon region.  We will not scan between cycles."). 
If I remove the BackgroundPowerSaver then the didExitRegion and didEnterRegion events fire perfectly.
### Steps to reproduce this behavior
1) Start monitoring for an ibeacon region with uuid,major,minor
2) Enable BackgroundPowerSaver
3) Run the app and put it in the background
4) Make the ble device stop the advertising (ex remove the battery or connect with it) (No didExitRegionEvents)
5) Disable the BackgroundPowerSaver
6) The didEnter-Exit events fire.
### Mobile device model and OS version
Google Pixel 2
Android 8.1
### Android Beacon Library version
2.13.1
## Sample code
```java
iBeaconManager = BeaconManager.getInstanceForApplication(cordovaActivity);
iBeaconManager.setForegroundBetweenScanPeriod(LocationManager._FOREGROUND_BETWEEN_SCAN_PERIOD);
iBeaconManager.setForegroundScanPeriod(LocationManager._FOREGROUND_SCAN_PERIOD);
iBeaconManager.setBackgroundBetweenScanPeriod(LocationManager._BACKGROUND_BETWEEN_SCAN_PERIOD); 
iBeaconManager.setBackgroundScanPeriod(LocationManager._BACKGROUND_SCAN_PERIOD); 
iBeaconManager.setRegionStatePersistenceEnabled(false);
iBeaconManager.applySettings();
try {
       iBeaconManager.updateScanPeriods();
} catch (Exception e) {
}
backgroundPowerSaver = new BackgroundPowerSaver(applicationContext);
```

IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

davidgyoung, 2018-05-04:
What are the values of _BACKGROUND_BETWEEN_SCAN_PERIOD and the other similar constants?

Kitsopappas, 2018-05-04:
Hi, sorry I forgot about that.

```java
private static final int _BACKGROUND_SCAN_PERIOD = 10000;
private static final int _BACKGROUND_BETWEEN_SCAN_PERIOD = 10000;
private static final int _FOREGROUND_SCAN_PERIOD = 3500;
private static final int _FOREGROUND_BETWEEN_SCAN_PERIOD = 500;
```

davidgyoung, 2018-05-04:
@Kitsopappas, are you sure you wait long enough?  With the settings shown, it will take 20 seconds for a scan cycle to complete in the background.   And if the beacon was turned off near the beginning of the scan cycle, one packet might have been detected that scan cycle and been reported  at the end of it.  Then at the end of the second scan cycle, (potentially up to 40 seconds after the beacon was shut off) the library will only measure 20 seconds since the beacon was last seen, and finally fire the didExitRegion(...) callback.

Are you sure you have waited 40 seconds?

Kitsopappas, 2018-05-29:
Hi, I did some tests and the did exit region event might take 5-10 minutes to fire. 
Although, if the BackgroundPowerSaver object is not initialized the didExit event files within 1sec.

The debug log that worries me is "We are inside a beacon region. We will not scan between cycles.". 
Is it posible to have instant didExit events in the background with the BackgroundPowerSaver initialized?

davidgyoung, 2018-05-29:
The BackgroundPowerSaver saves battery by making beacon  scanning behave differently in the background:

1. If no beacons ate visible, it will do a constant low power scan with hardware filters to quickly detect the first appearance of a beacon.
2. If beacons ate visible, it cannot perform the above technique because the beacon(s) would constantly be detected, causing CPU to be used to parse the beacons and using too much power.  (This causes the debug line you mention). Instead, the  library will stop scanning entirely for the backgroundBetweenScanPeriod then scan briefly for the backgroundScanPeriod to look for changes in what beacons are visible.

You can make detections faster for case 2 by reducing the time of the backgroundBetweenScanPeriod, but understand this will use more battery -- there is no free lunch.  Also, on Android 8+, you must use a foreground service if you want backgroundBetweenScanPeriod values less than 15 minutes, as the OS blocks apps from being active in the vackgroundore than once every 15 min

davidgyoung, 2019-01-09:
closing issue as system works as designed.

closed 2019-01-09 by davidgyoung