AltBeacon/android-beacon-library #685: Not all beacons being ranged (at start up)

tomacco, 2018-05-11:
### Expected behavior

When the library starts ranging a region (iBeacon - UUID only). It is expected to return all beacons in the region and not only one of them.

### Actual behavior

 1 out of 2 beacons are being ranged.
This only happens at start up.
From my tests it happens about 70% of the time

### Steps to reproduce this behavior

Let's have two beacons with the same UUID and different majors and minors. Let's call them A and B.
Both beacons are emitting (10Hz rate) and are close enough to the device (few centimeters).

- Ensure both beacons A and B are emitting before running the app.
- Start the application. 

(Details of how altBeacon API is being used)
- The AltBeacon library starts monitoring the region defined by their common UUID. 
- Immediately after start monitoring the region, it calls the `didEnterRegion` (as the beacons are already emitting nearby). So the `startRangingForRegion` is called, and immediately the the `didRangeBeaconsInRegion` callback is called with the list of ranged beacons. So far so good. 

- The problem is that the list of beacons contains *a single item instead of two (only A)*, If I stop the beacon B from emitting and immediately activate it back, *AltBeacons reports both A and B.*

### Mobile device model and OS version
Bq Aquaris Android 5.1.1

### Android Beacon Library version
2.13.1

IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

davidgyoung, 2018-05-11:
If you do not intervene, does the second beacon show up in ranging results after some time has passed?

tomacco, 2018-05-11:
@davidgyoung After let it run for several minutes, it is not detected unless I manually cut the signal for a brief interval (actually cutting it for less than a second does the trick).

I've recorded a video that could help:
https://youtu.be/LEOMcrLe1DM

Please also note that just after the app starts, altBeacon reports the second beacon briefly after starting up, but after that it disappears.
I've also printed the count in the beaconlist for every callback of `didRangeBeaconsInRegion`. This is the result (The very first time it reports 2 beacons and afterwards, only 1)

```
05-11 17:30:24.212 10861-10933/com.mocaplatform.mocapplication V/MOCA: Beacons in id1: a7826da6-4fa2-4e98-8024-bc5b71e0893e id2: null id3: null: 	2
05-11 17:30:25.317 10861-10934/com.mocaplatform.mocapplication V/MOCA: Beacons in id1: a7826da6-4fa2-4e98-8024-bc5b71e0893e id2: null id3: null: 	1
05-11 17:30:26.428 10861-10935/com.mocaplatform.mocapplication V/MOCA: Beacons in id1: a7826da6-4fa2-4e98-8024-bc5b71e0893e id2: null id3: null: 	1
05-11 17:30:27.535 10861-10936/com.mocaplatform.mocapplication V/MOCA: Beacons in id1: a7826da6-4fa2-4e98-8024-bc5b71e0893e id2: null id3: null: 	1
05-11 17:30:28.638 10861-10937/com.mocaplatform.mocapplication V/MOCA: Beacons in id1: a7826da6-4fa2-4e98-8024-bc5b71e0893e id2: null id3: null: 	1
05-11 17:30:29.751 10861-10938/com.mocaplatform.mocapplication V/MOCA: Beacons in id1: a7826da6-4fa2-4e98-8024-bc5b71e0893e id2: null id3: null: 	1
05-11 17:30:30.861 10861-10940/com.mocaplatform.mocapplication V/MOCA: Beacons in id1: a7826da6-4fa2-4e98-8024-bc5b71e0893e id2: null id3: null: 	1
05-11 17:30:31.967 10861-10941/com.mocaplatform.mocapplication V/MOCA: Beacons in id1: a7826da6-4fa2-4e98-8024-bc5b71e0893e id2: null id3: null: 	1
05-11 17:30:33.073 10861-10942/com.mocaplatform.mocapplication V/MOCA: Beacons in id1: a7826da6-4fa2-4e98-8024-bc5b71e0893e id2: null id3: null: 	1
05-11 17:30:34.184 10861-10943/com.mocaplatform.mocapplication V/MOCA: Beacons in id1: a7826da6-4fa2-4e98-8024-bc5b71e0893e id2: null id3: null: 	1
05-11 17:30:35.287 10861-10944/com.mocaplatform.mocapplication V/MOCA: Beacons in id1: a7826da6-4fa2-4e98-8024-bc5b71e0893e id2: null id3: null: 	1
05-11 17:30:36.418 10861-10945/com.mocaplatform.mocapplication V/MOCA: Beacons in id1: a7826da6-4fa2-4e98-8024-bc5b71e0893e id2: null id3: null: 	1
05-11 17:30:37.522 10861-10946/com.mocaplatform.mocapplication V/MOCA: Beacons in id1: a7826da6-4fa2-4e98-8024-bc5b71e0893e id2: null id3: null: 	1
05-11 17:30:38.627 10861-10947/com.mocaplatform.mocapplication V/MOCA: Beacons in id1: a7826da6-4fa2-4e98-8024-bc5b71e0893e id2: null id3: null: 	1
05-11 17:30:39.732 10861-10948/com.mocaplatform.mocapplication V/MOCA: Beacons in id1: a7826da6-4fa2-4e98-8024-bc5b71e0893e id2: null id3: null: 	1
05-11 17:30:40.839 10861-10949/com.mocaplatform.mocapplication V/MOCA: Beacons in id1: a7826da6-4fa2-4e98-8024-bc5b71e0893e id2: null id3: null: 	1
05-11 17:30:41.945 10861-10950/com.mocaplatform.mocapplication V/MOCA: Beacons in id1: a7826da6-4fa2-4e98-8024-bc5b71e0893e id2: null id3: null: 	1
05-11 17:30:43.054 10861-10951/com.mocaplatform.mocapplication V/MOCA: Beacons in id1: a7826da6-4fa2-4e98-8024-bc5b71e0893e id2: null id3: null: 	1
05-11 17:30:44.159 10861-10952/com.mocaplatform.mocapplication V/MOCA: Beacons in id1: a7826da6-4fa2-4e98-8024-bc5b71e0893e id2: null id3: null: 	1
05-11 17:30:45.269 10861-10953/com.mocaplatform.mocapplication V/MOCA: Beacons in id1: a7826da6-4fa2-4e98-8024-bc5b71e0893e id2: null id3: null: 	1
05-11 17:30:46.383 10861-10954/com.mocaplatform.mocapplication V/MOCA: Beacons in id1: a7826da6-4fa2-4e98-8024-bc5b71e0893e id2: null id3: null: 	1
05-11 17:30:47.486 10861-10955/com.mocaplatform.mocapplication V/MOCA: Beacons in id1: a7826da6-4fa2-4e98-8024-bc5b71e0893e id2: null id3: null: 	1
05-11 17:30:48.589 10861-10956/com.mocaplatform.mocapplication V/MOCA: Beacons in id1: a7826da6-4fa2-4e98-8024-bc5b71e0893e id2: null id3: null: 	1
05-11 17:30:49.698 10861-10957/com.mocaplatform.mocapplication V/MOCA: Beacons in id1: a7826da6-4fa2-4e98-8024-bc5b71e0893e id2: null id3: null: 	1
05-11 17:30:50.803 10861-10958/com.mocaplatform.mocapplication V/MOCA: Beacons in id1: a7826da6-4fa2-4e98-8024-bc5b71e0893e id2: null id3: null: 	1
05-11 17:30:51.908 10861-10959/com.mocaplatform.mocapplication V/MOCA: Beacons in id1: a7826da6-4fa2-4e98-8024-bc5b71e0893e id2: null id3: null: 	1

```

This is the argument in the callback.
[image]

davidgyoung, 2018-05-11:
Wow -- the video certainly shows the behavior you describe.

@tomacco, it's hard for me to imagine what might cause this on the scanning side.  I do lots of testing with the library where scanning is started with multiple beacons advertising, and I certainly don't see behavior like this.  There has to be yet another variable here that we aren't thinking of.

Is it always the *same* device that does not get detected or does it vary between one test run to the next?

Is it possible that the second transmitter *actually does* stop transmitting between the time you launch your app and the time you turn it off and turn it back on?  I'm not sure what might cause it to stop (maybe some BLE communication when you start your app?)  But just to eliminate the possibility, can you try two things:

(1) Use a different beacon scanning app like Beacon Locate instead of your app (which also uses the library) to try to reproduce.

(2) Try your app with a couple of hardware beacons (if you have any) to see if these keep being tracked on app launch.

tomacco, 2018-05-11:
@davidgyoung 

I've switched the beacon IDs in the two phones, and yes, the one that fails is always the same device, which is weird. 
I've added a control device (iOS) in order to check that the signal is being emitted during the restarting process, so I can be sure both phones are emitting.
 
[image]

In the not-being-detected phone, I've installed Beacon Locate, and results are the same:

[image]

I'll perform tests with real beacons. 

Might be related: Both beacons share the same UUID *and* the same MAJOR

davidgyoung, 2018-05-11:
Interesting.

I suspect that the problematic phone's advertisement is being altered somehow by the operating system (perhaps due to a second app or a system level app also transmitting.)  It may change the packet slightly so that it is not detected by the library.

If you turn on debug logging in your app with BeaconManager.setDebug(true); then you can see the raw bytes of the detections in the log at the time Device B fails to be detected.  There is probably something different about the byte sequence in this case, but you should be able to find it by searching for the minor value that is different -- it will be in hex bytes.

If you can run this test and paste the log lines, I bet we can get to the bottom of it.

davidgyoung, 2019-01-09:
closing due to inactivity

closed 2019-01-09 by davidgyoung