AltBeacon/android-beacon-library #686: Multiple advertising with timeout

parthrparekh93, 2018-05-17:
I wanted to broadcast multiple beacons at a time for a couple of seconds, wait for a minute and broadcast new multiple beacons for a couple of seconds. Therefore, I used AdvertiseSettings timeout parameter. What I am observing is that if I broadcast a single beacon for 5 seconds after every minute, it works perfectly. Once I start broadcasting multiple beacons with timeout, it broadcasts for the first time but the next time, it gets stuck. Basically I receive no success or failure callbacks. (Occassionally I get ADVERTISE_FAILED_INTERNAL_ERROR) I feel multi advertising with timeout is causing issues.  
I see in the AndroidBeaconLibrary, the timeout is set to 0. Why did you skip the option to set timeout from the library itself.

### Expected behavior
Multiple beacon advertisement with timeout 

### Actual behavior
Multiple beacon advertisement with timeout 0 works, with timeout causes no callback from the startAdvertisement call

### Steps to reproduce this behavior
Advertise 2 beacons with a timeout value < 1 minute. Wait for 1 minute and broadcast the same 2 beacons again with the same timeout.

### OS version
Android 5.1.1

davidgyoung, 2018-05-17:
I am not sure how you are having a problem with a timeout since the library's BeaconTransmitter always sets the timeout to zero, effectively disabling any timeout.  How have you made it work otherwise?

When starting a BeaconTransmitter, I have seen the callback work fine on a wide variety of devices using the sample code shown here:

https://altbeacon.github.io/android-beacon-library/beacon-transmitter.html

Does this sample code not working for you?

The simple reason the library does not expose the timeout parameter is because I saw no need when writing it.  You can simply call start and stop, and use a timer if you want to stop after a specific time.  I would be open to adding this configuration ability as a new feature, but again it is still not obvious to me there is a real need.

davidgyoung, 2019-01-09:
closing due to inactivity

closed 2019-01-09 by davidgyoung