AltBeacon/android-beacon-library #689: didEnterRegion not called after reboot the device

aashu09, 2018-05-29:
Hi Support,

I am using Altbeacon library from more than a year but now facing an issue.
Please resolve it ASAP.

### Expected behavior
After rebooting the device, if device is within the range of beacon then didEnterRegion method should called.

### Actual behavior
After rebooting the device, if device is within range of beacon however didEnterRegion not called.

### Steps to reproduce this behavior
1. open the app
2. detect beacon
3. reboot device
4. again open app, didEnterRegion will not called.

### Mobile device model and OS version
Samsung s8
version - 6.0

### Android Beacon Library version
2.14

IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

davidgyoung, 2018-05-29:
@aashu09, understand that if the app has seen the beacon recently when the phone reboots, it will still be in an in region state and no new didEnterRegion event will be called.   This is by design.

Please verify whether this is the  cause of what you are seeing by performing the following tests:

1. Run your app in the foreground and turn off your beacons until you can verify you got a did ExitRegion event.

2. Reboot the device

3. Wait up to 15 min for a didEnterRegion event.

If you do not get the event, then please repeat steps 1-3 with the reference app here: https://github.com/AltBeacon/android-beacon-library-reference

Please let me know your results.

Understand that this free library is a volunteer effort, and we need all users to assist in troubleshooting to get to the bottom of such issues.

aashu09, 2018-05-29:
@davidgyoung You are right. It was because of staying in didEnterRegion state while rebooting device.
But I need to call didEnterRegion while rebooting, so can I manually call didExitRegion or didEnterRegion ?
Please response.

Thank you for your support

davidgyoung, 2018-05-29:
You may disable the library remembering region state across reboots with this line:

`beaconManager.setRegioinStatePeristenceEnabled(false);`

closed 2019-01-09 by davidgyoung