AltBeacon/android-beacon-library #690: Not detecting beacons after app killed

aashu09, 2018-05-29:
Hi Support,
I have been using Altbeacon library since last year but now facing problems in some device.
Please resolve it.

### Expected behavior
After killing the app, beacon detection should work same as foreground or background.

### Actual behavior
After killing the app, beacon detection never works.

### Steps to reproduce this behavior
1. Open the app
2. Detect beacon
3. Kill the App

### Mobile device model and OS version
device model - HUAWEI_NEM-L22
OS version - 6.0

### Android Beacon Library version
2.14

IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

davidgyoung, 2018-05-29:
Please repeat this test with the reference app here:

https://github.com/AltBeacon/android-beacon-library-reference

Then reboot the phone with a beacon transmitting and attach a LogCat trace for 1 minute before and  5 minutes after the phone reboot.

If you can attach these logs I will review and investigate

davidgyoung, 2019-01-09:
closing due to inactivity

closed 2019-01-09 by davidgyoung

chintan-mishra, 2019-02-08:
In most Chinese ROMs, the Android OS is customized for extensive battery conservation. As a result, scheduled jobs, pending intents, `START_STICKY` services, etc. never get delivered.

davidgyoung, 2019-02-08:
I have not seen myself that Huawei blocks START_STICKY, but if confirmed, it may be possible to use a scheduled job to restart the library's scanning services on a 15 minute cycle for manufacturers known to do this.