AltBeacon/android-beacon-library #694: Stopped picking beacons after updating to 2.14

glaubergp, 2018-06-01:
### Expected behavior
to see a list of at least 1 beacon on .addRangeNotifier() with the app running in foreground

### Actual behaviour
beacons.size() == 0

### Steps to reproduce this behaviour
upgraded sdk from 2.9.1 to 2.14

### Mobile device model and OS version
Pixel 2 XL 8.1

### Android Beacon Library version
2.14

IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

davidgyoung, 2018-06-02:
@glaubergp, can you please set

beaconManager.setDebug(true);

Then attach a LogCat excerpt of starting up the app and not detecting beacons for at least 10 seconds.

Please also try an off the shelf beacon detector like   Beacon Scope https://play.google.com/store/apps/details?id=com.davidgyoungtech.beaconscanner to make sure the beacon is actually transmitting.

glaubergp, 2018-06-02:
there you go: 

`06-01 17:22:19.804 27053-27053/tech.glaubergpdev D/ScanJobScheduler: Scheduling ScanJob (job:208352940/tech.glaubergpdev/org.altbeacon.beacon.service.ScanJob) to run every 300000 millis
06-01 17:22:19.873 27053-27053/tech.glaubergpdev I/ScanJob: Using immediateScanJobId from manifest: 208352939
    Running immediate scan job: instance is org.altbeacon.beacon.service.ScanJob@9e93a54
06-01 17:22:20.992 27053-27053/tech.glaubergpdev D/Callback: attempting callback via local broadcast intent: org.altbeacon.beacon.range_notification
06-01 17:22:22.099 27053-27053/tech.glaubergpdev D/Callback: attempting callback via local broadcast intent: org.altbeacon.beacon.range_notification
06-01 17:22:24.093 27053-27053/tech.glaubergpdev D/Callback: attempting callback via local broadcast intent: org.altbeacon.beacon.range_notification
06-01 17:22:26.105 27053-27053/tech.glaubergpdev D/Callback: attempting callback via local broadcast intent: org.altbeacon.beacon.range_notification
06-01 17:22:28.214 27053-27053/tech.glaubergpdev D/Callback: attempting callback via local broadcast intent: org.altbeacon.beacon.range_notification
06-01 17:22:30.310 27053-27053/tech.glaubergpdev D/Callback: attempting callback via local broadcast intent: org.altbeacon.beacon.range_notification
06-01 17:22:32.421 27053-27053/tech.glaubergpdev D/Callback: attempting callback via local broadcast intent: org.altbeacon.beacon.range_notification
06-01 17:22:34.504 27053-27053/tech.glaubergpdev D/Callback: attempting callback via local broadcast intent: org.altbeacon.beacon.range_notification
06-01 17:22:36.617 27053-27053/tech.glaubergpdev D/Callback: attempting callback via local broadcast intent: org.altbeacon.beacon.range_notification
06-01 17:22:38.706 27053-27053/tech.glaubergpdev D/Callback: attempting callback via local broadcast intent: org.altbeacon.beacon.range_notification
06-01 17:22:40.804 27053-27053/tech.glaubergpdev D/Callback: attempting callback via local broadcast intent: org.altbeacon.beacon.range_notification
06-01 17:22:42.903 27053-27053/tech.glaubergpdev D/Callback: attempting callback via local broadcast intent: org.altbeacon.beacon.range_notification
06-01 17:22:45.012 27053-27053/tech.glaubergpdev D/Callback: attempting callback via local broadcast intent: org.altbeacon.beacon.range_notification
06-01 17:22:47.108 27053-27053/tech.glaubergpdev D/Callback: attempting callback via local broadcast intent: org.altbeacon.beacon.range_notification
06-01 17:22:49.207 27053-27053/tech.glaubergpdev D/Callback: attempting callback via local broadcast intent: org.altbeacon.beacon.range_notification
06-01 17:22:51.305 27053-27053/tech.glaubergpdev D/Callback: attempting callback via local broadcast intent: org.altbeacon.beacon.range_notification
06-01 17:22:53.415 27053-27053/tech.glaubergpdev D/Callback: attempting callback via local broadcast intent: org.altbeacon.beacon.range_notification
06-01 17:22:55.505 27053-27053/tech.glaubergpdev D/Callback: attempting callback via local broadcast intent: org.altbeacon.beacon.range_notification
06-01 17:22:57.612 27053-27053/tech.glaubergpdev D/Callback: attempting callback via local broadcast intent: org.altbeacon.beacon.range_notification
06-01 17:22:59.710 27053-27053/tech.glaubergpdev D/Callback: attempting callback via local broadcast intent: org.altbeacon.beacon.range_notification
06-01 17:23:01.803 27053-27053/tech.glaubergpdev D/Callback: attempting callback via local broadcast intent: org.altbeacon.beacon.range_notification
06-01 17:23:03.903 27053-27053/tech.glaubergpdev D/Callback: attempting callback via local broadcast intent: org.altbeacon.beacon.range_notification
06-01 17:23:06.009 27053-27053/tech.glaubergpdev D/Callback: attempting callback via local broadcast intent: org.altbeacon.beacon.range_notification
06-01 17:23:08.111 27053-27053/tech.glaubergpdev D/Callback: attempting callback via local broadcast intent: org.altbeacon.beacon.range_notification
06-01 17:23:10.207 27053-27053/tech.glaubergpdev D/Callback: attempting callback via local broadcast intent: org.altbeacon.beacon.range_notification
06-01 17:23:12.306 27053-27053/tech.glaubergpdev D/Callback: attempting callback via local broadcast intent: org.altbeacon.beacon.range_notification
06-01 17:23:14.406 27053-27053/tech.glaubergpdev D/Callback: attempting callback via local broadcast intent: org.altbeacon.beacon.range_notification
06-01 17:23:16.505 27053-27053/tech.glaubergpdev D/Callback: attempting callback via local broadcast intent: org.altbeacon.beacon.range_notification
06-01 17:23:18.607 27053-27053/tech.glaubergpdev D/Callback: attempting callback via local broadcast intent: org.altbeacon.beacon.range_notification
06-01 17:23:20.702 27053-27053/tech.glaubergpdev D/Callback: attempting callback via local broadcast intent: org.altbeacon.beacon.range_notification
06-01 17:23:22.803 27053-27053/tech.glaubergpdev D/Callback: attempting callback via local broadcast intent: org.altbeacon.beacon.range_notification
06-01 17:23:25.027 27053-27053/tech.glaubergpdev D/Callback: attempting callback via local broadcast intent: org.altbeacon.beacon.range_notification
06-01 17:23:27.009 27053-27053/tech.glaubergpdev D/Callback: attempting callback via local broadcast intent: org.altbeacon.beacon.range_notification
06-01 17:23:29.104 27053-27053/tech.glaubergpdev D/Callback: attempting callback via local broadcast intent: org.altbeacon.beacon.range_notification
[... cut ...]
06-01 17:24:01.172 24753-24753/life.jointheactive.activepay I/ScanJob: Scan job runtime expired: org.altbeacon.beacon.service.ScanJob@1924a8
06-01 17:24:01.433 24753-24753/life.jointheactive.activepay I/ScanJob: Using immediateScanJobId from manifest: 208352939
    Running immediate scan job: instance is org.altbeacon.beacon.service.ScanJob@43c2fb9`

And yes, I'm using iBeacon to transmit the beacon signal. I can detect it via Beacon Toy for Android.

Also, thanks a lot for replying and the support.

davidgyoung, 2018-06-02:
Strange, it seems there *must* be some log lines missing there.  I suspect something is filtering them out.   Can you try doing the same capture with `adb logcat` on the command line so you capture the log lines from both your app and operating system processes.  This will help me determine if bluetooth scanning is being blocked somehow.

glaubergp, 2018-06-02:
I filtered by **altbeacon**. Did you want the whole project log?

davidgyoung, 2018-06-02:
I probably need to see everything, yes, and not just for the project, but for the operating system for the first 10 seconds after app start up.  If there is anything you cannot post in a public form, fee free to redact it, or you can send it to me privately at tech@davidgyoung.com

glaubergp, 2018-06-02:
sent you an email. Thanks!

davidgyoung, 2018-06-02:
A log shared privately shows these lines:

```
06-01 17:50:37.727 1456-2078/? I/bt_stack: [INFO:gatt_api.cc(1004)] GATT_Register
06-01 17:50:37.727 1456-2078/? E/bt_stack: [ERROR:gatt_api.cc(1032)] can't Register GATT client, MAX client reached: 32
06-01 17:50:37.727 1456-2078/? E/bt_btif: Register with GATT stack failed.
06-01 17:50:37.727 1456-2078/? I/bt_stack: [INFO:gatt_api.cc(1004)] GATT_Register
06-01 17:50:37.727 1456-2078/? E/bt_stack: [ERROR:gatt_api.cc(1032)] can't Register GATT client, MAX client reached: 32
06-01 17:50:37.727 1456-2078/? E/bt_btif: Register with GATT stack failed.
06-01 17:50:37.727 29956-30092/tech.glaubergpdev D/BluetoothLeScanner: onScannerRegistered() - status=133 scannerId=0 mScannerId=0
06-01 17:50:37.728 29956-29956/tech.glaubergpdev E/CycledLeScannerForLollipop: Scan failed: app cannot be registered
```

This appears to be a hangup in the bluetooth stack and not anything specific to the library version.

Does turning bluetooth off and then on again resolve the problem?  Can any apps detect beacons once the phone is in this state? Does rebooting help?

davidgyoung, 2018-06-02:
The log lines like below indicate that the Android bluetooth stack has exhausted the ability of new apps to start bluetooth scans for beacons:

`06-01 17:50:37.727 1456-2078/? E/bt_stack: [ERROR:gatt_api.cc(1032)] can't Register GATT client, MAX client reached: 32`

This might mean that 32 apps on the phone have already used all the bluetooth clients, but more likely it is caused by an OS-level bug that has exhausted all the client slots prematurely, potentially by leaving some orphaned.

While this is an OS-level issue, this  library has been equipped with a BluetoothMedic to help solve problems like this.  If you find that this is a rare issue, you might want to employ this to protect against the problem.  In this case, the Bluetooth Medic can fix the issue by cycling power to bluetooth, which releases all the client slots.  In order to do this, you must configure the medic to cycle power on failures like this:

    BluetoothMedic medic = BluetoothMedic.getInstance(); 
    medic.enablePowerCycleOnFailures(context); 
    medic.enablePeriodicTests(context, BluetoothMedic.SCAN_TEST);  // this last part is optional

glaubergp, 2018-06-04:
Strangely, it works now. I got at work, sat, ran and bang. Go figure lol My guess is that I had some sort of bluetooth interference here or something of that kind. I really appreciate all your support and congrats for the awesome lib, you guys rock!

I'm closing this issue btw.

Have a great week!

closed 2018-06-04 by glaubergp