AltBeacon/android-beacon-library #696: Can't include AndroidBeaconLibrary as a project dependency wiyh source

kundelstein, 2018-06-05:
Hi.

I'm trying to add android-beacon-library as the project dependency, but WITH the source code. I want to be able to change the source so I could add some functionality to this great library.

I don't want to add it as aar or by adding "compile 'org.altbeacon:android-beacon-library:2+'". I want to add the library as a source code, but when I'm trying to File->New->Import Module.. I can't select the project, instead I get "Select Modules to Import" error message.

Can you please let me know what I'm doing wrong and what should I do in order to have this library as a library dependency with source code?

davidgyoung, 2018-06-05:
1. In your top-level project folder  (where your outermost build.gradle file exists), edit the `settings.gradle` file to add a reference to the library:

```
include ':android-beacon-library' # add this line
include ':app' # existing line 
```

2. Clone the library into the same folder where the above settings.gradle file exists, so there is a android-beacon-library folder next to it.

3. Change the way you reference the dependency to the library in your innermost build.gradle file to:

```
dependencies {
    //compile 'org.altbeacon:android-beacon-library:2+'
    compile project(':android-beacon-library')
    ...
}
```

kundelstein, 2018-06-05:
Thank you for your answer. Unfortunately when I follow the provided steps I receive following error after syncing:
`Cannot create variant 'android-lint' after configuration ':android-beacon-library:debugRuntimeElements' has been resolved`

I'm using the most recent of Android Studio 3.1.2, can you help me to resolve this issue?

davidgyoung, 2018-06-05:
This sounds like an incompatibility between the gradle version used by your project and that used by the library.   This is a common and frustrating problem as gradle's evolution often creates backward incompatibilities.   I see a similar issue here:  https://github.com/google/ExoPlayer/issues/3708

You might try upgrading the gradle version in the android-beacon-library to be the same as your project (I would try to get it working by itself first before integrating with your project. And if you can get it working, I would welcome a pull request on the library to upgrade gradle.)  If you can't get this working, you might try downgrading your project's gradle version to match that of the library.

kundelstein, 2018-06-06:
Thank you for your help. All I needed to do is to add
`doFirst { ... }`
as stated in linked solution.

closed 2018-06-06 by kundelstein

tomacco, 2018-10-09:
Just to be more specific for anyone who reads this thread:
- In the file 'build.gradle' of the altbeacon project change the line:

`            classpath = files(variant.javaCompile.classpath.files, ext.androidJar)`

to

```
        doFirst {
            classpath = files(variant.javaCompile.classpath.files, ext.androidJar)
        }
```