AltBeacon/android-beacon-library #699: Not working on Android 8

breakline87, 2018-06-07:
Running a service in the foreground to scan nearby beacons. I don't see any Beacons on my Android 8 device, but it works on my Android 7 device. So I suppose background restrictions don't apply, or does it matter for the Beacon library itself? Because my client wants an app which constantly monitors a list of beacons while the app is open basically. 

I'm using version 2.13 which supposedly supports Android 8.

kundelstein, 2018-06-07:
Hi.
Famous 4 words: "It works on mine". I have recently made the app (so used a new SDK and all that fancy stuff) and it worked on at least 4 different Android 8 devices. Furthermore I visited some retail stores and confirmed that the tool using it (namely: "Locate") works well on several more devices. You are doing something wrong. My version was a "stock one", applied in the matter which is included by adding dependency "by the book": `compile 'org.altbeacon:android-beacon-library:2+'`

davidgyoung, 2018-06-07:
@breakline87, please try running the [reference application](https://github.com/altbeacon/android-beacon-library-reference) on your Android 8 device to see if this known working setup detects.  If it does not detect, please modify it to set `beaconManager.setDebug(true);` and attach a LogCat capture for 60 seconds after launching the app.  Please also include the device model and exact build number from Settings.

breakline87, 2018-06-13:
So basically rewrote the whole part (it was done by another developer) and now it works. I supposed first that simply updating the lib would have worked but I guess it did not. The reference app helped. Thanks, gonna close it.

closed 2018-06-13 by breakline87