AltBeacon/android-beacon-library #701: Null Pointer Exception: ScanHelper.stopAndroidOBackgroundScan

EmmettWilson, 2018-06-08:
### Steps to reproduce this behavior

```Caused by java.lang.NullPointerException
Attempt to invoke virtual method 'java.util.Set com.android.bluetooth.gatt.ScanManager.getBatchScanQueue()' on a null object reference
android.os.Parcel.readException (Parcel.java:2010)
android.os.Parcel.readException (Parcel.java:1950)
android.bluetooth.IBluetoothGatt$Stub$Proxy.stopScanForIntent (IBluetoothGatt.java:972)
android.bluetooth.le.BluetoothLeScanner.stopScan (BluetoothLeScanner.java:283)
at
org.altbeacon.beacon.service.ScanHelper.stopAndroidOBackgroundScan (ScanHelper.java:194)
org.altbeacon.beacon.service.ScanJob.restartScanning (ScanJob.java:187)
org.altbeacon.beacon.service.ScanJob.startScanning (ScanJob.java:225)
org.altbeacon.beacon.service.ScanJob.onStartJob (ScanJob.java:81)
android.app.job.JobService$1.onStartJob (JobService.java:71)
android.app.job.JobServiceEngine$JobHandler.handleMessage (JobServiceEngine.java:108)
android.os.Handler.dispatchMessage (Handler.java:106)
android.os.Looper.loop (Looper.java:164)
android.app.ActivityThread.main (ActivityThread.java:6494)
java.lang.reflect.Method.invoke (Method.java)
com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run (RuntimeInit.java:438)
com.android.internal.os.ZygoteInit.main (ZygoteInit.java:807)```

### Mobile device model and OS version
Device Pixel 2 KL
Android: 8.1.0

### Android Beacon Library version
2.14

This has also been reported by other users as part of #636

davidgyoung, 2018-06-08:
Anybody know how to reproduce this?

All the reports I have heard of come from remote  crash log reports which give no indication of what conditions cause this, how frequently it happens etc.

Without more info to reproduce it will be very hard to solve.

EmmettWilson, 2018-06-08:
We saw this 23 times for one user in the past 24 hours. It happened once a minute for a period of time.

davidgyoung, 2018-06-08:
What we really need is some hands on testing on this device to try to find the conditions to reproduce.  And get a LogCat excerpt from a few secs before to figure out context.

Blutooth on or off?
Screen on or off?
App in foreground or background?
Any special power modes?

bthrall, 2018-06-11:
I'm working with Emmett on this.

We are trying to reproduce it ourselves, but we do know:

Bluetooth was on
App in background (according to Crashlytics)
Wifi connected
Phone was plugged in

davidgyoung, 2018-06-11:
It's unclear what conditions cause this exception, but it originates inside operating system code that uses AIDL to .  You can see the commit that added it here:

https://de.osdn.net/projects/android-x86/scm/git/frameworks-base/commits/461111bc3d9b53f745a9be686504aee161f36c28

Unfortunately the line of code where the exception actually originates is not open source, so we can't see it.  And while exceptions are normally not supposed to propagate up beyond the AIDL boundary, I see lots of reports that RuntimeExceptions like the NullPointerException in this case do propagate up:

https://blog.classycode.com/dealing-with-exceptions-in-aidl-9ba904c6d63

So while finding the conditions that cause this could be useful in preventing it from happening, if it is not always possible to prevent, the only recourse may be to catch a RuntimeException  like this:

        try {
            final BluetoothManager bluetoothManager =
                    (BluetoothManager) mContext.getApplicationContext().getSystemService(Context.BLUETOOTH_SERVICE);
            BluetoothAdapter bluetoothAdapter = bluetoothManager.getAdapter();
            if (bluetoothAdapter == null) {
                LogManager.w(TAG, "Failed to construct a BluetoothAdapter");
            } else if (!bluetoothAdapter.isEnabled()) {
                LogManager.w(TAG, "BluetoothAdapter is not enabled");
            } else {
               BluetoothLeScanner scanner =  bluetoothAdapter.getBluetoothLeScanner();
               if (scanner != null) {
                   scanner.stopScan(getScanCallbackIntent());
               }
            }
        } catch (SecurityException e) {
            LogManager.e(TAG, "SecurityException stopping Android O background scanner");
        } catch (RuntimeException e) {
            LogManager.e(TAG, "Unexpected runtime exception stopping Android O background scanner", e);
        }

Without understanding the root cause, simply adding the above catch might just cause another exception elsewhere.  However, if any of you experiencing this in your project want to send an app update with the above change to see how it works, I will prepare a test build.

If you would like to volunteer to send you users such a test build, please reply here.

fedorenkoalex, 2018-08-15:
Also I have a lot of same crashes. Here you can find a details: http://crashes.to/s/784e31104fe

Caused by java.lang.NullPointerException: Attempt to invoke virtual method 'java.util.Set com.android.bluetooth.gatt.ScanManager.getBatchScanQueue()' on a null object reference
       at android.os.Parcel.readException(Parcel.java:1969)
       at android.os.Parcel.readException(Parcel.java:1899)
       at android.bluetooth.IBluetoothGatt$Stub$Proxy.stopScanForIntent(IBluetoothGatt.java:970)
       at android.bluetooth.le.BluetoothLeScanner.stopScan(BluetoothLeScanner.java:290)
       at org.altbeacon.beacon.service.ScanHelper.stopAndroidOBackgroundScan(SourceFile:194)
       at org.altbeacon.beacon.service.ScanJob.restartScanning(SourceFile:187)
       at org.altbeacon.beacon.service.ScanJob.startScanning(SourceFile:225)
       at org.altbeacon.beacon.service.ScanJob.onStartJob(SourceFile:81)
       at android.app.job.JobService$1.onStartJob(JobService.java:71)
       at android.app.job.JobServiceEngine$JobHandler.handleMessage(JobServiceEngine.java:108)
       at android.os.Handler.dispatchMessage(Handler.java:105)
       at android.os.Looper.loop(Looper.java:251)

davidgyoung, 2018-08-15:
Fixed in #724.  Please go to that PR to download a test build with the fix.

PR #724 (2018-09-01): Catch of exception thrown by Android stopping scan
This is an attempt to fix #701.   I have not tested that this works, as there are no known ways to reproduce the problem.  It has only been seen in crash logs.
closed 2018-08-15 by davidgyoung