AltBeacon/android-beacon-library #704: iBeacon not detected after app restart on Android pre-8

geoffday67, 2018-06-15:
### Expected behavior

iBeacons should continue to be detected after the app is quit by swiping from the "running apps" menu.

### Actual behavior

The app is restarted but only AltBeacons are detected.

### Steps to reproduce this behavior

Create an app using RegionBootstrap on a pre-8 Android phone. Configure it for iBeacon detection using a BeaconParser with the iBeacon layout. iBeacons are detected if the app is in the foreground or background. Quit the app using the "running apps" menu on Android and wait until the app is restarted. Now only AltBeacons are detected rather than iBeacons.

*Note* I think what's happening is that the call to add the iBeacon BeaconParser is not having any effect. If I modify the AltLibrary code to make iBeacon the default then it works.

### Mobile device model and OS version

Various phones running Android pre-8.

### Android Beacon Library version

2.13.1

IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

davidgyoung, 2018-06-15:
Hi, @geoffday67, I'm happy to help with this, but the issue is most certainly app-specific with the way the parser is registered in the app's custom code.   I'd suggest you post a question on StackOverflow.com as that is the preferred forum for app troubleshooting assistance, posting an excerpt of your code that shows where you register the BeaconParser as part of the app lifecycle.  Please feel free to link to that question from a comment on this issue.

In the unlikely even that this turns out to be a bug in the library, I'll re-open this issue here.

closed 2018-06-15 by davidgyoung

geoffday67, 2018-06-18:
Added a question on StackOverflow: https://stackoverflow.com/questions/50910448/altbeacon-not-detecting-ibeacon-after-app-restart-on-android-pre-8