AltBeacon/android-beacon-library #707: Oreo: App never wakes up

quentin7b, 2018-06-21:
### Expected behavior

Following the tutorial to [wake up the application](https://altbeacon.github.io/android-beacon-library/samples.html)

I want to start a Service **each time** an iBeacon with a **specific id** is found.

In oder to simulate the iBeacon, i use another android phone and an iBeacon simulator.

The Service should be started when I enable the iBeacon simulator.

### Actual behavior

The service never starts.

### Steps to reproduce this behavior

Start an app with:

`AltBeaconApplication.kt`

```kotlin
class AltBeaconApplication : Application(), BootstrapNotifier {
    private var regionBootstrap: RegionBootstrap? = null

    override fun onCreate() {
        super.onCreate()
        Log.d("BLE Test", "App started up")
        val beaconManager = BeaconManager.getInstanceForApplication(this)
        BeaconManager.setDebug(true) // Deprecated 
        beaconManager.beaconParsers.add(BeaconParser().setBeaconLayout("m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24")) // iBeacon
        beaconManager.backgroundBetweenScanPeriod = 30 * 1000 // 30 seconds
        beaconManager.backgroundScanPeriod = 10000 // 10 seconds (debug)
        val identifier = Identifier.parse("4cfbafdc-4e55-43a9-9bb4-f6a90b46f47a") // Set in the app that simulate the beacon
        val region = Region("fr.myapp.boostrapRegion", identifier, null, null)
        regionBootstrap = RegionBootstrap(this, region)
        BackgroundPowerSaver(this)
    }

    override fun didDetermineStateForRegion(arg0: Int, arg1: Region) {
        // Don't care
    }

    override fun didEnterRegion(arg0: Region) {
        Log.d("BLE Test", "Got a didEnterRegion call")
        val intent = Intent(this, AdvertisingForegroundService::class.java)
        if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.O) {
            this.startForegroundService(intent)
        } else {
            this.startService(intent)
        }
    }

    override fun didExitRegion(arg0: Region) {
        Log.d("BLE Test", "Got a didExitRegion call")
        val intent = Intent(this, AdvertisingForegroundService::class.java)
        this.stopService(intent)
    }
}
```

`AdvertisingForegroundService.kt`
```kotlin
class AdvertisingForegroundService : Service() {

    override fun onBind(intent: Intent?): IBinder {
        TODO("not implemented") //To change body of created functions use File | Settings | File Templates.
    }

    override fun onStartCommand(intent: Intent?, flags: Int, startId: Int): Int {
        Log.d("BLE Test", "onStartCommand triggered")

        context.showNotification() // Display a notification for the test
        startForeground(1, notification)
        // stopForeground(true) // Never ends for the test
        // Ended
        Log.d("BLE Test", "onStartCommand return")
        return START_STICKY
    }
}
```

1. I start the application, then I kill it.
2. I start the iBeacon simulator (with the id `4cfbafdc-4e55-43a9-9bb4-f6a90b46f47a`)
3. Nothing never happen.

### Mobile device model and OS version

Using Oneplus 6 with Android Oreo 8.1.0.

### Android Beacon Library version

2.14

### NB

> I don't know if the bug is in the library, or in my implementation, so I posted here, if it is my implementation, I'll open a StackOverflow.com question.

davidgyoung, 2018-06-21:
I think there may be a number of issues with this implementation:

* On Android 8+ scans are performed via the JobScheduler which runs at most every 15 minutes in the background.  (So it will ignore the 30 second backgroundBetweenScanPeriod configured).   See here for detailed discussion of how this works: http://www.davidgyoungtech.com/2017/08/07/beacon-detection-with-android-8

* The code shown starts a foreground service, which has no effect on scanning rates so long as JobScheduler is being used for scans (which is the default on Android 8+), and can be turned off with `beaconManager.setEnableScheduledScanJobs(false)`

* Killing the app with the task switcher adds other complexities.  If you kill it when it is actively scanning for beacons (in the foreground, or during the rare background scans) then the next scan opportunity will be in ~15 minutes when the JobScheduler kicks in.  It may take that long to re-launch on detection.  If you kill the app in the background during the between scan period, then detection will be possible right away because a hardware filter will be active that will launch the app based on beacon intent.

Bottom line, there are lots of variables here which could cause inconsistent behavior, but if you get a detection within ~15 minutes in most of these cases, I'd say the library is working as designed.  

This probably isn't a good fit for StackOverflow.com unless you can narrow the variables.  I'm going to close this as an issue here because I do think it is working as designed.  I am happy to reopen if you do further tests and find that there is a library bug or if you are suggesting a specific library design change.

closed 2018-06-21 by davidgyoung

quentin7b, 2018-06-22:
Hi David,

First of all, thanks for the reply. I think I got it.
I understand the issue closing. But for the record I'll try to sum up.

**The need is to start a foreground service each time a beacon with a specific id is found, even if the app is closed (works with a wake up)**

1. On Oreo, with the Fast Detection it should work for the first beacon, but the fact that I'm killing the app to test it is causing of the non working behavior ?
2. In the case that Fast Detection works and the `BootstrapNotifier` is triggered, the minimum amount of time before another trigger is 15min (because of Oreo min time)

I'm sorry for all those questions, I just don't get how Android (and the library) is dealing with the BLE and the fast detection (I understand the background limitations and I've read http://www.davidgyoungtech.com/2017/08/07/beacon-detection-with-android-8 . it just seems dark to me as `startScan(x,y, PendingIntent)` should work everytime without background limitations ?)

Thanks for the time you're spending, I really think it will help understand the library (and Android Oreo) behavior.

davidgyoung, 2018-06-22:
The library can be used to detect beacons in two very different ways (which is hidden from casual users to hide the complexity):

1. Using a long-running scanning service.  This is the default on Android 4.3-7.x, and will NOT use intent-based delivery of scan results.  It delivers results directly to the long-running scanning service via a callback. If you want to use this mechanism on Android 8+, you must do two things.  (1) enable it with `beaconManager.setEnableScheduledScanJobs(false)`.  (2) Do something to stop the OS from killing the scanning service after 10 min in the background.  A foreground service might help here, but I have heard inconsistent reports of whether a second foreground service will allow the beacon scanning service to keep going.  Making the scanning service itself work as a foreground service is a new feature not yet released, but available here:  #663 

2. Using a JobScheduler job that runs every ~15 minutes in the background.  Between these background runs, the library will use intent-based delivery of scan results between background scan periods ONLY if no beacons are visible.  However, if you kill the app during an active scan period, this won't be set up, and it will be 15 minutes before it is enabled again.

Bottom line:  Killing the app in scenario 2 *MAY* cause a 15 minutes period where no beacons may be detected.  After this period is over, fast detections will happen normally.  This is more of a testing issue than it is a real life issue, because real users will not kill the app form the task switcher and then expect immediate detections.

quentin7b, 2018-06-26:
Again, thanks for the reply. 

I got it. In my case the ~15min is not acceptable because I have to wake the app up maybe 2 times in less than 5 minutes. 
So the only option I have seems to be a foreground service. 

Thanks again !