AltBeacon/android-beacon-library #710: Device stops scanning after a long time (usually one day)

pedreduardo, 2018-07-13:
### Expected behavior
Device scanning for a long time (more than one day)

### Actual behavior
Device stop scanning after a while (some hours)
The code and logcat can be seen [here](https://stackoverflow.com/questions/51154377/altbeacon-stop-scanning-after-a-long-time)

### Steps to reproduce this behavior
Just let the device scanning for a long time

### Mobile device model and OS version
Tablet Multilaser M10A

### Android Beacon Library version
'org.altbeacon:android-beacon-library:2+'

IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

KonsomeJona, 2020-07-08:
I am running into the same problem:

We setup foreground service and call startRangingBeaconsInRegion, everything works fine.

After 12~24h (it's never consistent), all advertisers suddenly stop being detected (didExitRegion is called). 

Turning off/on Bluetooth doesn't fix it. Turning off/on Location services doesn't fix it. Killing and restarting the application doesn't fix it. The only way to fix it is to reboot the device somehow.

This happens to most of the devices we tested (Samsung, Aquos, ...), any version of Android. 

Searching around, it might be related to the Bluetooth stack:
https://devzone.nordicsemi.com/f/nordic-q-a/61119/android-ble-library-scanner-stops-scanning-after-many-hours
https://blog.classycode.com/a-short-story-about-android-ble-connection-timeouts-and-gatt-internal-errors-fa89e3f6a456

Has someone found a work around for this so far?

davidgyoung, 2020-07-08:
@KonsomeJona, this issue is reserved for Tablet Multilaser M10A.  Please comment here only if you have something to add about that specific device

I will note that your problem report is too broad to be actionable.   I have build a number of apps that do constant ranging with a foreground service that run for many days successfully.  While Android is quite fragmented and many devices have Bluetooth stack problems, I can confirm that you can make this work on common newer Samsung devices.  I would suggest you focus troubleshooting on the newest and most common Samsung or Pixel device where you see this, analyzing OS logging around the time results stop coming in.  Find the cause of results stopping on that one device, fix it and test other device models one by one.   If you find a specific problem please open a new issue that includes details like a log capture.

Again, this issue is about "Tablet Multilaser M10A".  Unless you have comments about problems with that device, please open a new issue that is appropriately narrow.