AltBeacon/android-beacon-library #711: Ibeacons are automatically unregistering if beacons are more than 500?

saty932, 2018-07-16:
When the app launches for the very first time I am registering some beacons to monitor it’s able to register and monitor too but if i relaunch the app again all beacons are unregistered I mean getMonitoredRegions() returning empty array this is only happening for more than 500 beacons.Is there any reason for it?

davidgyoung, 2018-07-16:
This behavior is as designed. 

A maximum of 50 monitored regions will has their state persisted across app restarts. If you go beyond 50 monitored regions, then this persistence will no longer happen, and you will need to re-register the regions at each app startup.  The app will receive duplicate region entry events for nearby beacons each time the operating system restarts the app due to memory pressure.

The library has this limit to keep from performance issues at app startup caused by deserializing a very large number of regions from persistent storage.

As an alternative to monitoring a very large number of regions, consider using broader wildcard patterns in a smaller number of regions (basically specify fewer identifiers that must be matched) and then read the specific identifiers in ranging callbacks and custom logic based on the identifier patterns you see there.

It is worth noting that with iOS CoreLocation, there is also no limit to the number of regions that may be ranged. The limit of 20 regions is for monitoring only, and applies to both geofence regions and beacon regions.

closed 2018-07-16 by davidgyoung

davidgyoung, 2018-07-17:
The library does not have any geofence support at all -- it is focussed on beacons.  if you are using Google Play Services geofences, the limit is 100.