AltBeacon/android-beacon-library #713: Distance calculation for EddystoneUID

rickul, 2018-07-24:
### Expected behavior
It calculates roughly the same distance as for IBeacons.

### Actual behavior
The calculated distance for EddystoneURL is only about a hundredth of IBeacons.

Background:
I have a bluetooth device which can send up to five different advertisements. I configured this device to send two IBeacons and one EddystoneUID. All have the same transmission power and reference rssi. The two IBeacons actually have about the same distance, but the calculated EddystoneUID distance is far smaller (by factor 100 approx.)

### Steps to reproduce this behavior
Configure bluetooth devices with at least one IBeacon and at least one EddystoneUID with the same transmission power and reference RSSI.

### Mobile device model and OS version
HMD Global TA-1012 (Nokia 8), Android 8.1.0

### Android Beacon Library version
2.14

IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

davidgyoung, 2018-07-24:
Frustratingly, Eddystone beacon formats use a different reference value for their Measured Power field.  On iBeacon and AltBeacon, MeasuredPower (aka txPower) is the expected rssi at 1 meter.  On Eddystone, it is the **expected rssi at 0 meters.**

It sounds like your Eddystone beacons are incorrectly transmitting the same MeasuredPower as your iBeacon.  For proper distance estimates, add 41 to the MeasuredPower value for iBeacon to get the value for Eddystone, which converts from a 1 meter reference to a 0 meter reference.

closed 2018-07-24 by davidgyoung

rickul, 2018-07-24:
Yep, thats it! Thanks. I just recalibrated the eddystoneUID beacon with a reference RSSI -9, now the calculated distance is reasonable.

prayagupd, 2019-11-23:
```kotlin
// 2nd byte is tx power at 0m 
// as per https://github.com/google/eddystone/tree/master/eddystone-uid#frame-specification
val measuredPowerAt_0_m = namespaceMetadataBytes.get(1).toInt() 

val measuredPowerAt_1_m = measuredPowerAt_0_m - 41
```

which is `-99dbm` for Ibks beacon.

The distance is not accurate in pixel(android 10) though

```kotlin
    fun estimatedDistance(): Double {
        val pathLossParam = 2
        val ratio = (measuredPower.toDouble() - rsStrengthIndicator) / (10 * pathLossParam)
        val ed = Math.pow(10.0, ratio)

        return trimmedDistance(ed)
    }

    private fun trimmedDistance(ed: Double): Double {
        val bd = BigDecimal(ed).setScale(4, RoundingMode.HALF_UP)
        return bd.toDouble()
    }
```

it gives me distance me centimeters while in real it's more than 1m.

I got the formula from conversation at https://stackoverflow.com/q/52962218/432903