AltBeacon/android-beacon-library #719: Not allowed to start service Intent : BeaconIntentProcessor

GwadaManiac, 2018-08-07:
Hello,

I am using an Android 8 phone (Pixel) and my app crashs in background when a beacon has been detected. The android beacon library version used is 2.13.1 .

> java.lang.IllegalStateException: Not allowed to start service Intent { cmp=fr.mypackage.app/org.altbeacon.beacon.BeaconIntentProcessor (has extras) }: app is in background uid UidRecord{65add7d u0a309 CAC  bg:+3m38s270ms idle change:idle procs:1 seq(0,0,0)}
>         at android.app.ContextImpl.startServiceCommon(ContextImpl.java:1521)
>         at android.app.ContextImpl.startService(ContextImpl.java:1477)
>         at android.content.ContextWrapper.startService(ContextWrapper.java:650)
>         at org.altbeacon.beacon.service.Callback.call(Callback.java:69)
>         at org.altbeacon.beacon.service.MonitoringStatus.updateNewlyOutside(MonitoringStatus.java:103)
>         at org.altbeacon.beacon.service.BeaconService$1.onCycleEnd(BeaconService.java:363)
>         at org.altbeacon.beacon.service.scanner.CycledLeScanner.finishScanCycle(CycledLeScanner.java:251)
>         at org.altbeacon.beacon.service.scanner.CycledLeScanner.scheduleScanCycleStop(CycledLeScanner.java:243)
>         at org.altbeacon.beacon.service.scanner.CycledLeScanner$1.run(CycledLeScanner.java:239)
>         at android.os.Handler.handleCallback(Handler.java:790)
>         at android.os.Handler.dispatchMessage(Handler.java:99)
>         at android.os.Looper.loop(Looper.java:164)
>         at android.app.ActivityThread.main(ActivityThread.java:6494)
>         at java.lang.reflect.Method.invoke(Native Method)
>         at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:438)
>         at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:807)
> 08-07 16:46:58.731 16467-16467/fr.mypackage.app E/ContextWrapper.java: [startService:650]Uncaught exception at  - Not allowed to start service Intent { cmp=fr.mypackage.app/org.altbeacon.beacon.BeaconIntentProcessor (has extras) }: app is in background uid UidRecord{65add7d u0a309 CAC  bg:+3m38s270ms idle change:idle procs:1 seq(0,0,0)}

davidgyoung, 2018-08-07:
Please check your code to confirm if you have added a call like this: `beaconManager.setEnableScheduledScanJobs(false);`

If you have done so, then you *will* get exceptions like this if you try to detect beacons in the background on Android 8+, because Android disallows apps from starting new services in the background.  To fix this, simply remove the line like above s the library uses its default value for this setting of `true` on Android 8+.  Alternately, you can explicitly set it to true with `beaconManager.setEnableScheduledScanJobs(true);`

GwadaManiac, 2018-08-08:
I have not call the function to disabled scheduled scan jobs. That's why i don't know why i get this exception. I have updated the library to the 2.15 one and call `beaconManager.setEnableScheduledScanJobs(true);` . Still the same.
In the Callback class, we enter in the useLocalBroadcast case to send the intent.

davidgyoung, 2018-08-08:
I cannot explain what you describe.  The stack trace indicates that this line of code is evaluating to false indicating that `getScheduledScanJobsEnabled()` returns false.

https://github.com/AltBeacon/android-beacon-library/blob/master/src/main/java/org/altbeacon/beacon/service/Callback.java#L58

How could this be possible?

GwadaManiac, 2018-08-10:
Indeed, I'm using a foreground service in that case.

davidgyoung, 2018-08-10:
It's unclear how the foreground service is set up, but if you have set it up per this library's documentation instructions [here](https://altbeacon.github.io/android-beacon-library/foreground-service.html) then you would indeed have set `beaconManager.setEnableScheduledScanJobs(false);` which is what would cause the problem.  If you set that flag, and then somehow don't have the foreground service successfully operating, then you will get this error because Android will block your service from starting up as needed unless it is a foreground service.

davidgyoung, 2018-08-11:
@GwadaManiac, I have a theory about what is going on here.  I put in a proposed fix that I have in a beta release here:  https://github.com/AltBeacon/android-beacon-library/releases/tag/2.15.1-beta1

Please give this a try and let me know if it solves the problem.

GwadaManiac, 2018-08-19:
Hello,

Thanks for the patch.
I don't reproduce this issue anymore !

davidgyoung, 2018-08-19:
Thanks for the confirmation.  This will be in the next release.

davidgyoung, 2018-09-01:
this is now in the latest release on Bintray/JCenter and in the release section of this site

balchen, 2018-09-24:
We've experienced the same issue and have upgraded to 2.15.1. For us, the issue still re-surfaces from time to time.

It seems BeaconManager.getInstanceForApplication(context).isMainProcess() returns false and as a result, the BeaconIntentProcessor is started as a service (causing the well-known java.lang.IllegalStateException: Not allowed to start service Intent).

This code seems to have been the target of the change made in 2.15.1, so I decided to post it here and re-open rather than creating a new issue.

Here's a log excerpt from today. Device is a Motorola Moto e5 with Android 8.1.  The app is a Xamarin.Android app using AltBeacon via Xamarin bindings.

`09-24 11:49:19.469	Motorola moto e5 play	Error	22615	AndroidRuntime	java.lang.IllegalStateException: Not allowed to start service Intent { cmp=no.utviklerne.forum.guide.dev/org.altbeacon.beacon.BeaconIntentProcessor (has extras) }: app is in background uid UidRecord{165d367 u0a110 CAC  bg:+1m0s113ms idle change:idle procs:1 seq(0,0,0)}
	at android.app.ContextImpl.startServiceCommon(ContextImpl.java:1536)
	at android.app.ContextImpl.startService(ContextImpl.java:1492)
	at android.content.ContextWrapper.startService(ContextWrapper.java:650)
	at org.altbeacon.beacon.service.Callback.call(Callback.java:72)
	at org.altbeacon.beacon.service.BeaconService.processRangeData(BeaconService.java:402)
	at org.altbeacon.beacon.service.BeaconService.access$300(BeaconService.java:78)
	at org.altbeacon.beacon.service.BeaconService$1.onCycleEnd(BeaconService.java:364)
	at org.altbeacon.beacon.service.scanner.CycledLeScanner.finishScanCycle(CycledLeScanner.java:278)
	at org.altbeacon.beacon.service.scanner.CycledLeScanner.scheduleScanCycleStop(CycledLeScanner.java:269)
	at org.altbeacon.beacon.service.scanner.CycledLeScanner$1.run(CycledLeScanner.java:265)
	at android.os.Handler.handleCallback(Handler.java:790)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loop(Looper.java:164)
	at android.app.ActivityThread.main(ActivityThread.java:6626)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:438)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:811)`

davidgyoung, 2018-09-24:
@balchen, to your knowledge, has your app disabled scheduled scan jobs? 

 I don't think the problem has to do with ` BeaconManager.getInstanceForApplication(context).isMainProcess()` returning false for your app, as that is no longer part of the criteria to decide whether to use local broadcasts for data delivery as of library version 2.15.1.  The code changed to this:

```
        boolean useLocalBroadcast = BeaconManager.getInstanceForApplication(context).getScheduledScanJobsEnabled();
```

From this:

```
        boolean useLocalBroadcast = BeaconManager.getInstanceForApplication(context).isMainProcess();
```

So if you (or the Cordova plugin) disables ScanJobs, it would cause this problem.  In hindsight, I can't think of a good reason the code is not this:

```
        boolean useLocalBroadcast = BeaconManager.getInstanceForApplication(context).getScheduledScanJobsEnabled() || BeaconManager.getInstanceForApplication(context).isMainProcess();
```

This code change might solve your problem.  I could prepare a test build for you to try, but I want to understand the problem before I do.  Can you confirm through looking at your LogCat logs whether ScanJob or the BeaconService is used to do scanning?

balchen, 2018-09-24:
@davidgyoung, I've checked and we do not disable scheduled scan jobs anywhere in our code. We set background and foreground scan and between scan periods, add beacon parsers, handle ranging and monitoring callbacks, start monitoring or ranging, and enable/disable background mode.

What would I look for in the logs to determine if one or the other is used to do scanning? I can't remember seeing any log entries relevant to this except for the service failure log entry.

I have found this gem once, though (my comment continues after log), which looks like it might imply scheduling is used.

`Time	Device Name	Type	PID	Tag	Message
09-23 12:59:17.692	Sony G8441	Info	18859	chatty	uid=10398(u0_a398) no.utviklerne.forum.guide.must expire 1 line
	at org.altbeacon.beacon.service.scanner.CycledLeScanner.scanLeDevice(CycledLeScanner.java:237)
	at org.altbeacon.beacon.service.scanner.CycledLeScanner.finishScanCycle(CycledLeScanner.java:311)
	at org.altbeacon.beacon.service.scanner.CycledLeScanner.scheduleScanCycleStop(CycledLeScanner.java:269)
	at org.altbeacon.beacon.service.scanner.CycledLeScanner.scanLeDevice(CycledLeScanner.java:237)
	at org.altbeacon.beacon.service.scanner.CycledLeScanner.finishScanCycle(CycledLeScanner.java:311)
	at org.altbeacon.beacon.service.scanner.CycledLeScanner.scheduleScanCycleStop(CycledLeScanner.java:269)
	at org.altbeacon.beacon.service.scanner.CycledLeScanner.scanLeDevice(CycledLeScanner.java:237)
	at org.altbeacon.beacon.service.scanner.CycledLeScanner.finishScanCycle(CycledLeScanner.java:311)
	at org.altbeacon.beacon.service.scanner.CycledLeScanner.scheduleScanCycleStop(CycledLeScanner.java:269)
	at org.altbeacon.beacon.service.scanner.CycledLeScanner.scanLeDevice(CycledLeScanner.java:237)
	at org.altbeacon.beacon.service.scanner.CycledLeScanner.finishScanCycle(CycledLeScanner.java:311)
	at org.altbeacon.beacon.service.scanner.CycledLeScanner.scheduleScanCycleStop(CycledLeScanner.java:269)
	at org.altbeacon.beacon.service.scanner.CycledLeScanner.scanLeDevice(CycledLeScanner.java:237)
	at org.altbeacon.beacon.service.scanner.CycledLeScanner.finishScanCycle(CycledLeScanner.java:311)
	at org.altbeacon.beacon.service.scanner.CycledLeScanner.scheduleScanCycleStop(CycledLeScanner.java:269)
	at org.altbeacon.beacon.service.scanner.CycledLeScanner.scanLeDevice(CycledLeScanner.java:237)
	at org.altbeacon.beacon.service.scanner.CycledLeScanner.finishScanCycle(CycledLeScanner.java:311)
	at org.altbeacon.beacon.service.scanner.CycledLeScanner.scheduleScanCycleStop(CycledLeScanner.java:269)
	at org.altbeacon.beacon.service.scanner.CycledLeScanner.scanLeDevice(CycledLeScanner.java:237)
	at org.altbeacon.beacon.service.scanner.CycledLeScanner.finishScanCycle(CycledLeScanner.java:311)
	at org.altbeacon.beacon.service.scanner.CycledLeScanner.scheduleScanCycleStop(CycledLeScanner.java:269)
	at org.altbeacon.beacon.service.scanner.CycledLeScanner.scanLeDevice(CycledLeScanner.java:237)
	at org.altbeacon.beacon.service.scanner.CycledLeScanner.finishScanCycle(CycledLeScanner.java:311)
	at org.altbeacon.beacon.service.scanner.CycledLeScanner.scheduleScanCycleStop(CycledLeScanner.java:269)
	at org.altbeacon.beacon.service.scanner.CycledLeScanner.scanLeDevice(CycledLeScanner.java:237)
	at org.altbeacon.beacon.service.scanner.CycledLeScanner.finishScanCycle(CycledLeScanner.java:311)
	at org.altbeacon.beacon.service.scanner.CycledLeScanner.scheduleScanCycleStop(CycledLeScanner.java:269)
	at org.altbeacon.beacon.service.scanner.CycledLeScanner.scanLeDevice(CycledLeScanner.java:237)
	at org.altbeacon.beacon.service.scanner.CycledLeScanner.finishScanCycle(CycledLeScanner.java:311)
	at org.altbeacon.beacon.service.scanner.CycledLeScanner.scheduleScanCycleStop(CycledLeScanner.java:269)
	at org.altbeacon.beacon.service.scanner.CycledLeScanner.scanLeDevice(CycledLeScanner.java:237)
	at org.altbeacon.beacon.service.scanner.CycledLeScanner.finishScanCycle(CycledLeScanner.java:311)
	at org.altbeacon.beacon.service.scanner.CycledLeScanner.scheduleScanCycleStop(CycledLeScanner.java:269)
	at org.altbeacon.beacon.service.scanner.CycledLeScanner.scanLeDevice(CycledLeScanner.java:237)
	at org.altbeacon.beacon.service.scanner.CycledLeScanner.finishScanCycle(CycledLeScanner.java:311)
	at org.altbeacon.beacon.service.scanner.CycledLeScanner.scheduleScanCycleStop(CycledLeScanner.java:269)
	at org.altbeacon.beacon.service.scanner.CycledLeScanner.scanLeDevice(CycledLeScanner.java:237)
	at org.altbeacon.beacon.service.scanner.CycledLeScanner.finishScanCycle(CycledLeScanner.java:311)
	at org.altbeacon.beacon.service.scanner.CycledLeScanner.scheduleScanCycleStop(CycledLeScanner.java:269)
	at org.altbeacon.beacon.service.scanner.CycledLeScanner.scanLeDevice(CycledLeScanner.java:237)
	at org.altbeacon.beacon.service.scanner.CycledLeScanner.finishScanCycle(CycledLeScanner.java:311)
	at org.altbeacon.beacon.service.scanner.CycledLeScanner.scheduleScanCycleStop(CycledLeScanner.java:269)
	at org.altbeacon.beacon.service.scanner.CycledLeScanner.scanLeDevice(CycledLeScanner.java:237)
	at org.altbeacon.beacon.service.scanner.CycledLeScanner.finishScanCycle(CycledLeScanner.java:311)
	at org.altbeacon.beacon.service.scanner.CycledLeScanner.scheduleScanCycleStop(CycledLeScanner.java:269)
	at org.altbeacon.beacon.service.scanner.CycledLeScanner.scanLeDevice(CycledLeScanner.java:237)
	at org.altbeacon.beacon.service.scanner.CycledLeScanner.finishScanCycle(CycledLeScanner.java:311)
[... cut ...]
	at org.altbeacon.beacon.service.scanner.CycledLeScanner.scheduleScanCycleStop(CycledLeScanner.java:269)
	at org.altbeacon.beacon.service.scanner.CycledLeScanner.scanLeDevice(CycledLeScanner.java:237)
	at org.altbeacon.beacon.service.scanner.CycledLeScanner.finishScanCycle(CycledLeScanner.java:311)
	at org.altbeacon.beacon.service.scanner.CycledLeScanner.scheduleScanCycleStop(CycledLeScanner.java:269)
	at org.altbeacon.beacon.service.scanner.CycledLeScanner.scanLeDevice(CycledLeScanner.java:237)
	at org.altbeacon.beacon.service.scanner.CycledLeScanner.finishScanCycle(CycledLeScanner.j

balchen, 2018-09-24:
Oh, hang on. Is this what you're looking for?

`Time	Device Name	Type	PID	Tag	Message
09-25 00:42:57.131	Sony G8441	Info	21546	ScanJob	Running immediate scan job: instance is org.altbeacon.beacon.service.ScanJob@5d613f9
`

davidgyoung, 2018-09-24:
yes, that confirms that it is using scan jobs.  and since this is now what
determines that local notifications will be used instead of an intent
service, I do not understand your crash.  I think you would have to capture
a log excerpt leading up to the crash to see what caused it.

balchen, 2018-09-25:
I've looked at the logs some more, and it does seem like we have some weird behaviour. We have two test devices; a Motorola and a Sony. Both run the same version of the same app. The Sony runs Android 8.0 and the Moto runs Android 8.1. The Sony doesn't exhibit any issues, and this is where I found the log entry for a ScanJob. The Motorola has issues, and we cannot find any mention of ScanJob in the Motorola log. It has entries for BeaconService.

```
Time	Device Name	Type	PID	Tag	Message
09-24 13:56:56.930	Motorola moto e5 play	Info	1468	chatty	uid=1000(system) Binder:1468_D expire 8 lines
09-24 14:08:05.878	Motorola moto e5 play	Info	1468	chatty	uid=1000(system) CpuTracker expire 6 lines
09-24 14:07:35.927	Motorola moto e5 play	Info	1468	chatty	uid=1000(system) batterystats-wo expire 12 lines
09-24 14:07:12.552	Motorola moto e5 play	Info	1468	chatty	uid=1000(system) Binder:1468_6 expire 7 lines
09-24 14:07:12.550	Motorola moto e5 play	Info	1468	chatty	uid=1000(system) Binder:1468_13 expire 1 line
09-24 14:07:12.110	Motorola moto e5 play	Error	29618	AndroidRuntime	Process: no.utviklerne.forum.guide.dev, PID: 29618
09-24 14:07:12.110	Motorola moto e5 play	Error	29618	AndroidRuntime	FATAL EXCEPTION: main
09-24 14:07:12.110	Motorola moto e5 play	Error	29618	AndroidRuntime	java.lang.IllegalStateException: Not allowed to start service Intent { cmp=no.utviklerne.forum.guide.dev/org.altbeacon.beacon.BeaconIntentProcessor (has extras) }: app is in background uid UidRecord{2678f24 u0a110 CAC  bg:+1m0s115ms idle change:idle procs:1 seq(0,0,0)}
	at android.app.ContextImpl.startServiceCommon(ContextImpl.java:1536)
	at android.app.ContextImpl.startService(ContextImpl.java:1492)
	at android.content.ContextWrapper.startService(ContextWrapper.java:650)
	at org.altbeacon.beacon.service.Callback.call(Callback.java:72)
	at org.altbeacon.beacon.service.BeaconService.processRangeData(BeaconService.java:402)
	at org.altbeacon.beacon.service.BeaconService.access$300(BeaconService.java:78)
	at org.altbeacon.beacon.service.BeaconService$1.onCycleEnd(BeaconService.java:364)
	at org.altbeacon.beacon.service.scanner.CycledLeScanner.finishScanCycle(CycledLeScanner.java:278)
	at org.altbeacon.beacon.service.scanner.CycledLeScanner.scheduleScanCycleStop(CycledLeScanner.java:269)
	at org.altbeacon.beacon.service.scanner.CycledLeScanner$1.run(CycledLeScanner.java:265)
	at android.os.Handler.handleCallback(Handler.java:790)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loop(Looper.java:164)
	at android.app.ActivityThread.main(ActivityThread.java:6626)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:438)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:811)
09-24 14:07:12.087	Motorola moto e5 play	Info	1468	chatty	uid=1000(system) Binder:1468_3 expire 16 lines
09-24 14:07:09.318	Motorola moto e5 play	Info	1468	chatty	uid=1000(system) UEventObserver expire 13 lines
09-24 14:07:00.027	Motorola moto e5 play	Info	629	chatty	uid=1000(system) EventThread expire 10 lines
09-24 14:06:25.972	Motorola moto e5 play	Info	1468	chatty	uid=1000(system) Binder:1468_C expire 5 lines
09-24 14:06:25.634	Motorola moto e5 play	Info	1468	chatty	uid=1000(system) HwBinder:1468_1 expire 1 line
09-24 14:06:25.330	Motorola moto e5 play	Warning	29867	ContextImpl	Calling a method in the system process without a qualified user: android.app.ContextImpl.sendBroadcast:984 android.content.ContextWrapper.sendBroadcast:442 com.android.settings.location.LocationSettingsBase.updateLocationMode:97 com.android.settings.location.LocationSettingsBase.setLocationMode:118 com.android.settings.location.LocationSettings.onSwitchChanged:406 
09-24 14:06:11.903	Motorola moto e5 play	Info	1468	chatty	uid=1000(system) InputDispatcher expire 7 lines
09-24 14:05:56.216	Motorola moto e5 play	Info	1468	chatty	uid=1000(system) Binder:1468_2 expire 11 lines
09-24 14:05:45.962	Motorola moto e5 play	Info	1468	chatty	uid=1000(system) ActivityManager expire 11 lines
09-24 14:05:43.526	Motorola moto e5 play	Info	629	chatty	uid=1000(system) /system/bin/surfaceflinger expire 35 lines
09-24 14:05:42.611	Motorola moto e5 play	Info	1468	chatty	uid=1000(system) Binder:1468_F expire 1 line
09-24 14:05:41.822	Motorola moto e5 play	Warning	29867	ContextImpl	Calling a method in the system process without a qualified user: android.app.ContextImpl.sendBroadcast:984 android.content.ContextWrapper.sendBroadcast:442 com.android.settings.location.LocationSettingsBase.updateLocationMode:97 com.android.settings.location.LocationSettingsBase.setLocationMode:118 com.android.settings.location.LocationSettings.onSwitchChanged:404 
09-24 14:05:37.958	Motorola moto e5 play	Info	1468	chatty	uid=1000(system) UEventObserver expire 10 lines
09-24 14:05:27.520	Motorola moto e5 play	Info	629	chatty	uid=1000(system) EventThread expire 17 lines
09-24 14:05:13.377	Motorola moto e5 play	Info	629	chatty	uid=1000(system) /system/bin/surfaceflinger expire 12 lines
09-24 14:05:12.512	Motorola moto e5 play	Warning	29618	ActivityThread	serviceCreate starts in 9737 ms, CreateServiceData = CreateServiceData{token=android.os.BinderProxy@2afb749 className=org.altbeacon.beacon.service.BeaconService packageName=no.utviklerne.forum.guide.dev intent=null}
09-24 14:04:55.885	Motorola moto e5 play	Warning	29618	ActivityThread	Application no.utviklerne.forum.guide.dev can be debugged on port 8100...
09-24 14:04:53.364	Motorola moto e5 play	Info	1468	chatty	uid=1000(system) android.display expire 14 lines
```

balchen, 2018-09-25:
And I still want to reiterate that I cannot wrap my head around how Callback.java is supposed to call getScheduledScanJobsEnabled(). We're several people who've looked at the code and we all concluded that the current version of Callback.java calls isMainProcess().

[image]

davidgyoung, 2018-09-25:
@balchen, yes, you are right.  Sorry, I had it backwards -- I should know better since I'm the one who made the change! In 2.15.1 it uses the isMainProcess() check to decide whether to use notifications to do the callbacks.  So there are two mysterious behaviors on the  Motorola:

1. It is not using ScanJobs, but instead uses the BeaconService.
2. It is using intents for callbacks instead of notifications, perhaps because the service from step 1 is running in a different process.

I think we need to collect more info about what is happening on the Motorola.  Can you please put the library into debug mode with `beaconManager.setDebug(true);` and then capture a log excerpt at startup?  

Here is what I am looking for when the beacon service starts up:

    I/BeaconManager: BeaconManager started up on pid 13220 named 'XXXXX' for application package 'YYYYYY'.  isMainProcess=true
    D/BeaconManager: Binding to service
    I/BeaconService: beaconService version 2.12.2 is starting up on the main process

And here is what I am looking for when the ScanJob starts up:

    I/BeaconManager: BeaconManager started up on pid 13220 named 'XXXXX' for application package 'YYYYYY'.  isMainProcess=true
    D/BeaconManager: Not starting beacon scanning service. Using scheduled jobs
    I/ScanJob : scanJob version 2.12.2 is starting up on the main process

The lines above will probably be separated by other lines.

I know you are using Cordova, but if you have the ability to log two values, you might try to log these two lines in the `didDetermineState` callback.  I'm not sure the Cordova / JavaScript equivalents:

     Log.d(TAG, "Scan jobs enabled: "+beaconManager.getScheduledScanJobsEnabled());
     Log.d(TAG, "Android version: "+android.os.Build.VERSION.SDK_INT);

balchen, 2018-09-25:
@davidgyoung Sorry about the delays; the time zone difference must be significant.

We've done some more testing today (before you posted the latest message) and the results aren't very conclusive. 

Before 2.15.1, we were able to provoke crashes by disabling and enabling Bluetooth, or by running HockeyApp (oddly enough), or by running other beacon scanning apps. Either one would cause our apps to wake up and subsequently crash.

This time, though, the crash happened when Google location services settings were changed, and we haven't been able to reproduce it consistently.

We build multiple apps from the same codebase, and only one of them -- one that was deployed through Visual Studio (Xamarin) -- exhibited the issue. From this deployment, we've never seen any messages concerning scan jobs at all.

The other apps were deployed from HockeyApp and worked fine on all test devices. We're thinking there's a possibility that the app deployed from VS is a botched or incomplete deployment that has left v2.12 behind somehow. Is that consistent with the log messages?

We do not disable scheduled jobs, so my reasoning is getScheduledScanJobsEnabled() should still return true and Callback.java, even in v2.12, should use local broadcast instead of a service. Which means even v2.12 doesn't explain why the service is being used. So I'm not certain that what we're seeing is v2.12.

We'll do some more testing tomorrow, and if we see the issue popping up, we'll enable debug logging and log the values you suggested.

davidgyoung, 2018-09-25:
Hey, @balchen, I think there is a misunderstanding.  Those log lines I
pasted that mention 2.12 are just examples of what you might see.  I need
you to capture a log that shows the equivalent lines for your app running
on the Motorola (which should show 2.15.1) and let me know what they look
like

balchen, 2018-09-25:
@davidgyoung No misunderstanding. Like I said, this is what we were doing earlier today and our conclusions, uncertain as they are, were drawn before you posted your message. Inconsistent reproduction, no issues with apps from HockeyApp, and the possibility of a botched deployment from VS (which HAS happened before with Xamarin) means we'll go back and test more.

Really what I was asking about v2.12 and log messages is if the log messages I pasted yesterday would be consistent with v2.12, and if it's even possible to differentiate v2.12 and v2.15.1 based on the log messages I pasted.

Also, I'm wondering -- if it turns out we were running v2.12 somehow -- if the behaviour is consistent with that version. To me it seems like the service still shouldn't have been triggered.

davidgyoung, 2018-10-03:
Closing this issue as it sounds like this may be a build problem. Please reopen if you find it is a library issue.

closed 2018-10-03 by davidgyoung