AltBeacon/android-beacon-library #723: seem missing code

jerry1216m3tw, 2018-08-14:
Hi,
  I use this version org.altbeacon:android-beacon-library:2+ but unbind is like this,different from src code
please check

this.consumers.remove(consumer);
                    if(this.consumers.size() == 0) {
                        this.serviceMessenger = null;
                        this.mBackgroundMode = false;
                        if(this.mScheduledScanJobsEnabled) {
                            ;
                        }
                    }

src code
consumers.remove(consumer);
                if (consumers.size() == 0) {
                    // If this is the last consumer to disconnect, the service will exit
                    // release the serviceMessenger.
                    serviceMessenger = null;
                    // Reset the mBackgroundMode to false, which is the default value
                    // This way when we restart ranging or monitoring it will always be in
                    // foreground mode
                    mBackgroundMode = false;
                    // If we are using scan jobs, we cancel the active scan job
                    if (mScheduledScanJobsEnabled) {
                        if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.LOLLIPOP) {
                            LogManager.i(TAG, "Cancelling scheduled jobs after unbind of last consumer.");
                            ScanJobScheduler.getInstance().cancelSchedule(mContext);
                        }
                    }
                }

davidgyoung, 2018-08-16:
This change is visible in the master branch source code, but has not been put into a binary release yet.   I is planned for inclusion in 2.15.1 when released.  Until then, if you wish you may build your own copy of the library from source to get this change.

closed 2018-08-16 by davidgyoung

davidgyoung, 2018-09-01:
this is now in the latest release on Bintray/JCenter and in the release section of this site

jerry1216m3, 2018-09-02:
Can I ask a question?why start stop your lib fast for 8,9 time it will lost signal for about 10 second and I use android rescan it's the same

jerry1216m3, 2018-09-02:
Can I ask a question?why start stop your lib fast for 8,9 time it will lost signal for about 10 second and I use android lescan it's the same

davidgyoung, 2018-09-02:
Sorry, I don't think I understand the question, @jerry1216m3.   For troubleshooting assistance, please post a new question on StackOverflow.com.  I'm happy to answer there.

jerry1216m3, 2018-09-02:
Ok,I'm posted the question thank you very munch
https://stackoverflow.com/questions/52138917/android-beacon-library-on-and-off-with-a-fast-pace