AltBeacon/android-beacon-library #725: Foreground not working.

markrpg, 2018-08-19:
### Expected behavior
Connecting the BeaconService to a foreground notification should keep it alive even when the screen is off for hours. I have looked into enabling the longScanForcingEnabled flag however as I am using Xamarin to develop, it doesn't fully support manifest tools. Meaning using the tools:replace option to change the BeaconService metadata to true won't work. I have tried every conceivable option to enable this feature but its not possible without building my own bindings library using an AAR, which I have also tried and while trying to run the BeaconService, doesn't callback the OnBeaconServiceConnect method in the consumer. 

### Actual behavior
After around 30minutes the application stops working, until the screen is switched on.

### Mobile device model and OS version
Samsung OS 8.1

### Android Beacon Library version
2.14

IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

davidgyoung, 2018-09-01:
I understand the issue, but I am not enough of a Xamarin expert to offer a solution.  It would be possible to build a programmatic access to enable long scanning.

stepheaw, 2020-01-08:
@markrpg Did you ever resolve this? I am facing the same issue where on Samsung phones. It will not start scanning in the background until the screen is switched on. My BeaconService is also tied to a foreground notification. Phones from other manufacturers are working for this

markrpg, 2020-01-08:
> @markrpg Did you ever resolve this? I am facing the same issue where on Samsung. It will not start scanning until the screen is switched on. My BeaconService is also tied to a foreground notification. Phones from other manufacturers are working for this

I was able to get around this by forcing foreground scanning on by default, through editing the library and building my own binary.