AltBeacon/android-beacon-library #726: The altbeacon library only detect small numbers of beacons

haochen95, 2018-08-20:
### Expected behavior
detect enough number of beacons(In my cases, 160 beacons)
### Actual behavior
only detect small number of beacons (40-80 beacons) even when I put my smartphone just aside the beacons
### Steps to reproduce this behavior
1. I tried to increase the beacon's advertising interval to 300ms, but it still doesn't work
2. I also tried to increase the scan period to 3s, it still detect has similar problems
  My initial scan period setting is :  
`        beaconManager.setForegroundScanPeriod(1000);
        beaconManager.setForegroundBetweenScanPeriod(0);` as I need relatively high scan rate, In my case, I hope to detect beacon every seconds,  I then changed this to `beaconManager.setForegroundScanPeriod(3000);
        beaconManager.setForegroundBetweenScanPeriod(3000);` but it still doesn't work
### Mobile device model and OS version
Xiaomi ReadMi 5 Plus Android 7.1.2 API 25

I use RECOBeacon
### Android Beacon Library version
reco-sdk-android_1.1.4

davidgyoung, 2018-08-20:
The bad news is that the library is probably not what is limiting you here.  The number of beacons you can successfully scan simultaneously is largely dependent on two factors:

1. The bluetooth stack (hardware, firmware, and operating system software) of the Android device model being used.
2. The CPU power of the device.

The first factor is almost always the limiting factor.  The second factor only comes into play if there is heavy processing going on in the beacon callbacks (or of debug logging is turned on in the library.)

I have run tests on a number of Samsung device models, since these are some of the most common on the market.  The Galaxy S5 (Android 6) has better performance than the S8 (Android 7) or S9 (Android 8).  I have seen it detect about 350 unique beacon transmissions over a 5 minute period, but not all of these get detected in the same ranging cycle -- that is typically fewer than 100.  Each ranging cycle a different set of beacons get detected, but over time all of them do.

There is a bug difference from one device model to the next, and price is no predictor.  The S8 and S9 detect fewer than half as many beacons as the S5.  The Nexus 5X (Android 6) is simply terrible, and detects only 10% as many beacons as the S5  The Huawei P9 (Android 7) performs similarly to the Samsung Galaxy S8 and S9.

davidgyoung, 2018-08-20:
The didRangeBeaconsInRegion callback is supposed to do exactly what you say.  You get a list of all beacons that were detected in the ranging cycle NOT JUST NEW ONES.    If you don't see a beacon in the callback, that means that the bluetooth scanner on your phone did not detect it during that period.

haochen95, 2018-08-20:
I got it, Thanks for point

davidgyoung, 2018-09-01:
Closing this issue as this appears to be a device-specific bluetooth issue

closed 2018-09-01 by davidgyoung