AltBeacon/android-beacon-library #727: NullPointer Exception

dzaga71, 2018-08-21:
### Actual behavior
From time to time my app crashes with the following Exception

Caused by java.lang.NullPointerException: Attempt to invoke virtual method 'boolean org.altbeacon.beacon.service.RangedBeacon.isTracked()' on a null object reference
       at org.altbeacon.beacon.service.RangeState.finalizeBeacons(RangeState.java:74)
       at org.altbeacon.beacon.service.ScanHelper.processRangeData(ScanHelper.java:272)
       at org.altbeacon.beacon.service.ScanHelper.access$200(ScanHelper.java:60)
       at org.altbeacon.beacon.service.ScanHelper$1.onCycleEnd(ScanHelper.java:223)
       at org.altbeacon.beacon.service.scanner.CycledLeScanner.finishScanCycle(CycledLeScanner.java:388)
       at org.altbeacon.beacon.service.scanner.CycledLeScanner.scanLeDevice(CycledLeScanner.java:352)
       at org.altbeacon.beacon.service.scanner.CycledLeScanner.stop(CycledLeScanner.java:230)
       at org.altbeacon.beacon.service.ScanJob.stopScanning(ScanJob.java:162)
       at org.altbeacon.beacon.service.ScanJob.onStopJob(ScanJob.java:154)
       at android.app.job.JobService$1.onStopJob(JobService.java:76)
       at android.app.job.JobServiceEngine$JobHandler.handleMessage(JobServiceEngine.java:117)
       at android.os.Handler.dispatchMessage(Handler.java:106)
       at android.os.Looper.loop(Looper.java:164)
       at android.app.ActivityThread.main(ActivityThread.java:6494)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:438)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:807)

### Steps to reproduce this behavior
There is no strict sequence how to reproduce this behaviour. It happens from time to time.

### Mobile device model and OS version
Nexus 6P with Android 8

### Android Beacon Library version
2.12+

davidgyoung, 2018-08-21:
This is a duplicate of #638.   The good news is that a fix has already been put in #716, and will be included in the next binary release 2.15.1.

closed 2018-08-21 by davidgyoung

davidgyoung, 2018-09-01:
this is now in the latest release on Bintray/JCenter and in the release section of this site