AltBeacon/android-beacon-library #728: Leaked ServiceConnection on one smartwatch, but not on the other

maltepaskudafit, 2018-08-22:
### Expected behavior

I have an activity that starts a service that starts a beacon background scan. Init code looks like this:

     @Override
    public void onHandleIntent(Intent intent) {;
        Log.d(TAG, "Setting up beacon scanner");
        beaconManager = BeaconManager.getInstanceForApplication(this.getApplicationContext());
        beaconManager.setAndroidLScanningDisabled(false);
        // only setting a manual scan period allows the test device to find any beacons
        beaconManager.setForegroundScanPeriod(20000);
        beaconManager.setBackgroundScanPeriod(20000);
        beaconManager.setRegionExitPeriod(60000l);
        // ibeacon beacon layout
        beaconManager.getBeaconParsers().add(new BeaconParser().
                setBeaconLayout("m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24"));
        beaconManager.bind(this);
        backgroundPowerSaver = new BackgroundPowerSaver(this);
    }

This is supposed to work, as it works on the Huawei Watch 2 and we find beacons there.

### Actual behavior

On the Moto 360 Sport, we get this error message:

```
08-22 13:24:41.539 2523-2523/? E/ActivityThread: Service com.example.paskuda.sensordisplayj.LocationService has leaked ServiceConnection org.altbeacon.beacon.BeaconManager$BeaconServiceConnection@7db2a9e that was originally bound here
    android.app.ServiceConnectionLeaked: Service com.example.paskuda.sensordisplayj.LocationService has leaked ServiceConnection org.altbeacon.beacon.BeaconManager$BeaconServiceConnection@7db2a9e that was originally bound here
        at android.app.LoadedApk$ServiceDispatcher.<init>(LoadedApk.java:1336)
        at android.app.LoadedApk.getServiceDispatcher(LoadedApk.java:1231)
        at android.app.ContextImpl.bindServiceCommon(ContextImpl.java:1450)
        at android.app.ContextImpl.bindService(ContextImpl.java:1422)
        at android.content.ContextWrapper.bindService(ContextWrapper.java:636)
        at org.altbeacon.beacon.BeaconManager.bind(BeaconManager.java:423)
        at com.example.paskuda.sensordisplayj.LocationService.onHandleIntent(LocationService.java:64)
        at android.app.IntentService$ServiceHandler.handleMessage(IntentService.java:68)
        at android.os.Handler.dispatchMessage(Handler.java:102)
        at android.os.Looper.loop(Looper.java:154)
        at android.os.HandlerThread.run(HandlerThread.java:61)_
````

This is strange, as nothing at this point should have done lifecycle stuff with that connection. Also, beaconManager reports that nothing is bound so far when being asked, so there should not be a connection to leak.

Related or not, after this it goes on the Moto only one time into `didRangeBeaconsInRegion`, and it does not see the available beacons.

### Steps to reproduce this behavior

Use above init code to enable a background scan on the Huawei Watch 2 and the Moto 360 Sport.

### Mobile device model and OS version

Huawei Watch 2 reports:
 
 * Android 8.0.0
 * Wear OS 1.5

Moto 360 Sport reports:

 * Android 7.1.1
 * Wear OS 2.15

### Android Beacon Library version

Gradle says: 2.13.+

davidgyoung, 2018-08-22:
This problem is likely caused by your custom service terminating without calling `unbind` on the `BeaconManager`.   You may be able to prevent this exception by adding this call in a method like this in your custom service:

```
@Override
    protected void onDestroy() {
        super.onDestroy();
        BeaconManager.getinstanceForApplication(this).unbind(this);
    }
```

But that won't change the fact that your service is being terminated.  You may have to dig deeper to find out why that is happening.

If you need further assistance, please open a question on StackOverflow.com, as this is most certainly a project-specific issue and not an issue with the library itself.

closed 2018-08-22 by davidgyoung

maltepaskudafit, 2018-08-29:
Is it not a bit strange and pointing to an issue in the library that it works on one and not he other? I had assumed that rules out the "you use it wrong" argument, at least mostly.

But a short confirmation: The onDestroy call works and shows that the service gets ended immediately, the leak error message is indeed just asymptom for that We will try to look further into that. Thanks for the help so far.

davidgyoung, 2018-08-29:
It is common for mobile devices to customize Android, which explains why your app behaviour es differently on different models.  Motorola closely follows AOSP and Huawei customizes it heavily.  These Huawei customizations likely explain the difference you see.