AltBeacon/android-beacon-library #729: Run detect Beacon when App closed for android 8

longpro1601, 2018-08-27:
### Expected behavior
On Android 8, when app is closed (killed app) will run service detect Beacon like on Android 6, 7

### Actual behavior
On Android 8, when app is closed detect Beacon not work

### Steps to reproduce this behavior

### Mobile device model and OS version
Android version 8

### Android Beacon Library version
AltBeacon 2.15

IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

davidgyoung, 2018-08-27:
Hi, @longpro1601.  This feature has been throroughly tested on Android 8 with the reference app, so it may be an issue with your implementation.

Can you please verify that you can reproduce this by running the reference app on your device with the following steps:

1. Download and build the reference app from here: https://github.com/altbeacon/android-beacon-library-reference
2. Modify the beacon layout to detect your beacon in the foreground, and verify in indicates it detects it.
3. Turn off your beacon and verify the app indicates it stopped detecting.
4.  Kill the app
5. Turn on your beacon
6. Wait up to 25 minutes and look for a launch of the app indicated by an notification

If you do not see the notification, please attach a logcat excerpt (not filtering a specific app) for steps 2-6.

longpro1601, 2018-08-27:
Hi @davidgyoung ,
Download and build the reference app from here: https://github.com/altbeacon/android-beacon-library-reference
 download branch Master, right?

longpro1601, 2018-08-28:
Hi @davidgyoung ,
I tried, when killed app, then Running immediate scan job each 10 second.
When detected Beacon then app auto open, but when killed app again then service not continue run (We are inside a beacon region.  We will not scan between cycles.) so when I turn off Beacon then service not run didExitRegion.

Logs:
08-28 10:35:24.220 20852-20852/? I/zygote64: Late-enabling -Xcheck:jni
08-28 10:35:24.503 20852-20852/org.altbeacon.beaconreference I/InstantRun: starting instant run server: is main process
08-28 10:35:24.520 20852-20852/org.altbeacon.beaconreference I/BeaconManager: BeaconManager started up on pid 20852 named 'org.altbeacon.beaconreference' for application package 'org.altbeacon.beaconreference'.  isMainProcess=true
08-28 10:35:24.533 20852-20852/org.altbeacon.beaconreference D/BeaconParser: Parsing beacon layout: m:2-3=beac,i:4-19,i:20-21,i:22-23,p:24-24,d:25-25
08-28 10:35:24.537 20852-20852/org.altbeacon.beaconreference D/BeaconParser: Parsing beacon layout: m:0-3=4c000215,i:4-19,i:20-21,i:22-23,p:24-24
08-28 10:35:24.540 20852-20852/org.altbeacon.beaconreference D/BeaconReferenceApp: setting up background monitoring for beacons and power saving loko
08-28 10:35:24.794 20852-20852/org.altbeacon.beaconreference I/ScanJob: Using immediateScanJobId from manifest: 208352939
08-28 10:35:24.800 20852-20852/org.altbeacon.beaconreference I/ScanJob: Using periodicScanJobId from manifest: 208352940
08-28 10:35:24.810 20852-20852/org.altbeacon.beaconreference W/JobInfo: Specified interval for 208352940 is +5m0s0ms. Clamped to +15m0s0ms
    Specified flex for 208352940 is 0. Clamped to +5m0s0ms
08-28 10:35:24.811 20852-20852/org.altbeacon.beaconreference W/JobInfo: Specified interval for 208352940 is +5m0s0ms. Clamped to +15m0s0ms
08-28 10:35:24.812 20852-20852/org.altbeacon.beaconreference W/JobInfo: Specified flex for 208352940 is 0. Clamped to +5m0s0ms
08-28 10:35:24.899 20852-20852/org.altbeacon.beaconreference I/ScanJob: Using immediateScanJobId from manifest: 208352939
08-28 10:35:24.904 20852-20852/org.altbeacon.beaconreference I/ScanJob: Using periodicScanJobId from manifest: 208352940
08-28 10:35:24.904 20852-20852/org.altbeacon.beaconreference W/JobInfo: Specified interval for 208352940 is +5m10s0ms. Clamped to +15m0s0ms
08-28 10:35:24.905 20852-20852/org.altbeacon.beaconreference W/JobInfo: Specified flex for 208352940 is 0. Clamped to +5m0s0ms
    Specified interval for 208352940 is +5m10s0ms. Clamped to +15m0s0ms
    Specified flex for 208352940 is 0. Clamped to +5m0s0ms
08-28 10:35:24.916 20852-20852/org.altbeacon.beaconreference E/CheckPermission: _bluetooth code = 1 
08-28 10:35:24.964 20852-20852/org.altbeacon.beaconreference I/ScanJob: Using immediateScanJobId from manifest: 208352939
08-28 10:35:24.972 20852-20852/org.altbeacon.beaconreference I/ScanJob: Using periodicScanJobId from manifest: 208352940
08-28 10:35:24.972 20852-20852/org.altbeacon.beaconreference W/JobInfo: Specified interval for 208352940 is +5m10s0ms. Clamped to +15m0s0ms
    Specified flex for 208352940 is 0. Clamped to +5m0s0ms
08-28 10:35:24.973 20852-20852/org.altbeacon.beaconreference W/JobInfo: Specified interval for 208352940 is +5m10s0ms. Clamped to +15m0s0ms
    Specified flex for 208352940 is 0. Clamped to +5m0s0ms
08-28 10:35:25.036 20852-20852/org.altbeacon.beaconreference I/CycledLeScanner: Using Android O scanner
08-28 10:35:25.041 20852-20852/org.altbeacon.beaconreference I/ScanJob: Using immediateScanJobId from manifest: 208352939
08-28 10:35:25.042 20852-20852/org.altbeacon.beaconreference I/ScanJob: Running immediate scan job: instance is org.altbeacon.beacon.service.ScanJob@24121a8
08-28 10:35:25.045 20852-20852/org.altbeacon.beaconreference I/ScanJob: scanJob version 2.15 is starting up on the main process
08-28 10:35:25.055 20852-20852/org.altbeacon.beaconreference W/ModelSpecificDistanceCalculator: Cannot find match for this device.  Using default
    Cannot find match for this device.  Using default
08-28 10:35:25.061 20852-20852/org.altbeacon.beaconreference D/BluetoothAdapter: isLeEnabled(): ON
08-28 10:35:25.061 20852-20852/org.altbeacon.beaconreference E/CheckPermission: _bluetooth code = 10 
08-28 10:35:25.076 20852-20869/org.altbeacon.beaconreference D/BluetoothAdapter: isLeEnabled(): ON
08-28 10:35:25.077 20852-20869/org.altbeacon.beaconreference E/CheckPermission: _bluetooth code = 10 
08-28 10:35:25.077 20852-20852/org.altbeacon.beaconreference I/ScanJob: Scan job running for 10000 millis
08-28 10:35:25.084 20852-20866/org.altbeacon.beaconreference D/BluetoothLeScanner: onScannerRegistered() - status=6 scannerId=-1 mScannerId=0
08-28 10:35:35.079 20852-20852/org.altbeacon.beaconreference I/ScanJob: Scan job runtime expired: org.altbeacon.beacon.service.ScanJob@24121a8
08-28 10:35:35.090 20852-20869/org.altbeacon.beaconreference D/BluetoothAdapter: isLeEnabled(): ON
08-28 10:35:35.090 20852-20869/org.altbeacon.beaconreference D/BluetoothLeScanner: could not find callback wrapper
08-28 10:35:35.146 20852-20852/org.altbeacon.beaconreference D/BluetoothAdapter: isLeEnabled(): ON
08-28 10:35:35.146 20852-20852/org.altbeacon.beaconreference E/CheckPermission: _bluetooth code = 10 
08-28 10:35:36.454 20852-20852/org.altbeacon.beaconreference I/ScanJob: Using immediateScanJobId from manifest: 208352939
08-28 10:35:36.460 20852-20852/org.altbeacon.beaconreference I/ScanJob: Using periodicScanJobId from manifest: 208352940
[... cut ...]
08-28 10:40:25.876 20852-20852/org.altbeacon.beaconreference I/ScanJob: onStopJob called for immediate scan org.altbeacon.beacon.service.ScanJob@7b80ac
08-28 10:40:25.879 20852-21122/org.altbeacon.beaconreference D/BluetoothAdapter: isLeEnabled(): ON
08-28 10:40:25.882 20852-20852/org.altbeacon.beaconreference I/ScanJob: We are inside a beacon region.  We will not scan between cycles.
08-28 10:40:25.883 20852-21122/org.altbeacon.beaconreference D/BluetoothAdapter: isLeEnabled(): ON
08-28 10:40:25.883 20852-21122/org.altbeacon.beaconreference D/BluetoothLeScanner: could not find callback wrapper

davidgyoung, 2018-08-28:
if you kill the app while it is in region, the next BLE scan on Android 8 will take place in 15-25 minutes on the job scheduler.  If the beacon is not seen at that time, you will get a region exit.  if it is still seen, you will get no entry/exit callback.  This is the designed behaviour.

How is this different from what you expect?

longpro1601, 2018-08-28:
Hi @davidgyoung 
Thanks you supported.

I tried kill app  while it is in region, and I wait 15-25 minutes but can not seen job scheduler run.

davidgyoung, 2018-08-28:
@longpro1601, how do you know that the job scheduler did not run the ScanJob? Understand that If you are already in the region and the beacon is still detected, your app won't get an additional callback to didEnterRegion or didExitRegion by design of the library.  

If you want to test to see if it is running the ScanJob, you'd have to monitor LogCat for at least 25 minutes after killing the app and look for any log line with the name ScanJob.  You would have to remove all other filters on specific apps or process ids, because when the new ScanJob starts it will have a new process id.  It will look like this:

```
08-28 10:35:25.045 20852-20852/org.altbeacon.beaconreference I/ScanJob: scanJob version 2.15 is starting up on the main process
```

longpro1601, 2018-08-29:
@davidgyoung ,

```
08-28 10:40:12.299 20852-20852/org.altbeacon.beaconreference I/ScanJob: Scan job running for 10000 millis
08-28 10:40:14.823 20852-20852/org.altbeacon.beaconreference D/BeaconReferenceApp: did enter region.
auto launching MainActivity
08-28 10:40:25.882 20852-20852/org.altbeacon.beaconreference I/ScanJob: We are inside a beacon region. We will not scan between cycles.
```

After (08-28 10:41:00) I has turn off Beacon and wait to 30 minutes but can not see Logcat about ScanJob.

davidgyoung, 2018-08-29:
Huh.  Can you tell me what device model this is?  I wonder if this is a custom ROM that disables scheduled jobs after a task switcher kill.  Stock Android from AOSP does not do that.

longpro1601, 2018-08-29:
@davidgyoung 
I has test on device Oppo A3S, Android 8.1 and not run scheduled jobs.

I am testing Samsung S8, Android 8.0. I will report later.

Thanks.

davidgyoung, 2018-08-29:
You can also use this app: https://github.com/davidgyoung/stopped-app-detector

on the Oppo A3S to see if swiping your app off the screen puts it into the stopped app state.  If so, this would indicate the ROM has been customized in a manner that a swipe kill puts it in a stopped app state.  (Other customisations are possible, too.)

davidgyoung, 2018-09-04:
@longpro1601, do you still see this issue on the Oppo A3S?  If you have more info please share so others who encounter this issue might benefit.

longpro1601, 2018-09-04:
@davidgyoung 
I used app Stopped-app-detector,
when kill app (android-beacon-library-reference) on Oppo A3S then: stopped app state.

davidgyoung, 2018-10-03:
OK, understood.  So for the record, the Oppo A3S has a customized Android OS.  Unlike vanilla Android, if you swipe an app off the screen, it can no longer run in the background until you manually launch it again.  This includes for beacon detections.

closed 2018-10-03 by davidgyoung

JamieH-Cubik, 2020-08-25:
> OK, understood. So for the record, the Oppo A3S has a customized Android OS. Unlike vanilla Android, if you swipe an app off the screen, it can no longer run in the background until you manually launch it again. This includes for beacon detections.

I am having some strange behaviour working with an oppo a5 and beacons, incredibly tempermental. 
Does this issue apply to foreground services? I've got location working when the app is in the background through a foreground service but I haven't gotten reliable enough behaviour from the beacons to even check what they do when the app is in the background. Was wondering if the fact that it's in this service is playing a part in my headaches