AltBeacon/android-beacon-library #731: AltBeacon Library is writing on the UI Thread StrictMode

shawnvega, 2018-08-30:
### Expected behavior
Altbeacon Should not Write on the UI Thread 

### Actual behavior
StrictMode crashes the app

### Steps to reproduce this behavior
1. In Application OnCreate use the following to enable StrictMode 
StrictMode.setThreadPolicy(new StrictMode.ThreadPolicy.Builder()
                    .detectDiskWrites()
                    .detectNetwork()
                    .penaltyDeath()
                    .build());
2. Start scanning for beacons
3. Observe Crash 

### Mobile device model and OS version
Google Pixel 2 with Android 9.0
### Android Beacon Library version
AltBeacon Library version 2.15

### Stacktrace below
    StrictMode policy violation; ~duration=10 ms: android.os.strictmode.DiskWriteViolation
        at android.os.StrictMode$AndroidBlockGuardPolicy.onWriteToDisk(StrictMode.java:1460)
        at libcore.io.BlockGuardOs.write(BlockGuardOs.java:347)
        at libcore.io.IoBridge.write(IoBridge.java:526)
        at java.io.FileOutputStream.write(FileOutputStream.java:381)
        at java.io.ObjectOutputStream$BlockDataOutputStream.drain(ObjectOutputStream.java:1973)
        at java.io.ObjectOutputStream$BlockDataOutputStream.setBlockDataMode(ObjectOutputStream.java:1855)
        at java.io.ObjectOutputStream.writeObject0(ObjectOutputStream.java:1131)
        at java.io.ObjectOutputStream.writeObject(ObjectOutputStream.java:354)
        at java.util.HashSet.writeObject(HashSet.java:287)
        at java.lang.reflect.Method.invoke(Native Method)
        at java.io.ObjectStreamClass.invokeWriteObject(ObjectStreamClass.java:1037)
        at java.io.ObjectOutputStream.writeSerialData(ObjectOutputStream.java:1552)
        at java.io.ObjectOutputStream.writeOrdinaryObject(ObjectOutputStream.java:1488)
        at java.io.ObjectOutputStream.writeObject0(ObjectOutputStream.java:1234)
        at java.io.ObjectOutputStream.defaultWriteFields(ObjectOutputStream.java:1604)
        at java.io.ObjectOutputStream.writeSerialData(ObjectOutputStream.java:1565)
        at java.io.ObjectOutputStream.writeOrdinaryObject(ObjectOutputStream.java:1488)
        at java.io.ObjectOutputStream.writeObject0(ObjectOutputStream.java:1234)
        at java.io.ObjectOutputStream.writeObject(ObjectOutputStream.java:354)
        at org.altbeacon.beacon.service.ScanState.save(ScanState.java:195)
        at org.altbeacon.beacon.service.ScanState.applyChanges(ScanState.java:291)
        at org.altbeacon.beacon.service.ScanJobScheduler.applySettingsToScheduledJob(ScanJobScheduler.java:83)
        at org.altbeacon.beacon.service.ScanJobScheduler.applySettingsToScheduledJob(ScanJobScheduler.java:92)
        at org.altbeacon.beacon.BeaconManager.applyChangesToServices(BeaconManager.java:983)
        at org.altbeacon.beacon.BeaconManager.updateScanPeriods(BeaconManager.java:977)
        at org.altbeacon.beacon.BeaconManager.setBackgroundMode(BeaconManager.java:542)
        at org.altbeacon.beacon.powersave.BackgroundPowerSaver.onActivityResumed(BackgroundPowerSaver.java:66)
        at android.app.Application.dispatchActivityResumed(Application.java:239)
        at android.app.Activity.onResume(Activity.java:1343)
        at android.support.v4.app.FragmentActivity.onResume(FragmentActivity.java:485)
        at android.app.Instrumentation.callActivityOnResume(Instrumentation.java:1412)
        at android.app.Activity.performResume(Activity.java:7292)
        at android.app.ActivityThread.performResumeActivity(ActivityThread.java:3776)
        at android.app.ActivityThread.handleResumeActivity(ActivityThread.java:3816)
        at android.app.servertransaction.ResumeActivityItem.execute(ResumeActivityItem.java:51)
        at android.app.servertransaction.TransactionExecutor.executeLifecycleState(TransactionExecutor.java:145)
        at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:70)
        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1808)
        at android.os.Handler.dispatchMessage(Handler.java:106)
        at android.os.Looper.loop(Looper.java:193)
        at android.app.ActivityThread.main(ActivityThread.java:6669)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:493)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:858)

davidgyoung, 2018-08-31:
The library does write to storage (internal storage, generally) to save its state whenever the app moves from the foreground (as in this case), or whenever the beacon scanning settings are changed by the app programmatically.  And, yes, there is no protection to prevent his from being done on the UI thread.

However, given the small size of the state file that is written, I doubt this makes any practical difference.  I'm open to a change to address this if there is any evidence it causes problems.

agap, 2019-01-31:
It actually does cause the issues, I'll file a ticket soon.