AltBeacon/android-beacon-library #734: Start MonitoringActivity service when restart device Android 8

longpro1601, 2018-09-04:
Hi @davidgyoung,

I used  BootstrapNotifier,
beaconManager.setEnableScheduledScanJobs(false);
when device Samsung S8 (andrioid 8)  restarted, then has error: "Not allowed to start service Intent"

I tried edit function on RegionBootstrap.java as below:

```
public boolean bindService(Intent intent, ServiceConnection conn, int arg2) {
            this.serviceIntent = intent;
            if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.O) {
                RegionBootstrap.this.application.getApplicationContext().startForegroundService(intent);
            } else {
                RegionBootstrap.this.application.getApplicationContext().startService(intent);
            }
            return RegionBootstrap.this.application.getApplicationContext().bindService(intent, conn, arg2);
        }
```

After edit code and build library, I has test on Samsung S8 (andrioid 8).
Tested: when device restarted, after 1 minute,  service detect beacons ran success.

davidgyoung, 2018-09-05:
You are correct that this won't work on Android 8.  Android 8 blocks starting a background service when the app is not in the foreground.    But there is an existing way to use a foreground service with the library without modifying source code.

Two options:

1. Leave the default  of `beaconManager.setEnableScheduledScanJobs(true);` and let it use scheduled jobs instead of background services.

2. Configure the library to use a foreground service as described [here](https://altbeacon.github.io/android-beacon-library/foreground-service.html)

If you choose option 2, you need to add that configuration code before you set up RegionBootstrap so that when it tries to bind in the background it will automatically make it a background service per the configuration from step 2.

longpro1601, 2018-09-05:
@davidgyoung 
I tried option 2, but when device restart then has error "Not allowed to start service Intent"

davidgyoung, 2018-09-05:
@longpro1601, I'm not sure why you see this -- I don't see it on the Galaxy S9 with 8.0.  The regular service that gets started with `application.getApplicationContext().startService(intent);` is supposed to be converted to a foreground service by this code from BeaconService.java:

```
    private void startForegroundIfConfigured() {
        BeaconManager beaconManager = BeaconManager.getInstanceForApplication(
                this.getApplicationContext());
        Notification notification = beaconManager
                .getForegroundServiceNotification();
        int notificationId = beaconManager
                .getForegroundServiceNotificationId();
        if (notification != null &&
                Build.VERSION.SDK_INT >= Build.VERSION_CODES.JELLY_BEAN) {
            this.startForeground(notificationId, notification);
        }
    }
```

My understanding is that this is allowed on Android 8+,  and my tests show it to work.

I wonder if that code is not executing for your app for some reason.  Can you please see if 

`Notification notification = beaconManager.getForegroundServiceNotification();`

is returning a non-null value right before you set up RegionBootstrap?

davidgyoung, 2018-10-17:
This was reported again in #744 and fixed in #746, and released in 2.15.2.

closed 2018-10-17 by davidgyoung