AltBeacon/android-beacon-library #735: didRangeBeaconsInRegion() called multiple times in foreground mode

adrinieto, 2018-09-04:
### Expected behavior

Using the foreground mode, `didRangeBeaconsInRegion()` should be called once for every scan action.

### Actual behavior

The first time I call `startRangingBeaconsInRegion()` it works as expected. If I stop and start again, the listener is called twice. In version 2.15 it works.

### Steps to reproduce this behavior

We are using the Foreground Service Scanning.

The first time I call `startRangingBeaconsInRegion()` it works as expected. 

Then, if a call `stopRangingBeaconsInRegion()` and `startRangingBeaconsInRegion()`, the method `didRangeBeaconsInRegion()` is called twice for every scan. I'm calling `addRangeNotifier()` every time before  `startRangingBeaconsInRegion()`, but this is working with previous versions of the library.

If a do this a third time, the listener is called three times.

### Mobile device model and OS version

Google Pixel XL, Android 9

### Android Beacon Library version

2.15.1

davidgyoung, 2018-09-05:
@adrinieto, I've tried reproducing this by making the following change to the reference app's RangingActivity.java to that it calls addRangeNotifier twice and calls startRangingBeaconsInRegion twice.  Still, I don't see two callbacks.  (See logcat).  Can you please give me equivalent steps to reproduce on the [reference app](https://github.com/AltBeacon/android-beacon-library-reference)?

```
    @Override
    public void onBeaconServiceConnect() {

        RangeNotifier rangeNotifier = new RangeNotifier() {
           @Override
           public void didRangeBeaconsInRegion(Collection<Beacon> beacons, Region region) {
              if (beacons.size() > 0) {
                  Log.d(TAG, "didRangeBeaconsInRegion called with beacon count:  "+beacons.size());
                  Beacon firstBeacon = beacons.iterator().next();
                  logToDisplay("The first beacon " + firstBeacon.toString() + " is about " + firstBeacon.getDistance() + " meters away.");
              }
           }

        };
        try {
            beaconManager.startRangingBeaconsInRegion(new Region("myRangingUniqueId", null, null, null));
            beaconManager.addRangeNotifier(rangeNotifier);
            beaconManager.startRangingBeaconsInRegion(new Region("myRangingUniqueId", null, null, null));
            beaconManager.addRangeNotifier(rangeNotifier);
        } catch (RemoteException e) {   }
    }
```

Logcat shows:

```
09-04 21:24:10.924 31076-31076/org.altbeacon.beaconreference D/RangingActivity: didRangeBeaconsInRegion called with beacon count:  1
09-04 21:24:12.029 31076-31076/org.altbeacon.beaconreference D/RangingActivity: didRangeBeaconsInRegion called with beacon count:  1
09-04 21:24:13.133 31076-31076/org.altbeacon.beaconreference D/RangingActivity: didRangeBeaconsInRegion called with beacon count:  1
09-04 21:24:14.237 31076-31076/org.altbeacon.beaconreference D/RangingActivity: didRangeBeaconsInRegion called with beacon count:  1
09-04 21:24:15.342 31076-31076/org.altbeacon.beaconreference D/RangingActivity: didRangeBeaconsInRegion called with beacon count:  1
09-04 21:24:16.446 31076-31076/org.altbeacon.beaconreference D/RangingActivity: didRangeBeaconsInRegion called with beacon count:  1
09-04 21:24:17.552 31076-31076/org.altbeacon.beaconreference D/RangingActivity: didRangeBeaconsInRegion called with beacon count:  1
09-04 21:24:18.656 31076-31076/org.altbeacon.beaconreference D/RangingActivity: didRangeBeaconsInRegion called with beacon count:  1
09-04 21:24:19.760 31076-31076/org.altbeacon.beaconreference D/RangingActivity: didRangeBeaconsInRegion called with beacon count:  1
```

adrinieto, 2018-09-05:
I've tried to reproduce our problem in the reference app, but I couldn't.

With some experimentation in our app and with the library I think I've found the root of the problem. The BroadcastReceiver of `BeaconLocalBroadcastProcessor` is been leaked, unregister method is not called when service is destroyed. The problem was added in #720.

I fixed it calling `mBeaconNotificationProcessor.unregister()` in the `onDestroy()` method of `BeaconService`. I could submit a PR to fix that.

With further investigation i think there are more leaks, if I start and stop the scanning, the BeaconService is still in memory.

Here is a screenshot of the heap dump of our app starting and stopping the scanning 5 times. There are 5 instances of BeaconService and 5 receivers.

[image]

I think we are doing something wrong or there is something wrong in the library, there shouldn't be 5 instances of BeaconService running if I start and stop the scanning.

davidgyoung, 2018-09-05:
I would welcome a PR, yes.  For the service leaks I'd love steps to reproduce with the reference app.  Logging could be checked to see if the service has been commanded to stop but did not for some reason.

T-Spoon, 2018-10-03:
[image]

Also seeing the same issue here - `BeaconLocalBroadcastProcessor` is not being deregistered if the service is unbound.

To reproduce, you can call `beaconManager.bind(this)` in `Activity.onStart()`, and `beaconManager.unbind(this)` in `Activity.onStop()`

davidgyoung, 2018-10-03:
Thanks, @T-Spoon.  It looks like we just need to deregister when the service stops.  I will prepare a PR.

davidgyoung, 2018-10-03:
@T-Spoon, can you please take a look at #745 and see if the change looks correct to you?

closed 2018-10-03 by davidgyoung

adrinieto, 2018-10-04:
Thank you, that should fix my issue.
I think the fix is also needed in the `ScanJobScheduler` class. I can prepare a PR.

PR #748 (2018-10-08): Fix leaked NotificationProcessor in ScanJobsScheduler
This caused multiple `NotificationProcessor` registrations when binding and unbinding.

This PR and #745 will completely fix #735