AltBeacon/android-beacon-library #736: didExitRegion() is called after 30 minutes and after then the library fail to detect beacon

amazinglu, 2018-09-04:
### Expected behavior
the beacon detect service will run continually

### Actual behavior
After scanning for about 30 minutes, the didExitRegion() is called and the beacon can not be detected anymore. 
I try to turn off the bluetooth and turn it on again. And the beacon is detected for a minute. But after that the didExitRegion() is called again and the beacon is gone. The same loop goes over and over again.

The problem happens on the galaxy s9 (Android 8.0.0, API 26)
The Galaxy s6 (Android 7.0, API 24) so not have such issue

### Steps to reproduce this behavior
Use Android studio to run the app, and let it run for 30 minutes, and the problem occurs

### Mobile device model and OS version
galaxy s9 (Android 8.0.0, API 26)

### Android Beacon Library version
android-beacon-library-2.15.1-beta1.aar

IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

davidgyoung, 2018-09-04:
Hi @amazinglu can you please test this with the [reference app](https://github.com/AltBeacon/android-beacon-library-reference) and let me know if you can reproduce the same issue?  

I have not seen this with the  galaxy s9 in my testing, so it may be an application-specific problem.

amazinglu, 2018-09-04:
Can I know how to use your reference app, input the serial number?

davidgyoung, 2018-09-04:
To run the reference app:

1. Do a Git clone of the repository at https://github.com/AltBeacon/android-beacon-library-reference.git
2. Open the project in Android Studio, and run on your device.

davidgyoung, 2018-09-04:
> I knew that. I have the app running on the phone. But how can I use the
> reference app to detect beacon.
> I press start range and no beacon appears.

Oh, the reference app is not set up with any BeaconParsers other than the
one for AltBeacon.  If you are using iBeacon, for example, add the iBeacon
parser  to to BeaconReferenceApplication.java in the onCreate method.

amazinglu, 2018-09-04:
Hi, I did not test with the reference app yet. But I find out that the problem is only appear in Galaxy s9, I use my OnePlus 5 (Android 8.1.0, API 27) to test, no such issue appears

amazinglu, 2018-09-04:
I have tested my galaxy s9 with the reference app, within 30 minutes, I can get consistent detection. But after 30 minutes, sometime the app can detected the beacon, sometimes It cannot.

davidgyoung, 2018-09-04:
Can you please:

1. turn on debug logging with `beaconManager.setDebug(true);`

2. do a log capture showing at least 60 seconds when range results are delivered as you expect (during the first 30 minutes)

3. do a log capture showing at least 60 seconds when range results are not consistent.  Please try to have the time include at least one detection since you say it is intermittent.

amazinglu, 2018-09-05:
> Can you please:
>
> turn on debug logging with beaconManager.setDebug(true);
>
> do a log capture showing at least 60 seconds when range results are delivered as you expect (during the first 30 minutes)
>
> do a log capture showing at least 60 seconds when range results are not consistent. Please try to have the time include at least one detection since you say it is intermittent.

So you want me to capture the log record of my own app or the reference app?

davidgyoung, 2018-09-05:
If you can reproduce with the reference app, please capture from there.

amazinglu, 2018-09-05:
At the moment, seem like I cannot reproduce the issues. And after I added the bluetooth Medic into my app, the problem seems not appears any more. I will follow up if I can reproduce and capture the bug

amazinglu, 2018-09-10:
the problem appears again and I got this log msg: "We are newly outside the region because the lastSeenTime of 941968963 was 10097 seconds ago, and that is over the expiration duration of 10000" every time the didExitRegion() is called. What is this msg means?

davidgyoung, 2018-09-10:
That simply means that the beacon hasn't been seen in 10097 milliseconds (the log text incorrectly says seconds, it's actually milliseconds), and since the beacon hasn't been seen in a long time, which is why didExitRegion is firing.  This is normal.

You should check the logs to see if there are any operating system messages in the 10097 milliseconds prior to this message indicating any issues with bluetooth scanning.

amazinglu, 2018-09-13:
Is the library allow me to change the expiration duration that mention above?

davidgyoung, 2018-09-13:
Yes!  Just call `BeaconManager.setRegionExitPeriod(20000);` to change it to 20 seconds from 10 seconds.

amazinglu, 2018-09-17:
Hi, I just find out something. I am getting this msg when all the beacons are out of range.
"2018-09-17 16:04:40.467 22787-22787/com.parabit.parabeacon.app.demo.debugShow W/CycledLeScanner: Allowing a long running scan to be stopped by the OS.  To prevent this, set longScanForcingEnabled in the AndroidBeaconLibrary."

How can I set longScanForcingEnabled?

amazinglu, 2018-09-17:
I think is something to do with this code:
[image]

amazinglu, 2018-09-17:
Yes, for Android 8.0, I start a foreground service and leaving the app in the foreground for 30+ minutes and it is the requirement of the app.

amazinglu, 2018-09-17:
My scanning set up for Android 8.0:

private void setupOreoScanning() {
        /**
         * se up a notification for forground scanning (foreground service)
         * */
        NotificationManager mNotificationManager =
                (NotificationManager) getSystemService(Context.NOTIFICATION_SERVICE);
        NotificationChannel channel = new NotificationChannel(NOTIFICATION_CHANNEL_ID_SCANNING, "Beacons",
                NotificationManager.IMPORTANCE_DEFAULT);
        channel.setShowBadge(true);
        channel.setSound(null, null);
        mNotificationManager.createNotificationChannel(channel);
        Notification.Builder builder = new Notification.Builder(this,
                NOTIFICATION_CHANNEL_ID_SCANNING);

        int drawableResourceId = getResource("com.parabit.mmrbtsdk.trayIcon");

        builder.setSmallIcon(drawableResourceId);
        builder.setContentTitle("Scanning for Beacons");
        Intent intent = getLaunchIntent();
        PendingIntent pendingIntent = PendingIntent.getActivity(
                this, 0, intent, PendingIntent.FLAG_UPDATE_CURRENT
        );
        builder.setContentIntent(pendingIntent);
        // enable foreground scanning
        mBeaconManager.enableForegroundServiceScanning(builder.build(), 456);
        mBeaconManager.setEnableScheduledScanJobs(false);
        // Setting this value to false will disable ScanJobs when the app is run on Android 8+, \
        // which can prohibit delivery of callbacks when the app is in the background unless
        // the scanning process is running in a foreground service. This method may only be called if bind()
        // has not yet been called, otherwise an `IllegalStateException` is thown.

        mBeaconManager.setBackgroundScanPeriod(1125l);
        mBeaconManager.setForegroundScanPeriod(1125l);
        mBeaconManager.setBackgroundBetweenScanPeriod(0);
        mBeaconManager.setForegroundBetweenScanPeriod(0);

    }

amazinglu, 2018-09-17:
And my scanning setup for Android 7.0+:

 private void setupNormalScanning() {
        mBeaconManager.setAndroidLScanningDisabled(false);
        mBeaconManager.setForegroundScanPeriod(2000l);
        mBeaconManager.setBackgroundScanPeriod(2000l);
        mBeaconManager.setForegroundBetweenScanPeriod(0);
        mBeaconManager.setBackgroundBetweenScanPeriod(0);
        mBeaconManager.setRegionStatePersistenceEnabled(false);
    }

davidgyoung, 2018-09-17:
Ah, I see, so you have completely different configurations on Android 8 and 7 not mentioned in the issue.  It would have been really helpful to mention that in the first place!

I would think you'd have the same problem on Android 7 as on Android 8 -- I can't explain why you don't. But regardless, for you to not be blocked after 30 minutes, you'll need to configure long scanning as described in #529 if you really want to circumvent the operating system's limits.

One word of caution:  the operating system has these limits for a reason -- constant scanning for long periods of time will drain the battery quickly.  Make sure your users know this is happening and accept that the benefit your app provides is worth the battery cost, or they are likely to be unhappy and uninstall.

closed 2018-09-17 by davidgyoung

dinakarms, 2018-11-22:
hi David,
I am using the reference app for the beacons,as the entry region is detecting immediately.
But for the exit region detection time is 10 Sec.
So,Now I want to reduce the exit region detection time.
Could you please help me on this.
Thankyou in advance.

davidgyoung, 2018-11-22:
@dinakarms, you really cannot have an "immediate" exit region.  Understand that the whole concept of exiting a region is amounts to not seeing the beacon in a certain time frame.  You can reduce the intervals from 10 seconds, but it will increase false exits.