AltBeacon/android-beacon-library #738: Problem with detecting beacons(advertising interval 5000ms) in foreground service when device turned off.

dzmitrymikheyeu, 2018-09-05:
### Expected behavior
All beacons with advertising interval 5000ms should be found in foreground service during device turn off.
### Actual behavior
After turn off device with enabled foreground service beacons(5000ms) disappear from range. When turn on device beacons are appearing in range again. So, when device enabled(screen on) all beacons was found. 
### Steps to reproduce this behavior
Run foreground service -> run beacons range notifier in this service -> turn off device -> waiting 1-2 min -> beacons with advertising interval 5000ms will not be found until device won't turned on.
### Mobile device model and OS version
Samsung, android version 7.0
### Android Beacon Library version
android-beacon-library:2.15
IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

davidgyoung, 2018-09-06:
Does the app use the BackgroundPowerSaver class?  Does it customize `beaconManager.setForegroundScanPeriod(...)`, beaconManager.setBackgroundScanPeriod(...)` or the equivalent betweenScanPeriod settings?

dzmitrymikheyeu, 2018-09-06:
Yes, tried to disable power saver, but result the same. I playing around with scan periods for background and foreground - nothing to happens. By default set 10000ms for background and foreground, between scans interval 0ms. Seems only foreground scan period should be applied if i use foreground service.

dzmitrymikheyeu, 2018-09-06:
In addition, others devices/beacons are visible while device turned off. It's a very weird.

davidgyoung, 2018-09-06:
OK, so to clarify, only the beacon that advertises every 5000ms is not detected, but others are? What is different about the other devices? What are their advertising rates?

davidgyoung, 2018-09-06:
You might also try setting the foregroundBetweenScanPeriod to 1.  Setting it to a non-zero value causes the scan to stop briefly and restart every 10 seconds which might reset detections.

dzmitrymikheyeu, 2018-09-25:
There are some updates with this issue: it's reproduced only on devices with android 7 with any beacons(tested 100ms and 5000ms) in foreground service when device turned off. Android 8,9 worked perfect with the same beacons.

davidgyoung, 2018-09-25:
It sounds like specific devices may be blocking scans with the screen off.  Huawei devices, for example, have a customized Android OS that is known to do this is some cases.  Can you please confirm the device models where you see this behaviour?

dzmitrymikheyeu, 2018-09-25:
Issue appears on Samsung SM-G390F and Samsung SM-G920F with android version 7.0.

dzmitrymikheyeu, 2018-09-25:
Also i notice that range notifier doesn't work on my own foreground service when screen off if beaconManager does not configure enableForegroundServiceScanning().. So, Is there the way to use my own foreground service without library foreground service? In previous library version it's works until your added your own foreground solution. Maybe this is the reason of current issue? I tried to extend from org.altbeacon.beacon.service.BeaconService and move logic to this service, but i think it is not applicable for my own case.

davidgyoung, 2018-09-25:
I don't see any reason this should be the case on those Samsung devices .  You should be able to range normally on Android 7 for up to 30 minutes in the background (for scans beyond 30 Mon see here https://github.com/AltBeacon/android-beacon-library/issues/526)

If you are not seeing this work, please set `beaconManager.setDebug(true)` then capture a log excerpt from 30 seconds before scanning stops until 30 seconds after.

android-bash, 2019-01-03:
@dzmitrymikheyeu Did you find solution for this?