AltBeacon/android-beacon-library #739: didEnterRegion() not firing in background

MikeOrtiz, 2018-09-07:
### Expected behavior

didEnterRegion() callback should fire when the reference app is in the background.

### Actual behavior

didEnterRegion() doesn't fire in the background. It doesn't fire until the app is put in the foreground.

### Steps to reproduce this behavior

Open app. Turn on beacon. Observe didEnterRegion() is logged. Turn off beacon. Turn on again. It's logged again. Turn off beacon. Background app. Turn on beacon. Nothing is logged. Foreground app. didEnterRegion() is logged.

### Mobile device model and OS version
Nexus 5X Android 6.0.1

### Android Beacon Library version

2.15.1. Using latest Beacon Library Reference app.

IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

davidgyoung, 2018-09-07:
In these steps:

1. Open app.
2. Turn on beacon. 
3. Observe didEnterRegion() is logged. 
4. Turn off beacon. 
5. Turn on again.
6.  It's logged again.
7.  Turn off beacon.
8.  Background app. 
9. Turn on beacon. 
10. Nothing is logged. 
11. Foreground app. 
12. didEnterRegion() is logged.

The above may be normal unless you wait at least 30 seconds with default settings between steps 7 and 8 and confirm you received a didExitRegion event.

It is critical that a didExitRegion  is called before you expect to get the next didEnterRegion callback.  Once the app is put to the background, it scans slow down to save battery.  If the library still believed the beacon region to be in an "inside" state (the beacon was seen within the last 30 seconds) it will remain in the inside state until the next scan happens and it has not been seen.  And that next scan might not happen for a number of minutes in the background.

In order to save battery in the background the library won't do another scan to confirm a recently visible  beacon is still visible until the backgroundBetweenScanPeriod passes (5 minutes by default on Android 4-7, 15 minutes on Android 8).  No new didEnterRegion events will fire for the same region until the didExitRegion fires.  If you put the app to the background before this happens, it will remain in region.  If when the next scan cycle starts it is not detected, a didExitRegion will fire.  If it is visible at that time, no events will fire (the library will have completely missed that the beacon was briefly not transmitting.)

If, on the other hand, you verify you received a didExitRegion before the app goes to the background, the app will do a constant low power filtered hardware scan to look for reappearance of the beacon -- even in the background.   On most devices including the Nexus 5X, this will cause a detection within a few seconds if you turn the beacon on even if the app is in the background.  

Again the key difference is whether the app has received a didExitRegion event before it is put to the background.

MikeOrtiz, 2018-09-10:
> If, on the other hand, you verify you received a didExitRegion before the app goes to the background, the app will do a constant low power filtered hardware scan to look for reappearance of the [...]

I added logging in didExitRegion to confirm it fired before backgrounding the app. I turned on the beacon soon after backgrounding the app and didEnterRegion did not fire. I then kept the beacon turned on, and didEnterRegion did fire after 14 minutes.

My use case is that a BLE beacon can turn on for about a minute at a time (very sporadically, let's say a couple times a day), and while the beacon is on, the phone needs to scan it and initiate a data download. Does your library work with this use case?

davidgyoung, 2018-09-10:
Yes, this use case is supported.  If you see a didExitRegion before going to the background and it takes 15 minutes to detect, this means that hardware scan filters are not working for some reason.

* Do you have other beacons advertising when you do this test (even if they are not part of the region definition?)

* If not, can you please set `beaconManager.setDebug(true);` and attach a log excerpt that covers the time when the app is put to the background during this test?

MikeOrtiz, 2018-09-10:
I didn't set up any other beacons, but I noticed when I set debugging to true, that there were still a lot of BLE devices being detected (even when I tried to find an isolated space). Here are the [BLE logs](https://gist.github.com/MikeOrtiz/448668c99fa254e019133d6546281793). Here are my repro steps in this log:

1. Brought app to foreground.
2. Turned on BLE device and noted "did enter region" logged.
3. Turned off BLE device and noted "did exit region" logged after 15-30 seconds.
4. Backgrounded app.
5. Turned on BLE device. No logging for "did enter region".
6. Eventually, foregrounded app and "did enter region" was logged.
7. Saved logs.

davidgyoung, 2018-09-10:
Ah, I see the issue.  This is clearly a custom beacon format based on manufacturer data.  I see the manufacturer data on a good beacon packet looks like this:

`0201040302081807ff67010a00e0011409436f6e746f7572373830324850343033323433`

It appears that this BLE device has a manufacturer code of 0x0167, which is also the beacon type code you are using to match it with your custom beacon format.   That's all OK.

But here's the issue.  For Android's fast background BLE advertisement filters to work, you have to tell them the manufacturer code you are using.  Since this library is commonly used with AltBeacon (RadiusNetworks = 0x0118) and iBeacon (Apple =0x004C)  those manufacturer codes are pre-configured with the library.  Since you are using 0x0167, you simply need to add this to your BeaconParser to get it to work:

```
beaconParser.setHardwareAssistManufacturerCodes(new int[] {0x167});
```

While this method is in the Java Doc, this isn't a common use case, so it's not in the main samples and is easy to overlook this.  I think this is actually the second time anybody has ever asked me about this in about five years. 

Give it a shot, I think you'll see that once you add it you'll get fast background detections.

MikeOrtiz, 2018-09-11:
That did the trick! Unfortunately, I think I've run into another issue with a different beacon. I attempt to match Manufacturer ID 0x170. However, the beacon parser processes pdu type 0x16 rather than the custom 0xFF, which follows it. Thus, in order to match, I need to shift the "m" in the beaconLayout from 0-1 to 4-5. This workaround makes it so that the beacon will be recognized while the app is in the foreground. However, I can't seem to match the HardwareAssistManufacturerCodes while the app is backgrounded. [Here are the logs showing the matching in the foreground.](https://gist.github.com/MikeOrtiz/2a02386d3a0072721b9ec06bf1736ace)

davidgyoung, 2018-09-11:
That is a unusual advertisement packet for sure.  I don't think I've ever come across a device that sends out both a 0x16 PDU (service advertisement) and a 0xFF PDU (manufacturer advertisement) in the same packet.

```
len pdu bytes
02  01  05
03  02  0818
0f  09  6d657465722b3031343934343733
03  16  0818
09  ff  7001d521000100010319000400
```

Yes, the library's BeaconParser walks through the PDUs from the first to the last looking for one that is either a 0x16 or 0xFF, because that is what the popular beacon formats use in their advertisements.  For efficiency sake, it bails out with the first one it finds, rather than trying to match each one.  You can see that code here:

https://github.com/AltBeacon/android-beacon-library/blob/master/src/main/java/org/altbeacon/beacon/BeaconParser.java#L429

So that's why you have to match on the first one in your layout.  But I don't see why that should prevent this from working.  Seems like you could use something like this, which may be what you have: "s:0-1=1808,m:4-5=7001,i:6-16"

Because that layout is for a service advertisement, it won't use the HardwareAssistManufacturerCodes  for fast background matching -- they are unneeded.  The filter will use the  2-byte service UUID for matching instead.  You should probably check the logs to see what your filter looks like.  The log line for your manufacturer filter in the other case looks like this:

```
09-13 13:38:29.095 29924-29924/org.altbeacon.beaconreference D/ScanFilterUtils: Set up a scan filter: BluetoothLeScanFilter [mDeviceName=null, mDeviceAddress=null, mUuid=null, mUuidMask=null, mServiceDataUuid=null, mServiceData=null, mServiceDataMask=null, mManufacturerId=76, mManufacturerData=[], mManufacturerDataMask=[]]
```

The above indicates that it is filtering on just mManufacturerId=76 (0x004c == Apple, Inc.)  You can check what your new filter is based on the service layout to see anything might be wrong with it.

I'm not 100% sure how the hardware filter will match this funny packet were the PDU to be matched is the 4th one in the packet, but maybe it'll work.  I know that more typical beacon packets at least have a flags PDU up front, and they match fine.  Depending on your layout, it might try to match some bytes in the mServiceData field, which could be problematic if they are beyond the end of the PDU.

MikeOrtiz, 2018-09-13:
> Seems like you could use something like this, which may be what you have: "s:0-1=1808,m:4-5=7001,i:6-16"

I actually wasn't matching with **s**. Adding it did the trick! 

The setBeaconLayout has the following comment: _This is the preferred method and matching BeaconLayouts by serviceUUID will be deprecated in the future_. I didn't use **s**, because of that comment. I think I might've misinterpreted the comment, but didn't see another method matching by serviceUUID.

closed 2018-09-13 by MikeOrtiz

davidgyoung, 2018-09-13:
That deprecation note doesn't apply to your use case.  It has to do with using the service UUID to automatically combine multiple beacon frames together (designed for Eddystone-TLM).    You're not doing that, so don't worry.