AltBeacon/android-beacon-library #740: Update "Device Support For Beacon Transmission" page

breon, 2018-09-13:
Hi there!
It's been awhile since this page has been updated. We support broadcasting in our app and reference this page a bit for customer support.

https://altbeacon.github.io/android-beacon-library/beacon-transmitter-devices.html

Thanks
Breon

davidgyoung, 2018-09-13:
Thanks for the reminder.  I just updated the page with the latest stats.

closed 2018-09-13 by davidgyoung

wahram, 2021-08-22:
Hello,

is there a way to suggest new updates on the page?

davidgyoung, 2021-08-22:
@wahram, as of this date, I think that this page is largely obsolete.  Nearly every Android phone and tablet released in the last couple of years has support for this feature.  Any device that does not have such support is more likely due to an accidental engineering flaw rather than intent.

wahram, 2021-08-23:
Thank you for the answer!