AltBeacon/android-beacon-library #741: Ranging stopped when didEnterRegion() fired while app backgrounded

MikeOrtiz, 2018-09-19:
### Expected behavior
After didEnterRegion() fires and the app starts ranging for beacons, discovered beacons should be returned to didRangeBeaconsInRegion().

### Actual behavior
Beacons are recognized during ranging according to debug logs, but are not passed to didRangeBeaconsInRegion().

### Steps to reproduce this behavior
1. Enter app
2. Background app by hitting home or backing out of the app
3. Turn on beacon
4. Note didEnterRegion() logging and beaconManager.startRangingBeaconsInRegion() logging
5. Note that RangeNotifier callback never fires after beacon is ranged. 

### Mobile device model and OS version
Pixel 2, Android Pie

### Android Beacon Library version
2.15.1. Also, using latest Beacon Library Reference app, with some modifications. Added custom beacons and call beaconManager.startRangingBeaconsInRegion() in BeaconReferenceApplication didEnterRegion(). Add RangeNotifier in didEnterRegion() that prints logging.

### Logs
[Here are the logs](https://gist.githubusercontent.com/MikeOrtiz/126bc16d975869209e17e3f7e28be350/raw/c9972191ad47fb6873be6b6abbac2fd794aba03a/Stopped%2520Ranging) with debugging turned on. Here should be the most relevant snippet:

`09-18 19:09:17.812 14121-14226/org.altbeacon.beaconreference D/BeaconParser: This is not a matching Beacon advertisement. Was expecting 08 18 at offset 9 and 41 63 at offset 13.  The bytes I see are: 0201040302081807ff67010b005cfe1409436f6e746f75723738303248503430333232320000000000000000000000000000000000000000000000000000
    Ignoring pdu type 01
    Ignoring pdu type 02
    Processing pdu type FF: 0201040302081807ff67010b005cfe1409436f6e746f75723738303248503430333232320000000000000000000000000000000000000000000000000000 with startIndex: 9, endIndex: 14
    This is a recognized beacon advertisement -- 67 01 seen
    Bytes are: 0201040302081807ff67010b005cfe1409436f6e746f75723738303248503430333232320000000000000000000000000000000000000000000000000000
09-18 19:09:17.814 14121-14226/org.altbeacon.beaconreference D/ScanHelper: Beacon packet detected for: id1: 0x436f6e746f75723738303248503430333232 with rssi -62
    beacon detected : id1: 0x436f6e746f75723738303248503430333232
09-18 19:09:17.815 14121-14226/org.altbeacon.beaconreference D/ScanHelper: looking for ranging region matches for this beacon
09-18 19:09:27.833 14121-14121/org.altbeacon.beaconreference D/ScanJobScheduler: Applying settings to ScanJob
09-18 19:09:27.913 14121-14121/org.altbeacon.beaconreference D/ScanState: Scan state restore regions: monitored=1 ranged=1
09-18 19:09:27.914 14121-14121/org.altbeacon.beaconreference D/ScanState: ranged regions: old=1 new=0
    monitored regions: old=1 new=1
09-18 19:09:27.915 14121-14121/org.altbeacon.beaconreference D/ScanState: Stopping ranging region: id1: null id2: null id3: null
    Updated state with 0 ranging regions and 1 monitoring regions.
09-18 19:09:27.937 14121-14121/org.altbeacon.beaconreference D/ScanState: Temp file is /data/user/0/org.altbeacon.beaconreference/files/android-beacon-library-scan-state-temp
    Perm file is /data/user/0/org.altbeacon.beaconreference/files/android-beacon-library-scan-state
09-18 19:09:27.940 14121-14121/org.altbeacon.beaconreference D/MonitoringStatus: saveMonitoringStatusIfOn()
09-18 19:09:27.943 14121-14121/org.altbeacon.beaconreference D/ScanJobScheduler: Applying scan job settings with background mode true
09-18 19:09:27.943 14121-14121/org.altbeacon.beaconreference D/ScanState: ScanState says background mode for ScanJob is true
09-18 19:09:27.943 14121-14121/org.altbeacon.beaconreference D/ScanJobScheduler: Not scheduling an immediate scan because we are in background mode.   Cancelling existing immediate scan.
09-18 19:09:27.946 14121-14121/org.altbeacon.beaconreference I/ScanJob: Using immediateScanJobId from manifest: 208352939
09-18 19:09:27.948 14121-14121/org.altbeacon.beaconreference I/ScanJob: Using periodicScanJobId from manifest: 208352940
09-18 19:09:27.949 14121-14121/org.altbeacon.beaconreference W/JobInfo: Requested interval +5m10s0ms for job 208352940 is too small; raising to +15m0s0ms
    Requested flex 0 for job 208352940 is too small; raising to +5m0s0ms
09-18 19:09:27.949 14121-14121/org.altbeacon.beaconreference D/ScanJobScheduler: Scheduling ScanJob (job:208352940/org.altbeacon.beaconreference/org.altbeacon.beacon.service.ScanJob) to run every 310000 millis`

IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

MikeOrtiz, 2018-09-19:
Sure, here is my modified [BeaconReferenceApplication.java](https://gist.github.com/MikeOrtiz/183ed408c3643a55312136bc215c7cd8). It's the only file I modified from the reference app.

davidgyoung, 2018-09-24:
Looking at the code, I see that it starts ranging on region entry, then stops it 10 seconds later.  You will only get  a ranging callback if a beacon is detected in that 10 second period.  I cannot tell from the log attached when the region entry event fires, when ranging starts, and when ranging stops.  I suggest you simply add debug lines for each of these events.  If you do this and you can capture a log that shows that ranging starts and then a beacon is detected *before* ranging stops (but without a ranging callback) then I can look further.  

But I suspect this is simply a timing issue and your app may simply not be ranging when the beacon is detected.

MikeOrtiz, 2018-09-27:
Correct me if I'm wrong, but I think the log already provides all that info. It all occurs at the end of the log.

1. When the region entry event fires: "did Enter Region" is logged
2. When ranging starts: "did start ranging beacons. Region: backgroundRegion" is logged after beaconManager.startRangingBeaconsInRegion(mRegion).
3. When ranging stops: "Stopping ranging region: id1: null id2: null id3: null
    Updated state with 0 ranging regions and 1 monitoring regions." is logged at the very end of the file. That is the library's own debug logging. I did not include logging for this event in my changes to the reference app.

Between the start and the stop, there's logging indicating that the beacon is recognized: "This is a recognized beacon advertisement" and "Beacon packet detected for: id1: 0x436f6e746f75723738303248503430333232 with rssi -62
    beacon detected : id1: 0x436f6e746f75723738303248503430333232"

But, despite the beacon being recognized, the RangeNotifier is not fired. Note that the file is missing the following logging: "Did range beacons in region". Also note that the logging for stopping ranging occurs 10 seconds after the beacon is recognized, so it doesn't look to be a timing issue there.

Let me know if I misunderstood what you needed.

davidgyoung, 2018-09-28:
Sorry, I overlooked the fact that you had included a link to the full log.

I annotated some of the log lines to show some events of interest. See below.

Basically what is happening is that your beacon is detected in the background during an "Extra" scan period caused by the background detection.  This gives you a quick monitoring callback as expected, but the ranging callback doesn't give you results right away because the simple act of starting ranging in this case, actually resets the scan settings.  (See event 9)  This causes the normally 10 second "extra" scan period run due to a background detection event to be cut short.  The library does this because new settings are applied when you start ranging, and it basically says "I'm in the background, I shouldn't be ranging now anyway."

The ranging results would normally come on the next scan cycle, but your code shuts it off after 10 seconds, so it is just a timing problem that it totally fails in this unusual case.

It might be possible to fix this by detecting if the library is in an extra background scan cycle when the new settings are applied, and therefore not canceling the immediate ScanJob.  

But a far, far simpler solution is to just leave ranging on all the time in your program.  Is there any reason you don't want to do this?

```
// 1. log starts out in foreground mode doing a few scan cycles

09-18 19:09:04.805 14121-14121/org.altbeacon.beaconreference D/CycledLeScannerForLollipop: starting non-filtered scan in SCAN_MODE_LOW_LATENCY
09-18 19:09:05.906 14121-14121/org.altbeacon.beaconreference D/CycledLeScanner: Done with scan cycle
09-18 19:09:07.010 14121-14121/org.altbeacon.beaconreference D/CycledLeScanner: Done with scan cycle
09-18 19:09:08.112 14121-14121/org.altbeacon.beaconreference D/CycledLeScanner: Done with scan cycle
09-18 19:09:09.240 14121-14121/org.altbeacon.beaconreference D/CycledLeScanner: Done with scan cycle

// 2. app is switched to background mode here.  

09-18 19:09:09.228 14121-14121/org.altbeacon.beaconreference D/ScanJobScheduler: Applying scan job settings with background mode true
09-18 19:09:09.228 14121-14121/org.altbeacon.beaconreference D/ScanState: ScanState says background mode for ScanJob is true
09-18 19:09:09.228 14121-14121/org.altbeacon.beaconreference D/ScanJobScheduler: Not scheduling an immediate scan because we are in background mode.   Cancelling existing immediate scan.
09-18 19:09:09.229 14121-14121/org.altbeacon.beaconreference I/ScanJob: Using immediateScanJobId from manifest: 208352939
09-18 19:09:09.233 14121-14121/org.altbeacon.beaconreference I/ScanJob: Using periodicScanJobId from manifest: 208352940
09-18 19:09:09.233 14121-14121/org.altbeacon.beaconreference W/JobInfo: Requested interval +5m10s0ms for job 208352940 is too small; raising to +15m0s0ms
    Requested flex 0 for job 208352940 is too small; raising to +5m0s0ms
09-18 19:09:09.233 14121-14121/org.altbeacon.beaconreference D/ScanJobScheduler: Scheduling ScanJob (job:208352940/org.altbeacon.beaconreference/org.altbeacon.beacon.service.ScanJob) to run every 310000 millis

// 3. The foreground scan is ended due to the switch to background mode

09-18 19:09:09.243 14121-14121/org.altbeacon.beaconreference I/ScanJob: onStopJob called for immediate scan org.altbeacon.beacon.service.ScanJob@1fb88fa
09-18 19:09:09.244 14121-14199/org.altbeacon.beaconreference D/CycledLeScannerForLollipop: Stopping LE scan on scan handler
09-18 19:09:09.244 14121-14121/org.altbeacon.beaconreference D/CycledLeScanner: Scanning disabled. 

// 4. because no beacons have been recently seen, a filtered scan is started in the background

09-18 19:09:09.273 14121-14121/org.altbeacon.beaconreference D/ScanHelper: Started passive beacon scan

// 5. A few seconds after entering background mode, a beacon is detected with this passive scan, this kicks off an immediate scan job

09-18 19:09:17.529 14121-14121/org.altbeacon.beaconreference D/StartupBroadcastReceiver: onReceive called in startup broadcast receiver
09-18 19:09:17.532 14121-14121/org.altbeacon.beaconreference D/StartupBroadcastReceiver: Passive background scan callback type: 1
    got Android O background scan via intent
09-18 19:09:17.532 14121-14121/org.altbeacon.beaconreference D/ScanJobScheduler: scheduling an immediate scan job because last did 1537322957532seconds ago.
09-18 19:09:17.596 14121-14121/org.altbeacon.beaconreference D/ScanState: Scan state restore regions: monitored=1 ranged=0
    ScanState says background mode for ScanJob is true
09-18 19:09:17.596 14121-14121/org.altbeacon.beaconreference D/ScanJobScheduler: We just woke up in the background based on a new scan result.  Start scan job immediately.
    Scheduling immediate ScanJob to run in 0 millis
09-18 19:09:17.677 14121-14223/org.altbeacon.beaconreference D/ScanHelper: Beacon packet detected for: id1: 0x436f6e746f75723738303248503430333232 with rssi -50
09-18 19:09:17.702 14121-14121/org.altbeacon.beaconreference D/CycledLeScanner: Waiting to stop scan cycle for another 10000 milliseconds

// 6. Monitoring API didEnterRegion is called

09-18 19:09:17.711 14121-14121/org.altbeacon.beaconreference D/IntentHandler: Calling monitoring notifier: org.altbeacon.beaconreference.BeaconReferenceApplication@e9d807b
09-18 19:09:17.711 14121-14121/org.altbeacon.beaconreference D/Range: did Enter Region

// 7. Beacon detected

[... cut ...]
// 11. Beacon detected

09-18 19:09:17.814 14121-14226/org.altbeacon.beaconreference D/ScanHelper: Beacon packet detected for: id1: 0x436f6e746f75723738303248503430333232 with rssi -62
    beacon detected : id1: 0x436f6e746f75723738303248503430333232

// 12. Stopped ranging due to API call to do so.

09-18 19:09:27.915 14121-14121/org.altbeacon.beaconreference D/ScanState: Stopping ranging region: id1: null id2: null id3: null
    Updated state with 0 ranging regions and 1 monitoring regions.
```

MikeOrtiz, 2018-10-25:
> But a far, far simpler solution is to just leave ranging on all the time in your program. Is there any reason you don't want to do this?

I didn't realize that was how it should be done! I've had better results since switching, though occasionally still getting issues with scanning too frequently. I will open a new ticket if I can capture that behavior. Closing this ticket.

closed 2018-10-25 by MikeOrtiz