AltBeacon/android-beacon-library #742: If bluetooth headset is connected then didn't connect with Beacon

aashu09, 2018-09-21:
Hi Altbeacon support,

I am facing an issue related to Multi-bluetooth device connection described below - 

### Expected behavior
As of now Android introduced simultaneously multiple bluetooth device connections, so altbeacon should also connect with multiple bluetooth devices at the same time. I mean if I am connected with bluetooth headset however I should also connect with Beacon or I f I am connected with Beacon then I should also connect with Bluetooth headset.

### Actual behavior
If I am connected with Beacon and trying to connect with Bluetooth headset then Beacon automatically disconnected or If I am connected with Bluetooth headset and trying to connect with Beacon then It never connects with Beacon untill I am not disconnect with Bluetooth headset.

### Steps to reproduce this behavior
1. Open the app with scanning for bluetooth device
2. connect your device with bluetooth headset
3. Now you will not be connect with Beacon untill disconnect from Bluetooth headset

### Mobile device model and OS version
Every device with Android version 5.0 and above

### Android Beacon Library version
2.15.1

Please look at this issue and resolve it.

Thanks

IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

davidgyoung, 2018-09-21:
The library does not connect to Bluetooth beacons.  It only scans for them and relays detections.  So, I'm afraid the problem description does not quite make sense.

Are you perhaps having a problem detecting beacons?

if so, please provide a test case using the library reference app and describe the specific steps needed to create the condition and the specific hardware and operating system on which you saw the problem.

kapilojha, 2018-09-24:
HI @davidgyoung ,
I think you didn't completely understand the issue, let me clear you again in steps- 
1. Open altbeacon app
2. Scan for beacon
3. Keep your device in the range of beacon, so didEnterRange will be called
4. now try to connect bluetooth headset with your device
5. If bluetooth headset will be connected then didExitRegion for beacon will be called i.e. Beacon will be out of range even your device in the range of beacon.

My problem is - If I connect to the bluetooth headset then didEnterRegion will not be called that means I can only connect one bluetooth device at a time.

Is there any solution to connect multiple bl bluetooth devices simultaneously in altbeacon library.

Thanks

davidgyoung, 2018-09-24:
@kapilojha, can you please confirm if this is an accurate summary of the problem you are having:

* Whenever you connect your mobile device to a bluetooth headset, it is no longer able to detect beacons.

If this is accurate, then I don't think it is a universal issue with the library, but it may be specific to an issue with the bluetooth stack on your mobile device model.  Can you please confirm the mobile device model under test?

I can confirm that I can connect my Nexus 5X to a bluetooth speaker, play music, then detect beacon using the reference app.

kapilojha, 2018-09-25:
I am having Samsung s8 with Android version 8.0.
Also can you detect beacons while still connected to bluetooth speaker ?
If you can connect both simultaneously, can you please provide any recorded video of device or any screen shot ? It will be helpful to me in debugging of my device.

Thanks

davidgyoung, 2018-09-27:
I tried this on two different devices, first paring the device to a Bose Soundlink Mini bluetooth speaker and then running the library reference app.  I transmitted a single AltBeacon at 1 Hz, and played a podcast from the device to the speaker.

Huawei P9 Lite (Android 7.0)
Android Nexus 5X (Android 6.0)

The results were the same for both devices.  Beacons were detected.  Although I will note that there were several seconds between detections on the Nexus 5X, which is not the case when it is not connected to a speaker.

Here's a pic of the setup showing the beacon detections on the Nexus 5x:

[image]

closed 2018-10-03 by davidgyoung