AltBeacon/android-beacon-library #743: Scanning in background does not work with Android P(9.0)

piotrkosecki, 2018-09-27:
### Expected behavior
I would assume that scanning in background would return non-empty list of beacons - like it does with different versions of android. It works ok with backgroundMode set to false(but only when app is active).
### Actual behavior
When I set 
```
val manager get() = BeaconManager.getInstanceForApplication(this)
manager.backgroundMode = true
```
in my service, scans are not being send to my notifier.
### Steps to reproduce this behavior
Use Android P, create service and start scan using background mode.
### Mobile device model and OS version
Google Pixel Android P(9.0)
### Android Beacon Library version
2.15.1

davidgyoung, 2018-09-27:
Understand that if you set `beaconManager.setBackgroundMode(true);` then it changes the way the library behaves.  The library has four settings:

foregroundScanPeriod (1100 ms default)
foregroundBetweenScanPeriod (0ms default)
backgroundScanPeriod (10000 ms default)
backgroundBetweenScanPeriod (300000ms default)

In background mode, the background scan periods become active, which schedule 10 second scans to happen every 300 seconds (5 minutes).  But Android 8+ limits background activity to once every ~15 minutes (unless you configure the library to use a foreground service) so a 10 second scan will actually happen every ~15 minutes.

The expected behavior when raging on Android 9 with backgroundMode = true is that you will get one callback approximately every 15 minutes which will contain a list of beacons seen in the last 10 seconds.

If you want more background responsiveness, you need to configure the library to scan as a foreground service as described [here](https://altbeacon.github.io/android-beacon-library/foreground-service.html).

giladrom, 2018-10-02:
@davidgyoung I'm seeing strange behavior on Android 9 (not sure If it's the same on other versions yet ) - Seems like beacons are not being scanned/found in the background if the phone is locked and not charging. However, when I unlock the phone, I get a background scan result. Does this make sense?

davidgyoung, 2018-10-02:
@giladrom, I am unaware of any Android 9-specific changes like you describe.   However, Doze was introduced starting with Android 6.  If a phone is motionless for a time with the screen off and not charging, the OS will enter doze mode.  

You can tell the difference by disabling the library's background mode processing (e.g. don't use BackgroundPowerSaver or adjust scan periods) then start ranging and verify detections.  Then turn off the screen and seeing if detections immediately stop.  If they stop right away, it is not a doze-specific issue.  If they take several minutes to stop, it may be doze-specific.

giladrom, 2018-10-02:
@davidgyoung That probably explains it. I can see that it behaves as expected when the phone is plugged in. I'll continue testing across my user base. Thank you!

davidgyoung, 2018-10-03:
OK, I'll close this issue now, but please reopen if you see it is something specific to Android 9

closed 2018-10-03 by davidgyoung