AltBeacon/android-beacon-library #744: Reference App in Foreground mode fails at device restart on Android O

rohand27, 2018-09-28:
### Expected behavior
Reference app should restart / resume the scanning service after device has restarted

### Actual behavior
Reference app ends with an erorr message "Unable to create application org.altbeacon.beaconreference.BeaconReferenceApplication: java.lang.IllegalStateException: Not allowed to start service Intent { cmp=org.altbeacon.beaconreference/org.altbeacon.beacon.service.BeaconService }: app is in background uid UidRecord{ffa6d82 u0a108 RCVR idle change:uncached procs:1 seq(0,0,0)}"

### Steps to reproduce this behavior
1] Downloaded Reference App master. 
2] Start Android studio > Open > App location in extracted Folder in Downloads Folder 
3] Uncomment Foreground Notification Code - Left unchanged thereafter
4] Set up a new Notification Channel using Notification Manager [Required for new NotificationBuilder constructor for Android O+]
5] Set up a custom Region [Optional, but required to range our devices]
6] Start up on device
7] Confirm ranging occurs - Passed
8] Restart Device 
9] Observer Logcat with filter set to keyword "BEACON" or "org.altbeacon.beaconreference" 

### Mobile device model and OS version
Xaomi A2 Lite AndroidOne - Android 8.1.0 Kernel 3.18.71... Sept 18 2018

### Android Beacon Library version
org.altbeacon:android-beacon-library:2+ [from Reference App]

IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

Below is the stack trace after the restart: 
2018-09-28 12:12:36.018 2535-2535/? I/BeaconBle: Using BLE 'L' hardware layer
2018-09-28 12:12:36.290 2535-4123/? I/BeaconBle: Client requested scan, settings=BleSettings [scanMode=LOW_LATENCY, callbackType=ALL_MATCHES, reportDelayMillis=0, 6 filters, 1 clients, callingClientName=Nearby]
2018-09-28 12:12:36.320 2535-4123/? I/BeaconBle: 'L' hardware scan: 6 filters, scanMode=2, reportdelay=0, callback type=1, #clients=1, workSource=WorkSource{10015 com.google.android.gms}
2018-09-28 12:12:36.321 2535-4123/? I/BeaconBle: Starting scan on OS scanner. BT state: 12
2018-09-28 12:12:39.146 4535-4535/? I/BeaconManager: BeaconManager started up on pid 4535 named 'org.altbeacon.beaconreference' for application package 'org.altbeacon.beaconreference'.  isMainProcess=true
2018-09-28 12:12:39.157 4535-4535/? D/BeaconParser: Parsing beacon layout: m:2-3=beac,i:4-19,i:20-21,i:22-23,p:24-24,d:25-25
2018-09-28 12:12:39.161 4535-4535/? D/BeaconParser: Parsing beacon layout: m:2-3=beac,i:4-15,i:16-19,i:20-23,p:24-24,d:25-25
2018-09-28 12:12:39.170 4535-4535/? W/BeaconManager: Disabling ScanJobs on Android 8+ may disable delivery of beacon callbacks in the background unless a foreground service is active.
2018-09-28 12:12:39.170 4535-4535/? W/BeaconManager: Disabling ScanJobs on Android 8+ may disable delivery of beacon callbacks in the background unless a foreground service is active.
2018-09-28 12:12:39.170 4535-4535/? W/BeaconManager: Setting a short backgroundBetweenScanPeriod has no effect on Android 8+, which is limited to scanning every ~15 minutes
2018-09-28 12:12:39.170 4535-4535/? D/BeaconReferenceApp: setting up background monitoring for beacons and power saving
2018-09-28 12:12:39.180 1567-2137/? W/ActivityManager: Background start not allowed: service Intent { cmp=org.altbeacon.beaconreference/org.altbeacon.beacon.service.BeaconService } to org.altbeacon.beaconreference/org.altbeacon.beacon.service.BeaconService from pid=4535 uid=10108 pkg=org.altbeacon.beaconreference
2018-09-28 12:12:39.186 4535-4535/? E/AndroidRuntime: FATAL EXCEPTION: main
    Process: org.altbeacon.beaconreference, PID: 4535
    java.lang.RuntimeException: Unable to create application org.altbeacon.beaconreference.BeaconReferenceApplication: java.lang.IllegalStateException: Not allowed to start service Intent { cmp=org.altbeacon.beaconreference/org.altbeacon.beacon.service.BeaconService }: app is in background uid UidRecord{84e7fb1 u0a108 RCVR idle change:uncached procs:1 seq(0,0,0)}
        at android.app.ActivityThread.handleBindApplication(ActivityThread.java:5758)
        at android.app.ActivityThread.-wrap1(Unknown Source:0)
        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1659)
        at android.os.Handler.dispatchMessage(Handler.java:106)
        at android.os.Looper.loop(Looper.java:164)
        at android.app.ActivityThread.main(ActivityThread.java:6518)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:438)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:807)
     Caused by: java.lang.IllegalStateException: Not allowed to start service Intent { cmp=org.altbeacon.beaconreference/org.altbeacon.beacon.service.BeaconService }: app is in background uid UidRecord{84e7fb1 u0a108 RCVR idle change:uncached procs:1 seq(0,0,0)}
        at android.app.ContextImpl.startServiceCommon(ContextImpl.java:1521)
        at android.app.ContextImpl.startService(ContextImpl.java:1477)
        at android.content.ContextWrapper.startService(ContextWrapper.java:650)
        at org.altbeacon.beacon.startup.RegionBootstrap$InternalBeaconConsumer.bindService(RegionBootstrap.java:171)
        at org.altbeacon.beacon.BeaconManager.bind(BeaconManager.java:433)
        at org.altbeacon.beacon.startup.RegionBootstrap.<init>(RegionBootstrap.java:58)
        at org.altbeacon.beaconreference.BeaconReferenceApplication.onCreate(BeaconReferenceApplication.java:90)
        at android.app.Instrumentation.callApplicationOnCreate(Instrumentation.java:1120)
        at android.app.ActivityThread.handleBindApplication(ActivityThread.java:5755)
        at android.app.ActivityThread.-wrap1(Unknown Source:0) 
        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1659) 
        at android.os.Handler.dispatchMessage(Handler.java:106) 
        at android.os.Looper.loop(Looper.java:164) 
        at android.app.ActivityThread.main(ActivityThread.java:6518) 
        at java.lang.reflect.Method.invoke(Native Method) 
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:438) 
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:807) 
2018-09-28 12:12:42.273 2535-4123/? I/BeaconBle: Client requested scan, settings=BleSettings [scanMode=ZERO_POWER, callbackType=ALL_MATCHES, reportDelayMillis=0, 6 filters, 1 clients, callingClientName=Nearby]
2018-09-28 12:12:42.313 2535-4123/? I/BeaconBle: Stopping scan on delegate scanner. BT state: 12
2018-09-28 12:12:42.325 2535-4123/? I/BeaconBle: Resetting - new scanner available: true
2018-09-28 12:12:42.327 2535-4123/? I/BeaconBle: 'L' hardware scan: 6 filters, scanMode=-1, reportdelay=0, callback type=1, #clients=1, workSource=WorkSource{10015 com.google.android.gms}
2018-09-28 12:12:42.329 2535-4123/? I/BeaconBle: Starting scan on OS scanner. BT state: 12
2018-09-28 12:13:04.476 2535-4123/? I/BeaconBle: Client requested scan, settings=BleSettings [scanMode=LOW_LATENCY, callbackType=ALL_MATCHES, reportDelayMillis=0, 6 filters, 1 clients, callingClientName=Nearby]
2018-09-28 12:13:04.482 2535-4123/? I/BeaconBle: Stopping scan on delegate scanner. BT state: 12
2018-09-28 12:13:04.485 2535-4123/? I/BeaconBle: Resetting - new scanner available: true
2018-09-28 12:13:04.486 2535-4123/? I/BeaconBle: 'L' hardware scan: 6 filters, scanMode=2, reportdelay=0, callback type=1, #clients=1, workSource=WorkSource{10015 com.google.android.gms}
2018-09-28 12:13:04.487 2535-4123/? I/BeaconBle: Starting scan on OS scanner. BT state: 12
2018-09-28 12:13:08.103 2535-2535/? I/BeaconBle: Client requested scan, settings=BleSettings [scanMode=ZERO_POWER, callbackType=ALL_MATCHES, reportDelayMillis=0, 1 filters, 0 clients, callingClientName=Places]
2018-09-28 12:13:10.468 2535-4123/? I/BeaconBle: Client requested scan, settings=BleSettings [scanMode=ZERO_POWER, callbackType=ALL_MATCHES, reportDelayMillis=0, 6 filters, 1 clients, callingClientName=Nearby]
2018-09-28 12:13:10.488 2535-4123/? I/BeaconBle: Stopping scan on delegate scanner. BT state: 12
2018-09-28 12:13:10.495 2535-4123/? I/BeaconBle: Resetting - new scanner available: true
2018-09-28 12:13:10.500 2535-4123/? I/BeaconBle: 'L' hardware scan: 6 filters, scanMode=-1, reportdelay=0, callback type=1, #clients=2, workSource=WorkSource{10015 com.google.android.gms}
2018-09-28 12:13:10.502 2535-4123/? I/BeaconBle: Starting scan on OS scanner. BT state: 12

davidgyoung, 2018-09-28:
Thanks for this report. I believe we simply need to conditionally change the way the service is started if a foreground service is enabled per the Android migration guidelines:

> If your app needs to create a foreground service while the app is in the background, use the startForegroundService() method instead of startService()

rohand27, 2018-09-29:
Thanks for the response @davidgyoung. We have the same issue with our development app as well on v. 2.15.1. Any suggestion on how to get past this? We need to have the service ranging even if the device is restarted. It seems a service is attempting to start the app every minute or so, but fails to launch with the same Exception.

davidgyoung, 2018-10-03:
I think this change may solve this problem:  #746

davidgyoung, 2018-10-03:
@rohand27, I have prepared a binary test release in the linked PR.  If you can test this with your app and let me know if it solves the problem I'd appreciate the help.  Note that you have to set up a notification channel for the notification as described in the lined PR.

rohand27, 2018-10-04:
If I assume installed the patch aar file correctly... [through file > New Module > Import jar / aar method], there is no ranging activity happening at all. Below is the stack. 

`2018-10-04 15:55:25.325 31448-31448/? I/zygote64: Late-enabling -Xcheck:jni
2018-10-04 15:55:25.389 31448-31458/? E/zygote64: Failed sending reply to debugger: Broken pipe
2018-10-04 15:55:25.389 31448-31458/? I/zygote64: Debugger is no longer active
2018-10-04 15:55:25.542 31448-31448/? I/InstantRun: starting instant run server: is main process
2018-10-04 15:55:25.558 31448-31448/? I/BeaconManager: BeaconManager started up on pid 31448 named 'org.altbeacon.beaconreference' for application package 'org.altbeacon.beaconreference'.  isMainProcess=true
2018-10-04 15:55:25.567 31448-31448/? D/BeaconParser: Parsing beacon layout: m:2-3=beac,i:4-19,i:20-21,i:22-23,p:24-24,d:25-25
2018-10-04 15:55:25.571 31448-31448/? D/BeaconParser: Parsing beacon layout: m:2-3=beac,i:4-15,i:16-19,i:20-23,p:24-24,d:25-25
2018-10-04 15:55:25.577 31448-31448/? W/BeaconManager: Disabling ScanJobs on Android 8+ may disable delivery of beacon callbacks in the background unless a foreground service is active.
2018-10-04 15:55:25.577 31448-31448/? W/BeaconManager: Disabling ScanJobs on Android 8+ may disable delivery of beacon callbacks in the background unless a foreground service is active.
2018-10-04 15:55:25.577 31448-31448/? W/BeaconManager: Setting a short backgroundBetweenScanPeriod has no effect on Android 8+, which is limited to scanning every ~15 minutes
2018-10-04 15:55:25.577 31448-31448/? D/BeaconReferenceApp: setting up background monitoring for beacons and power saving
2018-10-04 15:55:25.608 31448-31448/? D/MonitoringActivity: onCreate
2018-10-04 15:55:25.752 31448-31475/? D/OpenGLRenderer: HWUI GL Pipeline
2018-10-04 15:55:25.764 31448-31448/? W/BluetoothCrashResolver: Can't read macs from BluetoothCrashResolverState.txt
2018-10-04 15:55:25.770 31448-31448/? I/CycledLeScanner: Using Android O scanner
2018-10-04 15:55:25.774 31448-31448/? I/BeaconService: beaconService version 2.15.1-1-gccb2071 is starting up on the main process
2018-10-04 15:55:25.778 31448-31448/? I/BeaconService: longScanForcingEnabled to keep scans going on Android N for > 30 minutes
2018-10-04 15:55:25.787 31448-31448/? W/ModelSpecificDistanceCalculator: Cannot find match for this device.  Using default
2018-10-04 15:55:25.787 31448-31448/? W/ModelSpecificDistanceCalculator: Cannot find match for this device.  Using default
2018-10-04 15:55:25.794 31448-31448/? I/BeaconService: starting with intent Intent { cmp=org.altbeacon.beaconreference/org.altbeacon.beacon.service.BeaconService }
2018-10-04 15:55:25.836 31448-31475/? I/Adreno: QUALCOMM build                   : 8e3df98, Ie4790512f3
    Build Date                       : 04/11/18
    OpenGL ES Shader Compiler Version: EV031.22.00.01
    Local Branch                     : 
    Remote Branch                    : quic/gfx-adreno.lnx.1.0.r36-rel
    Remote Branch                    : NONE
    Reconstruct Branch               : NOTHING
2018-10-04 15:55:25.841 31448-31475/? I/Adreno: PFP: 0x005ff087, ME: 0x005ff063
2018-10-04 15:55:25.846 31448-31475/? I/zygote64: android::hardware::configstore::V1_0::ISurfaceFlingerConfigs::hasWideColorDisplay retrieved: 0
2018-10-04 15:55:25.848 31448-31475/? I/OpenGLRenderer: Initialized EGL, version 1.4
2018-10-04 15:55:25.848 31448-31475/? D/OpenGLRenderer: Swap behavior 2
2018-10-04 15:55:26.028 31448-31448/? I/AssistStructure: Flattened final assist data: 2056 bytes, containing 1 windows, 6 views
2018-10-04 15:55:32.726 31448-31448/org.altbeacon.beaconreference I/BeaconService: starting with intent Intent { cmp=org.altbeacon.beaconreference/org.altbeacon.beacon.service.BeaconService }
2018-10-04 15:55:32.760 31448-31448/org.altbeacon.beaconreference I/AssistStructure: Flattened final assist data: 1396 bytes, containing 1 windows, 4 views
2018-10-04 15:55:51.105 31448-31456/org.altbeacon.beaconreference I/zygote64: Do partial code cache collection, code=30KB, data=24KB
2018-10-04 15:55:51.105 31448-31456/org.altbeacon.beaconreference I/zygote64: After code cache collection, code=30KB, data=24KB
2018-10-04 15:55:51.105 31448-31456/org.altbeacon.beaconreference I/zygote64: Increasing code cache capacity to 128KB
2018-10-04 15:55:51.400 31448-31448/org.altbeacon.beaconreference D/AndroidRuntime: Shutting down VM
2018-10-04 15:55:51.403 31448-31448/org.altbeacon.beaconreference E/AndroidRuntime: FATAL EXCEPTION: main
    Process: org.altbeacon.beaconreference, PID: 31448
    java.lang.RuntimeException: Unable to destroy activity {org.altbeacon.beaconreference/org.altbeacon.beaconreference.RangingActivity}: java.lang.IllegalArgumentException: Service not registered: org.altbeacon.beacon.BeaconManager$BeaconServiceConnection@605b3d5
        at android.app.ActivityThread.performDestroyActivity(ActivityThread.java:4386)
        at android.app.ActivityThread.handleDestroyActivity(ActivityThread.java:4404)
        at android.app.ActivityThread.-wrap5(Unknown Source:0)
        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1652)
        at android.os.Handler.dispatchMessage(Handler.java:106)
        at android.os.Looper.loop(Looper.java:164)
        at android.app.ActivityThread.main(ActivityThread.java:6518)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:438)
[... cut ...]
Settings.gradle contents
include ':app', ':android-beacon-library-2.15.1-1-gccb2071'

app.gradle dependencies: 
`dependencies {
    compile fileTree(dir: 'libs', include: ['*.aar'])
    compile 'com.android.support:appcompat-v7:27.0.0'
    //compile project(':android-beacon-library')
    //compile 'org.altbeacon:android-beacon-library:2+'
    //compile 'android-beacon-library:2.15.1-1-gccb2071@aar'
    //compile project(path:':android-beacon-library')
}
`
The app crashes when I hit the back button on RangingActivity

davidgyoung, 2018-10-04:
Yep, I see: `java.lang.RuntimeException: Unable to destroy activity {org.altbeacon.beaconreference/org.altbeacon.beaconreference.RangingActivity}: java.lang.IllegalArgumentException: Service not registered: org.altbeacon.beacon.BeaconManager$BeaconServiceConnection@605b3d5 at `.  Let me investigate.

davidgyoung, 2018-10-08:
@rohand27 I think I have a second test release that fixes it.  See the binary build here:  https://github.com/AltBeacon/android-beacon-library/releases/tag/fix-start-foreground-service-at-boot-release-2

The trick is that I needed to start the service the old-fashioned way as an intent service *after* starting  it the new Android 8+ way as a foreground service.  This ensured that it was properly bound to the client that started it.

I tested this on a Samsung S9 with Android 8.0, and rebooted the device, then verified I saw the library start up and begin beacon scanning.

Please give it a shot and let me know if you see the same.

rohand27, 2018-10-08:
@davidgyoung  TLDR; It seems to work and fire up the foreground service after device restarts on Android O!

The reference app seems to range, restart and range again. Below is my testing methodology. Let me know if that looks okay, as I was not able to use the steps listed on the release page [ I am still a noob at this and I don't think I have importing other projects nailed well enough=( ]

I have added my project w/ the module import to my bitbucket repo. I can give you access if needed. 

I tested the new release with a clean reference app configured w/ notification channels etc as according the previous comments. I used the <project name> Right Click > New > Module > Import Jar / AAR Package method to import the new release into the app as a library. 

settings.gradle file: 
`//include ':android-beacon-library'
include ':app', ':android-beacon-library-release'`

module gradle: 
`configurations.maybeCreate("default")
artifacts.add("default", file('android-beacon-library-release.aar'))`

project gradle: 
`// Top-level build file where you can add configuration options common to all sub-projects/modules.

buildscript {
    repositories {
        jcenter()
        maven {
            url 'https://maven.google.com/'
            name 'Google'
        }
        google()
    }
    dependencies {
        classpath 'com.android.tools.build:gradle:3.2.0'

        // NOTE: Do not place your application dependencies here; they belong
        // in the individual module build.gradle files
    }
}

allprojects {
    repositories {
        jcenter()
        maven {
            url 'https://maven.google.com/'
            name 'Google'
        }
        flatDir {
            dirs "src/main/libs"
        }
    }
}

task clean(type: Delete) {
    delete rootProject.buildDir
}
`
App gradle [Notice I commented out all compile references to android-beacon-lib.
`apply plugin: 'com.android.application'

android {
    compileSdkVersion 27
    buildToolsVersion '27.0.3'

    defaultConfig {
        applicationId "org.altbeacon.beaconreference"
        minSdkVersion 19
        targetSdkVersion 27
    }
    buildTypes {
        release {
            minifyEnabled false
            proguardFiles getDefaultProguardFile('proguard-android.txt'), 'proguard-rules.pro'
        }
    }
}

dependencies {
    compile fileTree(include: ['*.aar'], dir: 'libs')
    compile 'com.android.support:appcompat-v7:27.0.0'
    //compile project(':android-beacon-library')
    //compile 'android-beacon-library:fix-start-foreground-service-at-boot-release-1-1-g75f756f@aar'
    //compile 'org.altbeacon:android-beacon-library:2+'
}
/*
dependencies {

}*/
`

davidgyoung, 2018-10-17:
This is now in 2.15.2

closed 2018-10-17 by davidgyoung