AltBeacon/android-beacon-library #749: After ugprade to Android P, BLE does not scan anymore

ninchan8328, 2018-10-09:
### Expected behavior
Same software works just fine on Android O which the app start up and enable the BLE as foreground service.

### Actual behavior
After upgrade to Android P, service still started, and i see from the log that it still transmitting signal, but not scanning any signal.

### Steps to reproduce this behavior
just open the app

### Mobile device model and OS version
Pixel 2 Android P

### Android Beacon Library version
2.15

Here is my code to enable BLE:

        Log.d(logTag, "starting BLE foreground service");
        beaconManager.setForegroundScanPeriod(Config.SHORT_SCAN_PERIOD);
        beaconManager.setForegroundBetweenScanPeriod(Config.SHORT_BETTWEEN_SCAN_PERIOD);

        /**
         * For the above foreground scanning service to be useful, you need to disable
         * JobScheduler-based scans (used on Android 8+) and set a fast background scan
         * cycle that would otherwise be disallowed by the operating system.
         */
        beaconManager.setEnableScheduledScanJobs(false);
        beaconManager.setBackgroundBetweenScanPeriod(0);
        beaconManager.setBackgroundScanPeriod(1100);

        //setForegroundBetweenScanPeriod(sharedPreferences.getInt(Config.KEY_BLE_BETWEEN_SCAN_PERIOD, Config.SHORT_BACKGROUND_BETWEEN_SCAN_PERIOD));

        beaconManager.enableForegroundServiceScanning(appDefaultNotification.buildNotification(), Constants.NOTIFICATION_ID.FOREGROUND_SERVICE);

        /**
         * simply constructing this class and holding a reference to it in your custom Application
         * class will automatically cause the BeaconLibrary to save battery whenever the application
         * is not visible.  This reduces bluetooth power usage by about 60%
         */
        backgroundPowerSaver = new BackgroundPowerSaver(this);

        beaconManager.getBeaconParsers().clear();
        beaconManager.getBeaconParsers().add(new BeaconParser().
                setBeaconLayout("m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24"));      // iBeacon Layout

IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

davidgyoung, 2018-10-09:
Can you please set `beaconManager.setDebug(true);` then attach a 30 second log excerpt after starting the app with beacons in the vicinity?

ninchan8328, 2018-10-09:
10-08 22:19:27.417 11317-11317/com.firsapp.testing I/FirstApp.NearbyController: Starting NearBy feature
10-08 22:19:27.421 11317-11317/com.firsapp.testing I/FirstApp.BLEController: Started BLE transmitter
10-08 22:19:27.422 11317-11317/com.firsapp.testing D/BeaconParser: Parsing beacon layout: m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24
10-08 22:19:27.424 11317-11317/com.firsapp.testing D/BeaconTransmitter: new BeaconTransmitter constructed.  mbluetoothLeAdvertiser is android.bluetooth.le.BluetoothLeAdvertiser@f7a9c89
10-08 22:19:27.425 11317-11317/com.firsapp.testing D/BeaconParser: Identifier size is just right: c9 40 7f 30 f5 f8 46 6e af f9 25 55 6b 57 fe 6d
10-08 22:19:27.426 11317-11317/com.firsapp.testing D/BeaconParser: Identifier size is just right: 00 00
10-08 22:19:27.426 11317-11317/com.firsapp.testing D/BeaconParser: Identifier size is just right: 00 04
10-08 22:19:27.428 11317-11317/com.firsapp.testing D/BeaconTransmitter: Starting advertising with ID1: c9407f30-f5f8-466e-aff9-25556b57fe6d ID2: 0 ID3: 4 and data: 02 15 C9 40 7F 30 F5 F8 46 6E AF F9 25 55 6B 57 FE 6D 00 00 00 04 B5  of size 23
10-08 22:19:27.429 11317-11317/com.firsapp.testing D/BluetoothAdapter: isLeEnabled(): ON
10-08 22:19:27.430 11317-11317/com.firsapp.testing D/BluetoothAdapter: isLeEnabled(): ON
10-08 22:19:27.431 1596-2285/? D/BluetoothAdapterService: getAdapterService() - returning com.android.bluetooth.btservice.AdapterService@eec368d
10-08 22:19:27.432 11317-11317/com.firsapp.testing D/BeaconTransmitter: Started advertisement with callback: org.altbeacon.beacon.BeaconTransmitter$1@82fdca8
10-08 22:19:27.433 11317-11317/com.firsapp.testing D/FirstApp.BeaconScanner: BeaconScanList is updated
10-08 22:19:27.433 11317-11317/com.firsapp.testing D/Ning: ScanList item BeaconSignal{uuid='c9407f30-f5f8-466e-aff9-25556b57fe6d', major='0', minor='0', distance='0.0}
10-08 22:19:27.433 11317-11317/com.firsapp.testing D/FirstApp.BeaconScanner: BeaconScanList is updated
10-08 22:19:27.433 11317-11317/com.firsapp.testing D/FirstApp: ScanList item BeaconSignal{uuid='c9407f30-f5f8-466e-aff9-25556b57fe6d', major='0', minor='0', distance='0.0}
10-08 22:19:27.433 11317-11317/com.firsapp.testing D/FirstApp: ScanList item BeaconSignal{uuid='b9407f30-f5f8-466e-aff9-25556b57fe6d', major='0', minor='0', distance='0.0}
10-08 22:19:27.434 11317-11317/com.firsapp.testing D/FirstApp.BeaconScanner: startScanning
10-08 22:19:27.434 11317-11317/com.firsapp.testing D/FirstApp.BeaconScanner: Request received to START SCANNING
10-08 22:19:27.434 11317-11317/com.firsapp.testing D/FirstApp.BeaconScanner: Start service intent for BLE
10-08 22:19:27.435 11317-11317/com.firsapp.testing D/FirstApp.BeaconScanner: setScanning
10-08 22:19:27.436 11317-11317/com.firsapp.testing I/FirstApp.BLEController: beaconScanner should be started
10-08 22:19:27.437 11317-11317/com.firsapp.testing D/BluetoothMedic: Medic monitoring for transmission and scan failure notifications with receiver: org.altbeacon.bluetooth.BluetoothMedic$1@485d6d3
10-08 22:19:27.437 11317-11317/com.firsapp.testing D/BluetoothMedic: Medic scheduling periodic tests of types 3
10-08 22:19:27.438 11317-11317/com.firsapp.testing I/BluetoothTestJob: Using BluetoothTestJob JobId from manifest: 1799803768
10-08 22:19:27.443 11317-11317/com.firsapp.testing I/FirstApp.LocationController: Starting location
10-08 22:19:27.444 11317-11317/com.firsapp.testing I/FirstApp.LocationController: internal call to start location request
10-08 22:19:27.451 11317-11317/com.firsapp.testing I/FirstApp.LocationController: Location started
10-08 22:19:27.452 11317-11317/com.firsapp.testing I/FirstApp.Periodic.WorkManagerHelperFunctions: Scheduling job for PIMUserSyncRequestTag
10-08 22:19:27.471 11317-11317/com.firsapp.testing D/FirstApp.BeaconScanningHelper: onCreate BeaconScanningHelper
10-08 22:19:27.472 1157-2726/? I/GnssLocationProvider: WakeLock acquired by sendMessage(SET_REQUEST, 0, com.android.server.location.GnssLocationProvider$GpsRequest@cee7494)
10-08 22:19:27.473 11317-11317/com.firsapp.testing D/FirstApp.BeaconScanningHelper: starting BLE foreground service
10-08 22:19:27.473 11317-11317/com.firsapp.testing W/BeaconManager: Disabling ScanJobs on Android 8+ may disable delivery of beacon callbacks in the background unless a foreground service is active.
10-08 22:19:27.473 11317-11317/com.firsapp.testing W/BeaconManager: Setting a short backgroundBetweenScanPeriod has no effect on Android 8+, which is limited to scanning every ~15 minutes
10-08 22:19:27.475 11317-11317/com.firsapp.testing D/FirstApp.AppDefaultNotification: current foregroundBetweenScanPriod 4000
10-08 22:19:27.476 1157-1207/? I/GnssLocationProvider: WakeLock released by handleMessage(SET_REQUEST, 0, com.android.server.location.GnssLocationProvider$GpsRequest@cee7494)
10-08 22:19:27.504 829-906/? D/CHRE: @ 274.853: +: id 20, otherClientPresent 1, mode 3
10-08 22:19:27.506 829-906/? D/CHRE: @ 274.856: +: id 10, otherClientPresent 1, mode 3
10-08 22:19:27.507 829-906/? D/CHRE: @ 274.856: [ImuCal] Dynamic sensor configuration: high-performance.
10-08 22:19:27.507 829-906/? D/CHRE: @ 274.856: sensorType 9 allowed 1: mergedMode 3, otherClientPresent 1
10-08 22:19:27.508 829-906/? D/CHRE: @ 274.857: sensorType 12 allowed 1: mergedMode 3, otherClientPresent 1
10-08 22:19:27.509 829-906/? D/CHRE: @ 274.858: sensorType 14 allowed 1: mergedMode 3, otherClientPresent 1
[... cut ...]
10-08 22:19:38.934 1157-2009/? I/WifiService: releaseWifiLock uid=10013
10-08 22:19:39.237 1157-1455/? I/GnssLocationProvider: WakeLock acquired by sendMessage(REPORT_SV_STATUS, 0, com.android.server.location.GnssLocationProvider$SvStatusInfo@88df68d)
10-08 22:19:39.238 1157-1207/? I/GnssLocationProvider: WakeLock released by handleMessage(REPORT_SV_STATUS, 0, com.android.server.location.GnssLocationProvider$SvStatusInfo@88df68d)
10-08 22:19:39.588 11473-11554/? W/CrashLoopRemedyLog: unable to delete remedy log, instaCrash: true

davidgyoung, 2018-10-09:
Are you sure your beacon is transmitting correctly?  I see lots of BLE
detections in the logs, but they all show the byte sequence ff4c0005,
whereas the beacon layout you are looking for is supposed to start with
ff4c000215.  This appears to be why it is not detecting.

On Mon, Oct 8, 2018 at 11:33 PM ninchan8328 <notifications@github.com>
wrote:

> 10-08 22:19:27.417 11317-11317/com.firsapp.testing
> I/FirstApp.NearbyController: Starting NearBy feature
> 10-08 22:19:27.421 11317-11317/com.firsapp.testing
> I/FirstApp.BLEController: Started BLE [...]

ninchan8328, 2018-10-09:
I am pretty positive the code is working and was able to transmit and scan signal. My other older device is still able to detect the Android P since it is still transmitting okay, just that the Android P can no longer detects other device

ninchan8328, 2018-10-09:
David, seems like the problem was on the older device that BLE stopped working and i need to restart the device to get BLE working again. Sorry for the confusion and seems like Android P is working just fine with your library. Thanks for your time to look in to this.

closed 2018-10-09 by ninchan8328