AltBeacon/android-beacon-library #754: Beacons are still scanning after Bluetooth is turned off in Background Service

shaikm, 2018-10-22:
### Beacons should not be scanned if Bluetooth is turned off

### Beacons are still scanning after device Bluetooth is turned off

### Install the sample and uncomment the code of beacons manger in Reference Application related to service and run the code. Disable Bluetooth and check the Logcat, Still ablet to see the beacons count continuously.

### Nexus 5X, OS version: 8.1.0

### 2.15

rodmaz, 2018-10-22:
Based on our tests and evaluation, from Android 8 on, the behaviour in some smartphones is similar to the behaviour seen on iOS 11, i.e. BLE is not turned off completely and scanning works properly even when user turns it off from the UI.
We saw this behaviour also on other devices such as Samsung S8 (Android 8). Another device we tested was Motorola MotoG5S (Android 8.1) and here the behaviour was similar to previous version (no BLE, no scanning).
All is very odd, inconsistent and not documented anywhere. Please share if you know more information about this.

Steppschuh, 2018-10-25:
Here's an observation my team made yesterday (not related to this library but similar behaviour):

- Device A is scanning BLE advertising frames
- Device B is sending and scanning BLE advertising frames
- Disabled Bluetooth on Device B from the quick settings dropdown
- Device B keeps sending and scanning BLE advertising frames
- Device A keeps receiving BLE advertising frames from device B

Device A is an Raspberry Pi 3B running a custom Bluetooth stack.
Device B is a Pixel 2 running Android 9 and used [RxAndroidBle](https://github.com/Polidea/RxAndroidBle) 1.6.0.

As @rodmaz mentioned, the Bluetooth adapter is not actually turned off when disabling Bluetooth from the settings. It will still be used for location tracking, if not disabled in `Settings > Security & Location > Location > Advanced > Bluetooth Scanning`

rohand27, 2018-11-02:
I can confirm this behavior on Asus Zenfone Max Plus when we were doing a demo w/ Firebase. We switched off the Bluetooth stack, then made a few pings on the test device. The phone should not have detected the pings, but it did and the pings were captured by the app.

davidgyoung, 2018-11-09:
Unfortunately, I cannot test this on my Nexus 5X which is stuck on 6.0 as its USB port's data functions were killed by a power surge, it's rooted and not eligible for OTA upgrades.  However, I can confirm this is not universal on Android 8.1, as testing with a Nokia 6.1 plus (Android One) with 8.1 shows detections stop immediately when bluetooth is turned off from the Quick Settings.  

Per @Steppschuh's comment above, I verified in Quick Settings that `Settings > Security & Location > Location > Advanced > Bluetooth Scanning` is set to enable this:  "Bluetooth Scanning: Improve location by allowing system apps and services to detect Bluetooth devices at any time."  Yet even with this enabled, detections stop when bluetooth is turned off from Quick Settings.

That setting specifically mentions system apps being allowed to do scans in this case.  In my case the test app was not a system app.   I'm curious to hear if any of the above reports of scans continuing to work are for apps explicitly installed as system apps on rooted devices, of if this has actually been witnessed by non-system apps.

This issue was opened specifically referencing the Nexus 5X which runs stock Android, and this library's reference application which was presumably not installed as a system app.  That sounds like unintended behavior.

rodmaz, 2018-11-09:
In our case, tests were performed w/ our own developed apps using native BLE scanning (no AltBeacon library). The devices were not rooted or significantly modified in any way.

davidgyoung, 2019-01-09:
Update with Nokia 6.1 Plus (Android One) 9.0: Still not seeing any BLE scan results with Bluetooth turned off in quick settings, but Settings -> Location and Privacy -> Location -> Scanning -> Bluetooth Scanning (on)  "Allow apps and services to scan for nearby devices at any time even when Bluetooth is off.  This can be used, for example to improve location-based features and services."

pranavb25, 2019-01-28:
Any update on this issue? Not seeing any BLE scan results with Bluetooth off but Bluetooth scanning enabled on OnePlus 5 with Android 9. Tested on a custom app and on the nRF Connect app.

parthrparekh93, 2019-05-22:
I observed this issue on Android 8.0 Samsung galaxy S8. Turning OFF bluetooth did not stop ble scanning and I was able to observe beacons even after 10 minutes of disabling bluetooth. Based on the comments, I checked the Settings  > Connections > Location > Improve Accuracy > Bluetooth Scanning "Improve location accuracy by allowing apps and services to scan for and connect to nearby devices automatically via Bluetooth, even while Bluetooth is turned off" and found it to enabled. I disabled it and it finally stopped detecting beacons.
Note : I am not using Android Beacon Library. I call BLEScanner startScan to start scanning. I do not explicitly call BLEScanner stopScan method when Bluetooth is disabled.

davidgyoung, 2023-07-01:
Closed in favor of a new issue where this has been reproduced:  https://github.com/AltBeacon/android-beacon-library/issues/1148

closed 2023-07-01 by davidgyoung