AltBeacon/android-beacon-library #756: didRangeBeaconInRegion()(Android 8+) is not called when we unbind and bind again

AndroidTeam2019, 2018-10-24:
### Expected behavior
didRangeBeaconInRegion()(Android 8+) should be called when we unbind and bind again

### Actual behavior
didRangeBeaconInRegion()(Android 8+) is not called when we unbind and bind again

### Steps to reproduce this behavior
Please find the attached java file below
1)we created two buttons(bind,unbind)
2)In bind onClick,we wrote the following code
private void beaconSetup() {
        Log.e("beaconSetup","beaconSetup");
        beaconManager = BeaconManager.getInstanceForApplication(this);

        beaconManager.getBeaconParsers().add(new BeaconParser().
                setBeaconLayout("m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24"));
        beaconManager.setMonitorNotifier(this);

        region = new Region("regionId", null, null, null);
        beaconManager.bind(this);

    }
3)In unbind onClick, we wrote the following code
 public void unbind(){
        Log.e("unbind", "unbind");
        try {
            beaconManager.stopRangingBeaconsInRegion(region);
        } catch (RemoteException e) {
            e.printStackTrace();
        }
        //if (beaconManager.isBound(this)) beaconManager.unbind(this);
        beaconManager.unbind(this);
    }
4)Click on bind button didRangeBeaconInRegion() is called continously and the we click on unbind didRangeBeaconInRegion() stop calling and then again we click on bind button then didRangeBeaconInRegion() is not getting call.
This is working fine in all devices with OS version below 8
Not working in Android Version 8+

### Mobile device model and OS version
Mobile device: Samsung S8
OS: Android 8.0
### Android Beacon Library version
2.15.2

`
**### **CODE****
import android.os.RemoteException;
import android.support.v7.app.AppCompatActivity;
import android.os.Bundle;
import android.util.Log;
import android.view.View;
import android.widget.Button;
import com.wcm.noncache.R;
import org.altbeacon.beacon.Beacon;
import org.altbeacon.beacon.BeaconConsumer;
import org.altbeacon.beacon.BeaconManager;
import org.altbeacon.beacon.BeaconParser;
import org.altbeacon.beacon.RangeNotifier;
import org.altbeacon.beacon.Region;

import org.altbeacon.beacon.startup.BootstrapNotifier;

import java.util.ArrayList;
import java.util.Collection;

public class TestingActivity extends AppCompatActivity implements BeaconConsumer, BootstrapNotifier, RangeNotifier,View.OnClickListener {
    private BeaconManager beaconManager = null;
    private Region region;
    ArrayList<Beacon> beaconsList = new ArrayList<>();
    Button bind,unbind;

    @Override
    protected void onCreate(Bundle savedInstanceState) {
        super.onCreate(savedInstanceState);
        setContentView(R.layout.activity_testing);
        findView();

    }
    public void findView(){
        bind = findViewById(R.id.bind);
        unbind = findViewById(R.id.unbind);
        bind.setOnClickListener(this);
        unbind.setOnClickListener(this);
    }

    private void beaconSetup() {
        Log.e("beaconSetup","beaconSetup");
        beaconManager = BeaconManager.getInstanceForApplication(this);

        beaconManager.getBeaconParsers().add(new BeaconParser().
                setBeaconLayout("m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24"));
        beaconManager.setMonitorNotifier(this);

        region = new Region("regionId", null, null, null);
        beaconManager.bind(this);

    }

    @Override
    public void onResume() {
        super.onResume();
        //beaconSetup();
       /* Log.e("onresume", "onresume");
        Log.e("isBound", String.valueOf(beaconManager.isBound(this)));*/
    }

    @Override
    public void onBeaconServiceConnect() {
        Log.e("onBeaconServiceConnect","onBeaconServiceConnect");
        beaconManager.setRangeNotifier(this);
        try {
            Log.e("startRangingBeacons","startRangingBeaconsInRegion");
            beaconManager.startRangingBeaconsInRegion(region);
        } catch (RemoteException e) {
            e.printStackTrace();
        }
    }

    @Override
    public void onStop() {
        super.onStop();
        Log.e("onstop", "onstop");
        try {
            beaconManager.stopRangingBeaconsInRegion(region);
        } catch (RemoteException e) {
            e.printStackTrace();
        }
        if (beaconManager.isBound(this)) beaconManager.unbind(this);

    }

    public void unbind(){
        Log.e("unbind", "unbind");
        try {
            beaconManager.stopRangingBeaconsInRegion(region);
        } catch (RemoteException e) {
            e.printStackTrace();
        }
        //if (beaconManager.isBound(this)) beaconManager.unbind(this);
        beaconManager.unbind(this);
    }

    @Override
    public void didEnterRegion(Region region) {
        Log.e("didEnterRegion", "did enter region.");
    }

    @Override
    public void didExitRegion(Region region) {
        Log.e("didExitRegion", "didExitRegion");
    }

    @Override
    public void didDetermineStateForRegion(int i, Region region) {

    }

    @Override
    public void didRangeBeaconsInRegion(Collection<Beacon> collection, Region region) {
        Log.e("didRangeBeaconsInRegion", "didRangeBeaconsInRegion");
        beaconsList = (ArrayList<Beacon>) collection;
        Log.e("Size", String.valueOf(beaconsList.size()));
    }

    @Override
    public void onClick(View v) {
        switch (v.getId()){
            case R.id.bind:
                beaconSetup();
                break;
            case R.id.unbind:
                unbind();
                break;
        }
    }
}
`
IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

alin-rosu, 2018-10-26:
Please let me know when this is fixed. I am having the same issue with didRangeBeaconsInRegion not being called on an Samsung S9 and Huawei Mate 10 Pro (Android 8 both) but works on devices under Android 8

davidgyoung, 2018-10-26:
@AndroidTeam2019, can you please add `beaconManager.setDebug(true)` and attach a log excerpt that shows detections, then an unbind/bind operation followed by 30 seconds of failed detections?