AltBeacon/android-beacon-library #759: No method of setting ScanFilter device name

MikeOrtiz, 2018-10-26:
I have a BLE device that advertises as follows:

`02 01 06 03 02 F0 FF 14 09 4D 65 74 65 72 53 79 6E 63 20 42 6C 75 65 20 20 20 20 20 05 12 10 00 10 00 02 0A 00 00 00 00 00`

The library only looks at AD Types 0x16 (Service UUID) and 0xFF (Manufacturer Specific Data) when filtering devices, so it doesn't show up in scans. However, this device does list a device name (0x09), which is an applicable field for ScanFilter. Am I correct that there is currently no way to scan this BLE device with this library and it would require me to modify the library?

IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

davidgyoung, 2018-10-26:
That is correct.  This library has been built to parse and filter for popular BLE beacon formats (iBeacon, AltBeacon, UrlBeacon and Eddystone types) none of these use 0x09 AD type and the device name.

It is certainly possible to add this support, and I'd be glad to guide you if you are interested in doing the work. 

Especially if this is for a new beacon type that other might want to use, it could be a useful addition to the library and worthy of a PR to merge into the main branch.

MikeOrtiz, 2018-10-26:
Yes, I'm interested in doing the work. Here's what I think is required to add support:

- Add a new pattern to setBeaconLayout(), N_PATTERN, that expects a String.
- Modify BeaconParser.fromScanData() to check for 0x09 AD Type when matching.
- Modify ScanFilterUtils to use the name from BeaconParser when creating ScanFilters.

Is there anything else I'm missing?

davidgyoung, 2018-10-26:
Yep, that's basically it.  There is one lower layer that actually does the
parsing of the AD types -- it is the PDU.java class.  I believe you'll want
to add the 0x09 AD type there to the ones defined that the library cares
about.