AltBeacon/android-beacon-library #760: Device stuck in Crash Recovery loop

mannodermaus, 2018-10-29:
On a surface level, this problem seems to be related to https://github.com/AltBeacon/android-beacon-library/issues/282, but unfortunately that ticket couldn't be resolved due to inactivity of the original submitter.

A customer reached out to report problems using the beacon service, and upon investigation it turns out that their phone's `BluetoothService` crashes, and the attempted crash recovery doesn't seem to resolve positively. From their usage profile, it doesn't seem to be a permanent issue, but every once in a while they cannot connect, they say.

### Expected behavior
After a crash is detected in `BluetoothService`, an attempt to execute the crash recovery completes.

### Actual behavior
Sometimes, the crash recovery doesn't complete, and the phone is stuck inside a loop.

### Steps to reproduce this behavior
Currently unknown, unfortunately. We will reach out to ask if rebooting the phone, or turning the Bluetooth off/on resolves the problem for this customer. Also, will check what other Bluetooth services might be interfering on that phone.

### Mobile device model and OS version
Device: FUJITSU M02
OS:Android 5.1.1

### Android Beacon Library version
2.15

### Logs
```
(after BeaconManager#startRangingBeaconsInRegion() is called...)
2018-10-25 15:37:50.901 BLE (w): BluetoothService crash detected
2018-10-25 15:37:50.918 BLE (w): Recovery attempt started
2018-10-25 15:37:55.921 BLE (w): BluetoothAdapter.ACTION_DISCOVERY_STARTED never received. Recovery may fail.
2018-10-25 15:37:55.967 BLE (w): Skipping scan because crash recovery is in progress.
2018-10-25 15:37:55.968 BLE (w): BluetoothService crash detected
2018-10-25 15:37:57.076 BLE (w): Skipping scan because crash recovery is in progress.
2018-10-25 15:38:04.927 BLE (w): Skipping scan because crash recovery is in progress.
2018-10-25 15:38:06.035 BLE (w): Skipping scan because crash recovery is in progress.
2018-10-25 15:38:07.146 BLE (w): Skipping scan because crash recovery is in progress.
2018-10-25 15:38:08.325 BLE (w): Skipping scan because crash recovery is in progress.
2018-10-25 15:38:09.525 BLE (w): Skipping scan because crash recovery is in progress.
2018-10-25 15:38:10.725 BLE (w): Skipping scan because crash recovery is in progress.
2018-10-25 15:38:11.925 BLE (w): Skipping scan because crash recovery is in progress.
2018-10-25 15:38:13.123 BLE (w): Skipping scan because crash recovery is in progress.
2018-10-25 15:38:14.325 BLE (w): Skipping scan because crash recovery is in progress.
2018-10-25 15:38:15.524 BLE (w): Skipping scan because crash recovery is in progress.
2018-10-25 15:38:16.725 BLE (w): Skipping scan because crash recovery is in progress.
(...)
```

davidgyoung, 2018-11-07:
I put in a fix for this symptom in #768.  However, while it will make these log lines go away, you may find that it instead causes a phone in this state simply to not be able to scan for beacons.  I say this because the fact that you see these lines over and over again indicates the bluetooth chip is unable to complete a Bluetooth classic discovery.   This may indicate a problem with the bluetooth stack state  from which the library cannot help recover.

closed 2018-11-07 by davidgyoung

mannodermaus, 2018-11-08:
Thank you for the explanation!