AltBeacon/android-beacon-library #762: iOS devices distances parameters list

hadiidbouk, 2018-10-29:
Is there any list available for iOS devices like [this](https://github.com/AltBeacon/android-beacon-library/blob/master/src/main/resources/model-distance-calculations.json) list ?

I want to calculate the distance to a specific beacon using `CoreBluetooth` and `Eddystone`

davidgyoung, 2018-10-30:
Sorry, there are no such coeficients available as this library targets Android.  You can calculate then yourself if you collect rssi/distance measurements for various iOS devices.

hadiidbouk, 2018-10-30:
By following [this](https://altbeacon.github.io/android-beacon-library/distance-calculations2.html) right?

davidgyoung, 2018-10-30:
Exactly.  I did do measurements and calcs for the iPhone 4s a long time
ago, and they were almost identical to the Nexus 4.

davidgyoung, 2018-10-30:
> but the values of A, B and C are different in the ios-beacon-tools
> and the doc sheet
> <https://docs.google.com/spreadsheets/d/1iEusNfB0HgyzwPM25_oAsfNe8uxCUHix5fk1RPIeBGs/edit#gid=420885588&range=18:18>

You are right -- I trust the spreadsheet better than my memory!

closed 2018-11-07 by davidgyoung

sipandswap1234, 2020-06-17:
Is there anyway we can advertise and scan in CoreBluetooth format using the AltBeacon library through Android devices? IF not can you suggest an approach to achieve the same?

deeplathia, 2020-06-17:
Is there anyway we can advertise and scan in CoreBluetooth format using the AltBeacon library through Android devices? IF not can you suggest an approach to achieve the same?

@davidgyoung

davidgyoung, 2020-06-17:
@deeplathia please help us keep this issue on topic.  Please post your question on StackOverflow.com as that is the appropriate forum for questions on how to use the library.