AltBeacon/android-beacon-library #764: Android P system continue scanning after unbind is called

ninchan8328, 2018-11-05:
### Expected behavior
System should stop scanning after unbind method is called

### Actual behavior
System continue to scan even after unbind is called
The same code works on Android O or before

### Steps to reproduce this behavior
On Android P, when I click on the button to disable BLE, unbind method called, followed by disableForegroundServiceScanning, but the system continues to scan.

Here is the code:

    @Override
    public void onDestroy() {
        super.onDestroy();

        Log.d(logTag, "Calling onDestroy");

        Log.i(logTag, "Unbinding beacon manager");
        beaconManager.unbind(this);

        if (beaconScanner.getMode() == ScanningMode.MONITORING || beaconScanner.getMode() == ScanningMode.RANGING_MONITORING){
            stopBeaconMonitoring();
        }

        if (beaconScanner.getMode() == ScanningMode.RANGING || beaconScanner.getMode() == ScanningMode.RANGING_MONITORING){
            stopBeaconRanging();
        }
        Log.d(logTag, "Cancelling foreground notification");

        /** in Android O, i dont even have below code to cancel on notification and explicitly calling beaconManager.disableForegroundServiceScanning, i only have unbind above. **/
        NotificationManager notificationManager = (NotificationManager) getSystemService(Context.NOTIFICATION_SERVICE);
        notificationManager.cancel(beaconManager.getForegroundServiceNotificationId());

        beaconManager.disableForegroundServiceScanning();
        
        Log.d(logTag, "tag id: " + beaconManager.getForegroundServiceNotificationId());

    }

### Mobile device model and OS version
Android P on Pixel 2 XL

### Android Beacon Library version
    implementation 'org.altbeacon:android-beacon-library:2.15.2'

IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

davidgyoung, 2018-11-05:
Are you sure you have 2.15.2?    That method should be public on BeaconManager:

https://github.com/AltBeacon/android-beacon-library/blob/master/src/main/java/org/altbeacon/beacon/BeaconManager.java#L1404

Do you not see this?

ninchan8328, 2018-11-05:
Yes, as you can see from the code above, i did call beaconManager.disableForegroundServiceScanning(); but its not stopping the scanning, and i can still see the Foreground Service icon showed in notification drawer.

davidgyoung, 2018-11-05:
First, try to stop ranging and monitoring befire you call unbind.  Adduming that makes no difference,  please set beaconManager.setDebug(true); and then capture and post a log excerpt for a few seconds after calling unbind.

ninchan8328, 2018-11-06:
[androidPBleDebugFiltered.txt](https://github.com/AltBeacon/android-beacon-library/files/2551534/androidPBleDebugFiltered.txt)

Please see attached Debug log, and here is my updated code for the BLE stop

    public void onDestroy() {
        super.onDestroy();

        Log.d(logTag, "Calling onDestroy");

        if (beaconScanner.getMode() == ScanningMode.MONITORING || beaconScanner.getMode() == ScanningMode.RANGING_MONITORING){
            stopBeaconMonitoring();
        }

        if (beaconScanner.getMode() == ScanningMode.RANGING || beaconScanner.getMode() == ScanningMode.RANGING_MONITORING){
            stopBeaconRanging();
        }

        Log.i(logTag, "Unbinding beacon manager");
        beaconManager.unbind(this);
        Log.i(logTag, "Beacon Manager unbind");

        Log.i(logTag, "Cancelling foreground notification");
        NotificationManager notificationManager = (NotificationManager) getSystemService(Context.NOTIFICATION_SERVICE);
        notificationManager.cancel(beaconManager.getForegroundServiceNotificationId());
        Log.i(logTag, "Forground Service Notification cancelled");

        Log.i(logTag, "Disabling ForegroundServiceScanning");
        beaconManager.disableForegroundServiceScanning();
        Log.i(logTag, "ForegroundServiceScanning disabled");

        Log.d(logTag, "tag id: " + beaconManager.getForegroundServiceNotificationId());

    }

ninchan8328, 2018-11-06:
Tried to install the reference app, but got the following error, see screenshot
[image]

how should i resolve that?

davidgyoung, 2018-11-06:
Sorry, I mistakenly pushed the update with the "compile project" line
uncommented and the next one that loads the library from jCenter commented
out.  You can try to fix it by reversing which lines are commented out.
You may have to also comment out the equivalent project reference in
gradle.properties.  I will fix tomorrow.

ninchan8328, 2018-11-06:
Hi, i got it working by adding following:
In manifrests
    <uses-permission android:name="android.permission.FOREGROUND_SERVICE"/>

In app build.gradle
    implementation 'org.altbeacon:android-beacon-library:2.15.2'

I was able to load the app and run it couple times, first time i click on Disable button, it continue to scan, then i uninstall and install again, Disable button seems to be able to stop scanning, but the app crashed soon after.

see attached log
[BeaconReferenceAppDebug.txt](https://github.com/AltBeacon/android-beacon-library/files/2551669/BeaconReferenceAppDebug.txt)

davidgyoung, 2018-11-07:
@ninchan8328, I created this pull request as an experimental attempt to make the service stop more reliable: #767

ninchan8328, 2018-11-07:
@davidgyoung  how should i get your change to test with?

davidgyoung, 2018-11-07:
I've built a binary of this as a experimental release along with instructions on how to use it with your project [here](https://github.com/AltBeacon/android-beacon-library/releases/tag/more-reliable-service-stop).

ninchan8328, 2018-11-08:
@davidgyoung  I installed the patch, when i disable the BLE on my app, i now see the foreground notification icon is being removed, however, I still see the app performing the scan.

I tested the same on your reference app, and the app crashed like before without the patch.
See attached log files
[AndroidAltBeaconPullReq767Debug.txt](https://github.com/AltBeacon/android-beacon-library/files/2560140/AndroidAltBeaconPullReq767Debug.txt)
[AndroidPRefAppPullReq767Debug.txt](https://github.com/AltBeacon/android-beacon-library/files/2560141/AndroidPRefAppPullReq767Debug.txt)

As for the instruction, i need to add the following (flatDir) in the Project level gradle file

```
allprojects {
    repositories {
        google()
        jcenter()
        flatDir {
            dirs 'src/main/libs'
        }
    }
}
```

davidgyoung, 2018-11-08:
Thanks for the tests, @ninchan8328.

It appears that in the first log, scanning is stopped, but then a scan result comes in using the StartupBroadcastReceiver which then starts scanning again using a ScanJob.   Did you re-enable job-based scanning when you disabled the foreground service?

2018-11-07 20:45:02.491 12752-12752/com.bbapp.firstchoice I/ScanJob: scanJob version more-reliable-service-stop is starting up on the main process

In the second log, I do see the scanning stops after the service is destroyed.  The log then shows a *different* process id (11748-11748) doing scanning with a ScanJob (it also shows this process running earlier, so I think it is unrelated to the reference app.)  Bottom line, I think the reference app did stop scanning properlyl.

I'm not sure what is causing the crash in the reference app, that warrants further investigation.

see that it is the ScanJob that is doing the scan.    How is this possible that both the service and the schedule job are running?  If properly configured, the library is supposed to use one or the other.  I wonder if your setup code starts ranging or monitoring using the job scheduler (the default on Android 8+) and only later sets up a foreground service (which uses the BeaconService.)

ninchan8328, 2018-11-08:
@davidgyoung here is the code i start BLE, i dont think i enable any background scan anywhere else in my code, and the same code worked fine in my other device that is running Android 7. After i disable BLE from my app, it stop scanning immediately.

here is the code, let see if you spot anything or should i call any function to disable background scanning on Android 8+ or is there any particular function that i should search for?
```
    public void onCreate() {
        Log.d(logTag, "onCreate BeaconScanningHelper");
        beaconScanner = BeaconScanner.getInstance(getApplicationContext());
        beaconManager = BeaconManager.getInstanceForApplication(this);
        sharedPreferences = getSharedPreferences(LocalData.localStorage, Context.MODE_PRIVATE);

        appDefaultNotification = new AppDefaultNotification(this);
        Log.d(logTag, "starting BLE foreground service");
        beaconManager.setForegroundScanPeriod(Config.SHORT_SCAN_PERIOD);
        beaconManager.setForegroundBetweenScanPeriod(Config.SHORT_BETTWEEN_SCAN_PERIOD);

        /**
         * For the above foreground scanning service to be useful, you need to disable
         * JobScheduler-based scans (used on Android 8+) and set a fast background scan
         * cycle that would otherwise be disallowed by the operating system.
         */
        beaconManager.setEnableScheduledScanJobs(false);
        beaconManager.setBackgroundBetweenScanPeriod(0);
        beaconManager.setBackgroundScanPeriod(1100);

        //beaconManager.enableForegroundServiceScanning(appDefaultNotification.buildNotification(), Constants.NOTIFICATION_ID.FOREGROUND_SERVICE);
        beaconManager.enableForegroundServiceScanning(appDefaultNotification.buildNotification(), R.string.foreground_notification_channel_id);

        /**
         * simply constructing this class and holding a reference to it in your custom Application
         * class will automatically cause the BeaconLibrary to save battery whenever the application
         * is not visible.  This reduces bluetooth power usage by about 60%
         */
        backgroundPowerSaver = new BackgroundPowerSaver(this);

        beaconManager.getBeaconParsers().clear();
        beaconManager.getBeaconParsers().add(new BeaconParser().
                setBeaconLayout("m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24"));      // iBeacon Layout

        beaconManager.setDebug(true);

        super.onCreate();
    }
```

davidgyoung, 2018-11-08:
I think I see the problem [here](https://github.com/AltBeacon/android-beacon-library/blob/master/src/main/java/org/altbeacon/beacon/startup/StartupBroadcastReceiver.java).  Intent-based scans are used in Android O+ when between scan periods.  But the video that process the results (linked) always uses ScanJobs to process the results, even if the BeacinService is active as a foreground service.  This causes ScanJobs to get scheduled at the same time as the BeaconService.  Time fix this we need to change the library to:

1. Not schedule intent-based scans if using BeaconService
2. Cancel active intent-based scans if all consumers are unbound.

davidgyoung, 2018-11-08:
Upon review item 1 in my pervious comment was not necessary.  Only item 2 was necessary, and I added it to #767 and put it into the following adhoc release.

@ninchan8328, please try with this experimental release at your convenience:

https://github.com/AltBeacon/android-beacon-library/releases/tag/more-reliable-service-stop2

The above release also should fix the problem of the crashing reference app by calling stopForeground() before calling destroySelf() on the service.

ninchan8328, 2018-11-09:
@davidgyoung I tested this latest patch on both on my Android 9 and Android 7 systems, they seems to be working fine except this one time when i shutdown BLE on Android 9 it continue to scan. I then re-install the app and do more testing and can no longer reproduce the problem. So i think the solution works and i can report back if i can consistently reproduce the problem.

I think my last question will be when will you official release a newer version?

davidgyoung, 2018-11-09:
Glad to hear you had better results, @ninchan8328.  I expect to roll the next release on or before Wednesday Nov. 21.  I'm trying to get some other changes into that build, so if they are all done before then it may be sooner.  If you find other issues in your testing, please report back -- in the meantime I'll close this issue for now.

closed 2018-11-09 by davidgyoung

biswas123, 2018-12-21:
Has this been fixed? I tried on 2.15.4 version but seems the scan continues to run even when unbind is called on onDestroy() event.

aeremin, 2019-02-05:
Hi @davidgyoung , do you have any plans for the new release? It seems one in november never happened (or I am looking in a wrong place).

davidgyoung, 2019-02-05:
Yes, this change is in #767 put into release 2.15.3 in December 2018.

see:

https://github.com/AltBeacon/android-beacon-library/blob/master/CHANGELOG.md

aeremin, 2019-02-05:
Ah, I see, thanks! I was mislead by https://altbeacon.github.io/android-beacon-library/download.html which only mentions 2.15.2.