AltBeacon/android-beacon-library #769: Ranging is not available when screen is off (Galaxy note 9, Android 8.1)

wodnd23, 2018-11-08:
### Expected behavior
Should be able to Range beacons when the screen is off.

### Actual behavior
The size of the beacon list is zero, even though there is an beacon enabled around the device.

### Steps to reproduce this behavior
1. Just setup beacon scanning at 'onCreate' of my project
            mBeaconManager = BeaconManager.getInstanceForApplication(this.mContext);
            BeaconManager.setDebug(true);
            mBeaconManager.getBeaconParsers().add(new BeaconParser().setBeaconLayout(I_BEACON_LAYOUT));
            if (!mBeaconManager.isAnyConsumerBound()) {
                Log.d(TAG,"there is no consumer bound, bind new consumer");
                mBeaconManager.bind(this);

                Log.i(TAG, "Using [Android Beacon Library v" + org.altbeacon.beacon.BuildConfig.VERSION_NAME + "]");
            } else {
                Log.d(TAG,"there are consumer already bound!");
            }

2. Start ranging Beacon at 'onBeaconServiceConnect'
try {
            mBeaconManager.startRangingBeaconsInRegion(new Region("com.example.myapp.boostrapRegion", null, null,null));
        } catch (RemoteException e) {
            Log.e(TAG, "startRangingBeaconsInRegion exception!!!" + e);
        }

3. When screen is off, size of beacon list is always zero. At callback 'didRangeBeaconsInRegion'

p.s. Ranging activity of your reference app always call 'onPause' when screen is off. So I couldn't reproduce with reference app.

### Mobile device model and OS version
Samsung Galaxy Note 9 (SM-N960N), OS version : 8.1.0 (O_MR)

### Android Beacon Library version
Library version 2.15.2

[Log_ranging_not_available_screen_off.txt](https://github.com/AltBeacon/android-beacon-library/files/2562617/Log_ranging_not_available_screen_off.txt)

I've attached full log. Thank you.

IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

davidgyoung, 2018-11-09:
@wodnd23, I've been unable to reproduce this behavior using a modified version of the reference app.   The test device in my case is a Nokia 6.1 Plus running 8.1.0 (00WW_2_50A_SP01), which is an Android One device that runs stock Android.  It is certainly possible that this issue is specific to a Samsung custom ROM modification.  However, this is the only report I have heard of this issue on a Samsung device, and given how common they are, I'd expect to hear more.   I won't have access to a Samsung device with 8.1 for a few weeks, so perhaps you can re-run my exact test to see if you can reproduce on your device.  This might help eliminate any possible difference in your test code.

Also, please do capture a log line showing the library version.  If you are for some reason unexpectedly running an old version of the library, it may cause the symptoms you see.

To test this, simply clone the reference app and make the changes to the BeaconReferenceApplication.java in the diff at the bottom of this note.  The changes are basically to:

1. Disable the background power saver
2. Put the scanner into foreground mode (always scanning)
3. Turn on ranging in the default activity
4. Log each foreground / background change (so the results are obvious in the logs.)

I then run this, start an AltBeacon (you may use an iBeacon if you also add the proper parser), and move the app from foreground to background.  As you can see from the logs below, one beacon is always detected in each range cycle:

```

11-09 09:59:13.014 11707-11707/org.altbeacon.beaconreference I/ScanJob: scanJob version 2.15.2 is starting up on the main process
11-09 09:59:13.031 11707-11707/org.altbeacon.beaconreference W/ModelSpecificDistanceCalculator: Cannot find match for this device.  Using default
11-09 09:59:13.031 11707-11707/org.altbeacon.beaconreference W/ModelSpecificDistanceCalculator: Cannot find match for this device.  Using default
11-09 09:59:13.036 11707-11707/org.altbeacon.beaconreference D/BluetoothAdapter: isLeEnabled(): ON
11-09 09:59:13.045 11707-11707/org.altbeacon.beaconreference I/ScanJob: Scan job running for 300000 millis
11-09 09:59:13.045 11707-11740/org.altbeacon.beaconreference D/BluetoothAdapter: isLeEnabled(): ON
11-09 09:59:13.050 11707-11724/org.altbeacon.beaconreference D/BluetoothLeScanner: onScannerRegistered() - status=0 scannerId=9 mScannerId=0
11-09 09:59:13.215 11707-11707/org.altbeacon.beaconreference I/AssistStructure: Flattened final assist data: 2340 bytes, containing 1 windows, 7 views
11-09 09:59:13.237 11707-11751/org.altbeacon.beaconreference I/ScanHelper: Non-distinct packets detected in a single scan.  Restarting scans unecessary.
11-09 09:59:14.153 11707-11707/org.altbeacon.beaconreference D/BeaconReferenceApp: Ranged beacons with count: 1
11-09 09:59:15.266 11707-11707/org.altbeacon.beaconreference D/BeaconReferenceApp: Ranged beacons with count: 1
11-09 09:59:16.373 11707-11707/org.altbeacon.beaconreference D/BeaconReferenceApp: Ranged beacons with count: 1
11-09 09:59:17.484 11707-11707/org.altbeacon.beaconreference D/BeaconReferenceApp: Ranged beacons with count: 1
11-09 09:59:18.588 11707-11707/org.altbeacon.beaconreference D/BeaconReferenceApp: Ranged beacons with count: 1
11-09 09:59:19.708 11707-11707/org.altbeacon.beaconreference D/BeaconReferenceApp: Ranged beacons with count: 1
11-09 09:59:20.825 11707-11707/org.altbeacon.beaconreference D/BeaconReferenceApp: Ranged beacons with count: 1
11-09 09:59:21.936 11707-11707/org.altbeacon.beaconreference D/BeaconReferenceApp: Ranged beacons with count: 1
11-09 09:59:23.041 11707-11707/org.altbeacon.beaconreference D/BeaconReferenceApp: Ranged beacons with count: 1
11-09 09:59:24.051 11707-11714/org.altbeacon.beaconreference I/zygote64: Do full code cache collection, code=124KB, data=89KB
11-09 09:59:24.053 11707-11714/org.altbeacon.beaconreference I/zygote64: After code cache collection, code=113KB, data=65KB
11-09 09:59:24.151 11707-11707/org.altbeacon.beaconreference D/BeaconReferenceApp: Ranged beacons with count: 1
11-09 09:59:25.263 11707-11707/org.altbeacon.beaconreference D/BeaconReferenceApp: Ranged beacons with count: 1
11-09 09:59:26.376 11707-11707/org.altbeacon.beaconreference D/BeaconReferenceApp: Ranged beacons with count: 1
11-09 09:59:27.486 11707-11707/org.altbeacon.beaconreference D/BeaconReferenceApp: Ranged beacons with count: 1
11-09 09:59:28.609 11707-11707/org.altbeacon.beaconreference D/BeaconReferenceApp: Ranged beacons with count: 1
11-09 09:59:29.720 11707-11707/org.altbeacon.beaconreference D/BeaconReferenceApp: Ranged beacons with count: 1
11-09 09:59:30.831 11707-11707/org.altbeacon.beaconreference D/BeaconReferenceApp: Ranged beacons with count: 1
11-09 09:59:31.940 11707-11707/org.altbeacon.beaconreference D/BeaconReferenceApp: Ranged beacons with count: 1
11-09 09:59:33.054 11707-11707/org.altbeacon.beaconreference D/BeaconReferenceApp: Ranged beacons with count: 1
11-09 09:59:33.070 11707-11714/org.altbeacon.beaconreference I/zygote64: Do partial code cache collection, code=124KB, data=84KB
11-09 09:59:33.071 11707-11714/org.altbeacon.beaconreference I/zygote64: After code cache collection, code=124KB, data=84KB
11-09 09:59:33.071 11707-11714/org.altbeacon.beaconreference I/zygote64: Increasing code cache capacity to 512KB
11-09 09:59:34.163 11707-11707/org.altbeacon.beaconreference D/BeaconReferenceApp: Ranged beacons with count: 1
[... cut ...]
+        BeaconManager beaconManager = BeaconManager.getInstanceForApplication(this);
+        try {
+            beaconManager.startRangingBeaconsInRegion(region);
+        } catch (RemoteException e) {
+            e.printStackTrace();
+        }
+        beaconManager.addRangeNotifier(new RangeNotifier() {
+            @Override
+            public void didRangeBeaconsInRegion(Collection<Beacon> collection, Region region) {
+                Log.d(TAG, "Ranged beacons with count: "+collection.size());
+            }
+        });
     }
```

wodnd23, 2018-11-13:
@davidgyoung 
Sorry for late reply.
[Log_Screen_Off+Reference_App.txt](https://github.com/AltBeacon/android-beacon-library/files/2577748/Log_Screen_Off%2BReference_App.txt)

As you said, I modified the reference application to collect new log.
The results were same as when I tested with my code. Whether your app is in foreground or in background, number of AltBeacons ranged to zero when screen is off.

Please look at the following logs.

11-14 04:31:41.280 D 10157    10157    BeaconReferenceApp:                                                  Activity is in the foreground 
11-14 04:31:42.470 D 10157    10157    BeaconReferenceApp:                                                  Library version : v2.15.2, Ranged beacons with count: 1 
11-14 04:31:42.470 D 10157    10157    BeaconReferenceApp:                                                  Library version : v2.15.2, Ranged beacons with count: 1 
11-14 04:31:43.578 D 10157    10157    BeaconReferenceApp:                                                  Library version : v2.15.2, Ranged beacons with count: 1 
11-14 04:31:43.578 D 10157    10157    BeaconReferenceApp:                                                  Library version : v2.15.2, Ranged beacons with count: 1 
11-14 04:31:44.683 D 10157    10157    BeaconReferenceApp:                                                  Library version : v2.15.2, Ranged beacons with count: 1 
11-14 04:31:44.684 D 10157    10157    BeaconReferenceApp:                                                  Library version : v2.15.2, Ranged beacons with count: 1 
...

(Press home key to make activity in background, Still ranging normally)
11-14 04:32:00.797 D 10157    10157    BeaconReferenceApp:                                                  Activity is in the background 
11-14 04:32:01.365 D 10157    10157    BeaconReferenceApp:                                                  Library version : v2.15.2, Ranged beacons with count: 1 
11-14 04:32:01.365 D 10157    10157    BeaconReferenceApp:                                                  Library version : v2.15.2, Ranged beacons with count: 1 
11-14 04:32:02.469 D 10157    10157    BeaconReferenceApp:                                                  Library version : v2.15.2, Ranged beacons with count: 1 
11-14 04:32:02.470 D 10157    10157    BeaconReferenceApp:                                                  Library version : v2.15.2, Ranged beacons with count: 1 
11-14 04:32:03.580 D 10157    10157    BeaconReferenceApp:                                                  Library version : v2.15.2, Ranged beacons with count: 1 
11-14 04:32:03.580 D 10157    10157    BeaconReferenceApp:                                                  Library version : v2.15.2, Ranged beacons with count: 1 
11-14 04:32:04.699 D 10157    10157    BeaconReferenceApp:                                                  Library version : v2.15.2, Ranged beacons with count: 1 
...

(As soon as I turned off the screen, Ranging count goes to zero)
11-14 04:32:27.202 D 10157    10157    BeaconReferenceApp:                                                  Library version : v2.15.2, Ranged beacons with count: 0 
11-14 04:32:27.202 D 10157    10157    BeaconReferenceApp:                                                  Library version : v2.15.2, Ranged beacons with count: 0 
11-14 04:32:28.308 D 10157    10157    BeaconReferenceApp:                                                  Library version : v2.15.2, Ranged beacons with count: 0 
11-14 04:32:28.308 D 10157    10157    BeaconReferenceApp:                                                  Library version : v2.15.2, Ranged beacons with count: 0 
11-14 04:32:29.414 D 10157    10157    BeaconReferenceApp:                                                  Library version : v2.15.2, Ranged beacons with count: 0 
11-14 04:32:29.415 D 10157    10157    BeaconReferenceApp:                                                  Library version : v2.15.2, Ranged beacons with count: 0 
11-14 04:32:30.518 D 10157    10157    BeaconReferenceApp:                                                  Library version : v2.15.2, Ranged beacons with count: 0 
...

Please refer to the log I have uploaded, and check what the problem is.

davidgyoung, 2018-11-13:
Huh. Thanks for the tests.  

This certainly looks like an OS customization on your device.  I know I tested background ranging on the Galaxy S8 with 8.0 and it worked fine with the screen off.  I'm on travel now, so cannot test with that device again for a few weeks.  There was a change to stock Android 8.1 that blocked scans with the screen off unless a scan filter was added, so the library was modified to use a scan filter to solve this problem.  The version you are using has that change.  And I can confirm that does work on stock Android.

Do you see any system level log lines about scanning being stopped or paused when the screen goes off?  

Has anybody else seen this problem on Samsung devices?

Appesque, 2018-12-05:
I've seen this today on the exact same hardware/os version using latest lib & reference app (foreground service). As soon as the screen goes off beacons in range are 0 until screen is on again.

davidgyoung, 2018-12-05:
Thanks for this update, @Appesque. Anything you see in the logs from the OS when the screen goes off?  

What happenes if you disable the foreground service?  Does it let you scan when the screen is off at the library's regular 15 min background interval?

Appesque, 2018-12-06:
I've attached logs using a slightly modified ref app (only added log points) with foreground service on Note 9 Android 8.1 device where it returns 0 beacons in range after screen goes off. I've the working case showing first (highlighted) then the failure case where the screen went off in between. Running the same code on Samsung G955F Android 8.0 works fine. I've ensured no battery optimisation in the phone Settings either. The difference I noticed relates to the setDozeState, the Note 9 seems to use values greater than 1 whereas my logs for the S8+ phone doesn't go above 1 when the screen goes off. Perhaps a deep doze is kicking much faster on the Note9?

I'll try the background service approach if I get time today but it doesn't suit my needs unfortunately.

[Beacon Failure Note 9 Android 8.1.pdf](https://github.com/AltBeacon/android-beacon-library/files/2652357/Beacon.Failure.Note.9.Android.8.1.pdf)

davidgyoung, 2018-12-06:
Does the phone resume scanning if you move the phone (trigger the accelerometer) after the screen locks? Does it prevent going I to doze mode on screen lock if you move the phone shortly before?

Understood about your use case needing a foreground service.  The test, with background scanning however may be useful in identifying a solution if we find the scanning block is not triggered in that case.

Appesque, 2018-12-06:
Movement before or after had no effect on the outcome. Same applies to the background service, no beacons returned while screen is off. I ran the ForegroundService test over lunch and it didn't report any beacons during that time while screen was off (almost an hour).

davidgyoung, 2018-12-06:
This sure sounds like the default Android 8.1 behavior:

https://stackoverflow.com/a/48079800/1461050

However this default behavior is supposed to allow background scans if the scan has a Bluetooth filter, which was added in library version. 2.13:

https://github.com/AltBeacon/android-beacon-library/compare/2.12.4...2.13

You are certain you are running a 2.13 or later librafy version?  You can see the version sting  in the log when the BeafonService starts up.

davidgyoung, 2018-12-06:
You might try explicitly setting your library version to 2.13 in your build.gradle file.

Appesque, 2018-12-06:
I'm using latest from Master branch. Went back to use 2.13.1 and same behaviour unfortunately.

davidgyoung, 2018-12-06:
One other thought:  Are you certain you have obtained location permission at runtime?  Very similar symptoms can happen if this is not done successfully.  See the discussion [here](https://stackoverflow.com/questions/32708374/bluetooth-le-scan-doesnt-work-on-android-m-in-the-background/37015725#37015725) which notes a half-dozen pitfalls that can cause this problem.  You can verify this by going to Settings -> Apps -> [Your app] ->  Permissions and verifying you see a Location switch enabled.

davidgyoung, 2018-12-07:
Unfortunately, I don't have a Samsung 8.1 device to test with.  I have:

Galaxy S9 w/ 8.0.0 
Galaxy S8+ w/ 7.0 (No update available)
Galaxy Note 8 w/ 7.1.1

Testing with the S9 w/ Android 8.0, I see this when I lock the screen:

```
12-07 11:35:00.048 1479-1510/? I/DisplayManagerService: Display device changed state: "Built-in Screen", DOZE
12-07 11:35:00.266 1479-1510/? I/DisplayManagerService: Display device changed state: "Built-in Screen", DOZE_SUSPEND
12-07 11:37:00.329 19050-19050/org.altbeacon.beaconreference D/CycledLeScannerForLollipop: got record
12-07 11:37:00.330 19050-19257/org.altbeacon.beaconreference D/BeaconParser: Ignoring pdu type 01
12-07 11:37:00.330 19050-19257/org.altbeacon.beaconreference D/BeaconParser: Processing pdu type FF: 02011a1bff1801beac2f234454cf6d4a0fadf2f4911ba9ffa600010001c60000000000000000000000000000000000000000000000000000000000000000 with startIndex: 5, endIndex: 30
12-07 11:37:00.330 19050-19257/org.altbeacon.beaconreference D/BeaconParser: This is a recognized beacon advertisement -- be ac seen
12-07 11:37:00.330 19050-19257/org.altbeacon.beaconreference D/BeaconParser: Bytes are: 02011a1bff1801beac2f234454cf6d4a0fadf2f4911ba9ffa600010001c60000000000000000000000000000000000000000000000000000000000000000
```

So this is certainly not an issue as of Samsung Android 8.

@Appesque, one possibility is that Samsung Android 8.1 is more strict about background scanning with scan filters and requires that the scan filter be non-empty.  If you'd like, I can send you a test build of the library that creates a non-empty scan filter to see if this makes any difference on Samsung Android 8.1

Appesque, 2018-12-07:
ok let's try the non-empty scan filter - you can probably just highlight the code diff and i'll apply it for the test.

davidgyoung, 2018-12-07:
If you can compile the library from source then it's a very simple change to try a non-empty scan filter:

Here's the code in CycledLeScannerForLollipop.java:

```
            settings = (new ScanSettings.Builder().setScanMode(ScanSettings.SCAN_MODE_LOW_LATENCY)).build();
            // We create wildcard scan filters that match any advertisement so that we can detect
            // beacons in foreground mode even if the screen is off.  This is a necessary workaround
            // for a change in Android 8.1 that blocks scan results when the screen is off unless
            // there is a scan filter associatd with the scan.  Prior to 8.1, filters could just be
            // left null.  The wildcard filter matches everything.
            // We only add these filters on 8.1+ devices, because adding scan filters has been reported
            // to cause scan failures on some Samsung devices with Android 5.x
            if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.O_MR1) {
                filters = new ScanFilterUtils().createWildcardScanFilters();
            }
```

Change this:

```
                filters = new ScanFilterUtils().createWildcardScanFilters();
```

To this:

```
                LogManager.i(TAG, "Applying non-empty scan filter");
                filters = new ScanFilterUtils().createScanFiltersForBeaconParsers(mBeaconManager.getBeaconParsers());

```

The extra debug line above will let us verify in the logs that this change has taken effect.

Appesque, 2018-12-07:
Perfect, that does the trick! Thanks @davidgyoung

davidgyoung, 2018-12-07:
Wait, are you saying you have already confirmed that fixes scanning with
the screen off?

Appesque, 2018-12-08:
Correct, it fixes it on the Note 9 w/ Android 8.1. Without the fix it returns 0 beacons when the screen is off, works as expected now with the code change in place.

davidgyoung, 2018-12-08:
OK great news, @Appesque.  I will put a variant of this change into the next formal release.

I am thinking this should be a Samsung-specific change, as there are no reports of this affecting other models.  And there is a minor disadvantage to this change:

Beacons emitting an AltBeacon advertisement with a manufacturer code other than that of Radius Networks (the default) will not be matched with this scan filter unless the app developer edplicitly registers another manufacturer code with the library.  That won't affect many apps, but I don't want to break anything if I don't have to.

Along those lines, does anybody have a different Samsung device model with 8.1 (other than the Note 9) to see if this is potentially a Note-specidic issue?  Anybody have, say,  a S9 with 8.1?

Appesque, 2018-12-10:
@davidgyoung any idea when a next formal release is expected?

davidgyoung, 2018-12-10:
I'll put together a pull request with this Samsung-specific change today, than let's test again.  If no other issues, I can cut a release tomorrow.

Appesque, 2018-12-10:
Great, thanks!

davidgyoung, 2018-12-11:
Fixed in #798.  @Appesque, if you would like to give a final test of that build (which differs from what you have tested by adding the non-empty filter only if the device reports a manufacturer of "samsung"), then I will roll a release.

closed 2018-12-11 by davidgyoung

Appesque, 2018-12-11:
Thanks @davidgyoung I can confirm this patch fixes the issue on the Note 9 with Android 8.1 screen off.

paulpv, 2019-12-19:
FYI, I suspect this is not actually fixed:
https://stackoverflow.com/a/57654627/252308

> I faced the same issue. I had Scan filters in order to scan BLE devices even if the screen were locked. But on Samsung devices it didn't work, so I search on Samsung forum and I discovered Knox SDK (https://seap.samsung.com/sdk/knox-android).
> 
> And it was the solution of my problem. All you have to do is add it to your app, create a license and activate it and finally use this method addPackageToBatteryOptimizationWhiteList to unlock the scan when the Samsung device screen is lock.

If you want to scan BLE on a Samsung device when the screen is locked then you need to apply for and install and initialize/whitelist your app in their Knox SDK.

schauersbergern, 2020-05-30:
Hello @paulpv,
did you try to get a samsung knox key and implement the device admin?

I tried it with development and production key to only get errors. 

- The activate license just throws Error code unknown 206
- The addPackageToBatteryOptimizationWhiteList throws SecurityException that Admin doesn't have MDM_APP_MGMT or KNOX_APP_MGMT Permission (despite I listed these in manifest)

But maybe I have also to reach out for Samsung Support, as these functionality isn't really documented...

davidgyoung, 2020-05-30:
@schauersbergern, you are certainly welcome to pursue a Knox whitelist  workaround if needed for your use case, but I will reiterate that simply using a scan filter with the screen off is a reliable alternative, and has long been integrated with this library.

schauersbergern, 2020-06-02:
@davidgyoung, yes I know, but for weird reasons its not working e.g. on Samsung Galaxy A20e, S20, A71 and A5
When the device enters background the non empty scan filter (all props are null, manufacturer id is 76) is applied. 
The scan cycle goes on, but no beacons are detected anymore. I debugged the lib through but couldn't find a clue. I attached a log where the lib was detecting beacons, then goes into background and doesn't receive beacons anymore.
Maybe I am doing something wrong or Knox is really blocking scans in Background.

[log.txt](https://github.com/AltBeacon/android-beacon-library/files/4719197/log.txt)

paulpv, 2020-06-02:
@schauersbergern TLDR; I fought and fought and fought with getting background scans to work and never could. I had my own scanning code and pretty much gave up on it. I tried many other 3rd party libraries. I tried everything @davidgyoung suggested. I tried the Knox SDK and after days/weeks I did manage to get it to eventually install and register, but still the problem persisted.

Then on a whim I realized the problem.

The scan filter I was using filtered by name.
The names ***ARE CASE SENSITIVE***!
I was filtering for "abcd".
The device was named "ABCD".
As soon as I changed the filter to the exact casing of the device name then things started working.
Weeks/Months of effort wasted and tons of code then thrown away.
Turns out there was nothing wrong in my scanning code in this area.
It was just a wrongly cased filter the entire time.

Might this help?

schauersbergern, 2020-06-03:
Wow, @paulpv, good that you found out, persistence pays off :) 

So, AltBeaconLib is setting following ScanFilter which is not enough, if I understand that right:

BluetoothLeScanFilter [mDeviceName=null, mDeviceAddress=null, mUuid=null, mUuidMask=null, mServiceSolicitationUuid=null, mServiceSolicitationUuidMask=null, mServiceDataUuid=null, mServiceData=null, mServiceDataMask=null, mManufacturerId=4c, mManufacturerData=BE AC, mManufacturerDataMask=FF FF, mTransportBlock=null, mTransportBlockMask=null]

And if I want to receive beacons in the background I would have to just set the mDeviceName to the case sensitive name of the SENDING device (not the scanning device)?
Or do I have to also set other Scanfilter props?

So just to get the concept right. When I want to receive from multiple devices in the 
background the devices would have to send their name, which I have to save when 
receiving in foreground, for then adding that to the filters to receive beacons from
them in background? Or am I completely wrong here?

paulpv, 2020-06-03:
I never had much luck with any filter settings other than **name**.
You are probably on the wrong track that if you are working with interacting with multiple devices with different signatures/identities/fingerprints then you will need to dynamically adjust the scan filter to include anything that you know you want to work with in the background.

You should start simple though and make sure you can talk to one in the background first.
But you will need to design your code so that you can dynamically adjust the scan filter.
 
The devices already send their name in a basic BLE scan.
They may send other useful unique identifying info that you may get to work in a scan filter.
I don't have any proven advice, but don't be afraid to experiment with lots of large complex scan filters, and also with trying to be as specific as you can or as broad/general as you can to see which gives you the better desired result.

Good luck.
If @davidgyoung doesn't mind, please do chime back in here w/ your findings.
Or you can email me directly at pv@swooby.com.

schauersbergern, 2020-06-03:
@paulpv thanks for advice! I will do some testing and keep you informed about the results