AltBeacon/android-beacon-library #771: Android O ranging in background issue

AntonHellbe, 2018-11-12:
### Expected behavior
Contionusly range beacons when in proximity when app is in background / screen is off.

### Actual behavior

After somewhere between 15-30 minutes of running in the background the app is killed and restarted automatically. However when restarted it does not resume ranging beacons.

### Steps to reproduce this behavior

Perhaps I've missed something in my implementation of ranging beacons from the application class, see code below

```java

public class ExampleApplication extends Application implements BootstrapNotifier, BeaconConsumer, RangeNotifier {

    private RegionBootstrap regionBootstrap;
    private BackgroundPowerSaver backgroundPowerSaver;
    private MainActivity mainActivity = null;
    private BeaconManager beaconManager;
    private Region r;

    @Override
    public void onCreate() {
        super.onCreate();

        createNotificationChannel(
                Constants.BACKGROUND_NOTIFICATION_CHANNEL_ID,
                R.string.beacon_scan,
                R.string.beacon_scan_desc
        );

        beaconManager = org.altbeacon.beacon.BeaconManager.getInstanceForApplication(this);

        beaconManager.getBeaconParsers().clear();
        beaconManager.getBeaconParsers().add(new BeaconParser().
                setBeaconLayout(Constants.IBEACON_LAYOUT));

        // Uncomment the code below to use a foreground service to scan for beacons. This unlocks
        // the ability to continually scan for long periods of time in the background on Android 8+
        // in exchange for showing an icon at the top of the screen and a always-on notification to
        // communicate to users that your app is using resources in the background.
        //

        NotificationCompat.Builder mBuilder = new NotificationCompat.Builder(this, Constants.BACKGROUND_NOTIFICATION_CHANNEL_ID);
        mBuilder.setSmallIcon(R.drawable.my_logo);
        mBuilder.setContentTitle("Scanning for Beacons");
        Intent intent = new Intent(this, MainActivity.class);
        PendingIntent pendingIntent = PendingIntent.getActivity(
                this, 0, intent, PendingIntent.FLAG_UPDATE_CURRENT
        );

        mBuilder.setContentIntent(pendingIntent);
        beaconManager.enableForegroundServiceScanning(mBuilder.build(), 1);

        // For the above foreground scanning service to be useful, you need to disable
        // JobScheduler-based scans (used on Android 8+) and set a fast background scan
        // cycle that would otherwise be disallowed by the operating system.
        //

        beaconManager.setEnableScheduledScanJobs(false);
        beaconManager.setBackgroundBetweenScanPeriod(1L);
        beaconManager.setBackgroundScanPeriod(1100L);
        beaconManager.setForegroundBetweenScanPeriod(1100L);
        beaconManager.setForegroundScanPeriod(1L);
        this.beaconManager.setRssiFilterImplClass(RunningAverageRssiFilter.class);
        RunningAverageRssiFilter.setSampleExpirationMilliseconds(10000L);

        // wake up the app when a beacon is seen
        r = new Region("backgroundRegion",
                Identifier.parse(Constants.UUID_STRING), null, null);
        regionBootstrap = new RegionBootstrap(this, r);

        // simply constructing this class and holding a reference to it in your custom Application
        // class will automatically cause the BeaconLibrary to save battery whenever the application
        // is not visible.  This reduces bluetooth power usage by about 60%
        //backgroundPowerSaver = new BackgroundPowerSaver(this);

        beaconManager.bind(this);
        beaconManager.setDebug(true);
    }

    @Override
    public void didEnterRegion(Region region) {

        mainActivity.logToDisplay("Beacon in proximity");
    }

    @Override
    public void didExitRegion(Region region) {
        if (mainActivity != null) {

        }
    }

    @Override
    public void didDetermineStateForRegion(int state, Region region) {
        if (mainActivity != null) {
            String s = state == 0 ? "Out of distance" : "In distance";
            if(state == 1) {
                try {
                    beaconManager.startRangingBeaconsInRegion(region);
                } catch (RemoteException e) {
                    e.printStackTrace();
                }
            } else {
                try {
                    beaconManager.stopRangingBeaconsInRegion(region);
                } catch (RemoteException e) {
                    e.printStackTrace();
                }
            }
            mainActivity.logToDisplay("Beacon switched state: " + s);

        }
    }

    public void setMainActivity(MainActivity activity) {
        this.mainActivity = activity;
    }

    private void createNotificationChannel(String notification_channel, int resIdName, int resIdDescription) {
        if(Build.VERSION.SDK_INT >= Build.VERSION_CODES.O) {
            CharSequence name  = getString(resIdName);
            String description = getString(resIdDescription);
            int importance = android.app.NotificationManager.IMPORTANCE_DEFAULT;
            NotificationChannel channel = new NotificationChannel(notification_channel, name, importance);
            channel.setDescription(description);
            // Register the channel with the system; you can't change the importance
            // or other notification behaviors after this
            android.app.NotificationManager notificationManager = getSystemService(android.app.NotificationManager.class);
            notificationManager.createNotificationChannel(channel);
        }
    }

    @Override
    public void onBeaconServiceConnect() {
        beaconManager.addRangeNotifier(this);
    }

    @Override
    public void didRangeBeaconsInRegion(Collection<org.altbeacon.beacon.Beacon> collection, Region region) {
        if (collection.size() > 0) {

            Beacon firstBeacon = (Beacon) collection.iterator().next();
            double avgRSSI = firstBeacon.getRunningAverageRssi();

            if(mainActivity != null) {
                mainActivity.setTvBeaconRSSI(String.format("%.3f", avgRSSI));
            }

        }
    }

}

```

### Mobile device model and OS version

Mobile Device: Oneplus 5
OS Version: 8.1.0

### Android Beacon Library version
implementation 'org.altbeacon:android-beacon-library:2+'
2+

IMPORTANT:  This forum is reserved for feature requests or reproducible bugs with the library itself.  If you need help with using the library with your project, please open a new question on StackOverflow.com.

davidgyoung, 2018-11-12:
What do the logs say after your app resumes?  Please set beaconManager.setDebug(true); then capture a log after it resumes.

If you have another device. Please test there as well, as OnePlus has some undocumented power saving customizations from stock Android that may affect foreground service apps.

AntonHellbe, 2018-11-12:
_What do the logs say after your app resumes?_

I'm investigating that right now, I will post logs as soon as possible.

_If you have another device. Please test there as well_

Yeah, I read that in another issue posted here. Trying to get my hands on another device to test on.

Thanks,
Anton

AntonHellbe, 2018-11-12:
Here are the logs from when the app resumes
`
11-12 13:33:45.796  1310  1405 I ActivityManager: Killing 20543:se.hovding.hovding_beacon_app/u0a59 (adj 200): [BgDetect][RNK] kill se.hovding.hovding_beacon_app (uid 10059) usage 10
11-12 13:33:45.798  1310  1405 D ActivityManager: Process se.hovding.hovding_beacon_app has 1 services
11-12 13:33:45.798  1310  1405 W ActivityManager: Scheduling restart of crashed service se.hovding.hovding_beacon_app/org.altbeacon.beacon.service.BeaconService in 1000ms
11-12 13:33:45.798  1310  1407 W zygote64: kill(-20543, 9) failed: No such process
11-12 13:33:45.804  1310  1407 W zygote64: kill(-20543, 9) failed: No such process
11-12 13:33:45.806  5380  5380 D NotificationListener: onNotificationRemoved# hash: 181198927 sbn: StatusBarNotification(pkg=se.hovding.hovding_beacon_app user=UserHandle{0} id=1 tag=null key=0|se.hovding.hovding_beacon_app|1|null|10059: Notification(channel=1 pri=0 contentView=null vibrate=null sound=null defaults=0x0 flags=0x62 color=0x00000000 vis=PRIVATE))
11-12 13:33:45.809  1310  1407 W zygote64: kill(-20543, 9) failed: No such process
11-12 13:33:45.813  2980  2980 D StatusBar: updateMediaMetaData: hasArtwork = true, 1, true, true, 0, false, false, false, 0, false
11-12 13:33:45.813  2980  2980 D StatusBarWindowManager: applyFocusableFlag: true, true, true, 81860040
11-12 13:33:45.814  1310  1407 W zygote64: kill(-20543, 9) failed: No such process
11-12 13:33:45.820  1310  1407 W zygote64: kill(-20543, 9) failed: No such process
11-12 13:33:45.824  1310  2343 W InputDispatcher: channel '4feab72 se.hovding.hovding_beacon_app/se.hovding.hovding_beacon_app.MainActivity (server)' ~ Consumer closed input channel or an error occurred.  events=0x9
11-12 13:33:45.824  1310  2343 E InputDispatcher: channel '4feab72 se.hovding.hovding_beacon_app/se.hovding.hovding_beacon_app.MainActivity (server)' ~ Channel is unrecoverably broken and will be disposed!
11-12 13:33:45.825  1310  1407 W zygote64: kill(-20543, 9) failed: No such process
11-12 13:33:45.827  1310  3176 I WindowManager: WIN DEATH: Window{4feab72 u0 se.hovding.hovding_beacon_app/se.hovding.hovding_beacon_app.MainActivity}
11-12 13:33:45.828  1310  3176 W InputDispatcher: Attempted to unregister already unregistered input channel '4feab72 se.hovding.hovding_beacon_app/se.hovding.hovding_beacon_app.MainActivity (server)'
11-12 13:33:45.830   777  6221 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Received IBS_SLEEP_IND: 0xFE
11-12 13:33:45.830   777  6221 I vendor.qti.bluetooth@1.0-ibs_handler: SerialClockVote: vote for UART CLK OFF
11-12 13:33:45.832   777   777 I vendor.qti.bluetooth@1.0-ibs_handler: SerialClockVote: vote for UART CLK ON
11-12 13:33:45.832   777   777 I vendor.qti.bluetooth@1.0-ibs_handler: DeviceWakeUp: Writing IBS_WAKE_IND
11-12 13:33:45.832  1310  1407 W zygote64: kill(-20543, 9) failed: No such process
11-12 13:33:45.832  1310  1407 I zygote64: Successfully killed process cgroup uid 10059 pid 20543 in 33ms
11-12 13:33:45.832   777  6221 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Received IBS_WAKE_ACK: 0xFC
11-12 13:33:45.833   777  6221 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Received IBS_WAKE_IND: 0xFD
11-12 13:33:45.833   777  6221 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Writing IBS_WAKE_ACK
11-12 13:33:45.833  1310  1423 W ActivityManager: setHasOverlayUi called on unknown pid: 20543
11-12 13:33:45.885   777  6221 I vendor.qti.bluetooth@1.0-ibs_handler: ProcessIbsCmd: Received IBS_SLEEP_IND: 0xFE
11-12 13:33:46.814  1310  1405 I ActivityManager: Start proc 22411:se.hovding.hovding_beacon_app/u0a59 for service se.hovding.hovding_beacon_app/org.altbeacon.beacon.service.BeaconService
11-12 13:33:46.818 22411 22411 I zygote64: Late-enabling -Xcheck:jni
11-12 13:33:46.844   777  6226 I vendor.qti.bluetooth@1.0-ibs_handler: DeviceSleep: TX Awake, Sending SLEEP_IND
11-12 13:33:46.844   777  6226 I vendor.qti.bluetooth@1.0-ibs_handler: SerialClockVote: vote for UART CLK OFF
11-12 13:33:46.865 22411 22418 E zygote64: Failed sending reply to debugger: Broken pipe
11-12 13:33:46.866 22411 22418 I zygote64: Debugger is no longer active
11-12 13:33:46.868 22411 22424 I zygote64: The ClassLoaderContext is a special shared library.
11-12 13:33:46.869  5380  5380 D NotificationListener: onNotificationPosted# hash: 181198927 sbn: StatusBarNotification(pkg=se.hovding.hovding_beacon_app user=UserHandle{0} id=1 tag=null key=0|se.hovding.hovding_beacon_app|1|null|10059: Notification(channel=1 pri=0 contentView=null vibrate=null sound=null defaults=0x0 flags=0x62 color=0x00000000 vis=PRIVATE))
11-12 13:33:46.871  2980 22426 I MediaPlayer: constructor
11-12 13:33:46.877   940  9760 D MediaPlayerService: LF player type = 4
11-12 13:33:46.877   940  9760 D NuPlayerDriver: NuPlayerDriver(0xf2428e00) created, clientPid(2980)
11-12 13:33:46.880   940  9760 I ExtendedNuUtils: printFileName fd(8) -> /data/system_de/0/ringtones/notification_sound_cache
11-12 13:33:46.884   940 22432 D GenericSource: FileSource remote
11-12 13:33:46.889   936 20938 I MediaExtractor: MediaExtractor mime application/ogg flags 2
11-12 13:33:46.889   936 20938 I ExtendedExtractor: QTIParser is prefered .. flags 2
11-12 13:33:46.889   936 20938 I QComExtractorFactory: QTI parser is not preferred for mime: application/ogg
11-12 13:33:46.889   936 20938 E QCExtractor:  ExtendedExtractor failed to instantiate extractor 
[... cut ...]
11-12 13:33:47.018 22411 22411 D BluetoothCrashResolver: constructed
11-12 13:33:47.020 22411 22411 D BluetoothCrashResolver: Read 0 Bluetooth addresses
11-12 13:33:47.021 22411 22411 D BluetoothCrashResolver: started listening for BluetoothAdapter events
11-12 13:33:47.023 22411 22411 I CycledLeScanner: Using Android O scanner
11-12 13:33:47.026 22411 22411 I BeaconService: beaconService version 2.15.2 is starting up on the main process
11-12 13:33:47.026 22411 22411 D BeaconLocalBroadcastProcessor: Register calls: global=1 instance=1
`

AntonHellbe, 2018-11-12:
After googling around regarding the ActivityManager and OnePlus this definitely seems like a OnePlus issue, do you agree David?

chintan-mishra, 2018-11-12:
I am experiencing a similar issue with AltBeacon. It appears once the app is closed(by returning to the home screen or killing the app from the overview window) then the OxygenOS kills any services and cancels any pending jobs within few seconds.

@AntonHellbe how long does it take to detect the first beacon on OnePlus?

AntonHellbe, 2018-11-13:
Monitoring seems to startup correctly after the service restarts. However ranging does not.

Here is the log when the service starts up again: https://pastebin.com/vRK0NhKw

chintan-mishra, 2018-11-13:
@AntonHellbe can you try setting `mBeaconManager.setEnableScheduledScanJobs(false);` and noticing the behaviour

AntonHellbe, 2018-11-13:
Hi,

`mBeaconManager.setEnableScheduledScanJobs(false);` is already set. See the code I posted in the issue description.

Br,
Anton

AntonHellbe, 2018-11-13:
Hi,

I noticed that too when I reworked the application class, currently running tests to see if it works better now.

Br,
Anton

AntonHellbe, 2018-11-14:
Hi David,

After changing the didDetermineStateForRegion according to the following:

```java

    @Override
    public void didDetermineStateForRegion(int state, Region region) {
        String s = state == 0 ? "Out of distance" : "In distance";
        if(state == 1) {
            try {
                beaconManager.startRangingBeaconsInRegion(region);
            } catch (RemoteException e) {
                e.printStackTrace();
            }
        } else {
            try {
                beaconManager.stopRangingBeaconsInRegion(region);
            } catch (RemoteException e) {
                e.printStackTrace();
            }

        }

        if(mainActivity != null) {
            mainActivity.logToDisplay("Hovding Beacon switched state: " + s);
        }
    }

```
Ranging is now restarted when the application is killed by the OS. I have confirmed on an Sony Xperia X (F5121) running Android O (v8.0.0) that the foreground service is not killed. It seems like its OxygenOS that behaves differently when it comes to services.

davidgyoung, 2018-11-14:
Ok, good, @ AntonHellbe.  Just so I understand the killing behavior you see caused by OxygenOS... The OS kills the foreground service after 15-30 min or it starting, then after X minutes the service restarts itself.

How long is X?  Does this continue indefinitely?  If so, other than the dropouts when scanning is not happening during period X, are there  any other negative effects?  From the user perspective, does the foreground service notification just blink off and on during period X?

Answers to the above can at least let me document what library users can expect on OxygenOS.

AntonHellbe, 2018-11-14:
I will try to explain the behaviours I have noticed on OxygenOS

- The foreground service can run for anything between 1-50 minutes (50 minutes is the longest I observed, could be longer when I haven't paid attention).
- When it's killed the service starts within a couple seconds again, the old notification dissapears and is replaced with a new one (No blinking)
- Occasionally OxygenOS does a "force stop" on the foreground service, i.e it's killed completely and does not restart. When that happens a "High battery usage" notification appears telling that _MyApp_ has high battery usage.
- The "force stop" requires you to manually start the app from the launcher again

davidgyoung, 2018-11-15:
Thanks.  And the typical time between launch and the force stop / Hugh Battery usage notification?  Is it like one hour or one day?

AntonHellbe, 2018-11-29:
I apologize for the late answer.

The high battery notification is very sporadic, sometimes it appears several times an hour. Sometimes once an hour.

AntonHellbe, 2018-12-06:
Closing the issue.

Br,
Anton

closed 2018-12-06 by AntonHellbe

android-bash, 2019-08-16:
@AntonHellbe So what worked for you. How are you making sure Foreground service will still scan beacons for long time say 10 hours even when screen is off.